dresden-elektronik/deconz-rest-plugin #877: Lost all lights and sensor

jeopold, 2018-10-17:
The other day, I noticed that all lights, switches and sensors suddenly were gone. When I logged in to my deCONZ docker installation, it looked like a fresh installation of the Conbee stick without any nodes. 
In Phoscon, all the lights were gone, but the sensors were still in the configuration, marked as unavailable though. 

I have checked the zll.db for differences in the network settings, and it looks like the PAN ID has changed, but changing back to the same value as in the zll.db does not help. 

I also noticed that the Network Key differs, but I´m not able to enter the one from the zll.db.
(The Network Key in zll.db starts with 0000, but it looks like i´m forced to start with 0x in deCONZ settings. I´m not sure if this even is the problem. 

I tried to reset some nodes with touch link reset, and then they re-appear with their old names. 

With 50+ nodes in the network, including some outdoor lights and switches with fixed power wires, it´s not really an option to do a touchlink reset to get all nodes back. 

deCONZ version 2.05.34, firmware version 0x261F0500.

manup, 2018-10-17:
The network should be still working, some notes:

* as you noticed the network key from zbconf table must be prefixed with 0x 
* panid and channel should also be configured
* press Save
* close the network settings dialog
* press Leave
* press Join

Sorry that it is currently so cumbersome, in the future this should be automated or better not happen at all. Technically the firmware can check all channels on startup and look where the network is, and select the matching one from the zbconf table.

jeopold, 2018-10-18:
The nework key is displayed as 00006c644800b4a4d864bcb4fcfc9848 when I open the zll.db in "DB Browser for SQLite", is it mabye displayed in another format? 
[image]

It would really be useful with a backup/restore function, else this would result in hours of work for restoring. (And the hard wired nodes requires an electrician to demount).

manup, 2018-10-18:
>It would really be useful with a backup/restore function, else this would result in hours of work for restoring. (And the hard wired nodes requires an electrician to demount).

There is a backup/restore function in *Phoscon App > Settings > Gateway*. It is not perfect yet but  handles this case to restore network settings.

>The nework key is displayed as 00006c644800b4a4d864bcb4fcfc9848 when I open the zll.db in "DB Browser for SQLite", is it mabye displayed in another format?

They are both the same and have hex format, you only need to prefix it with `0x` when putting it in deCONZ settings.

jeopold, 2018-10-18:
Thank you for sorting that out, I had not noticed the backup feature. 

It worked with restoring the network key (starting with 0x), so the network is now up again with all nodes back. And with a fresh backup :)

Hopefully i wont suffer from any further crashes.

closed 2018-10-18 by jeopold