dresden-elektronik/deconz-rest-plugin #878: unable to edit some TRÅDFRI wireless dimmers in PWA

ErnstEeldert, 2018-10-17:
I have 4 TRÅDFRI wireless dimmers. Two of them I have been able to configure just fine. The other two however, show up in PWA, but if I click them in the groups overview, instead of showing me the group control screen, it just jumps to the last 'real' group.

So I can't assign lights to those dimmers. Previously, they had an old firmware version, but I've hooked them up to the IKEA gateway to update (as that didn't happen using the OTA plugin). Repaired to deconz, but the same issue remains.

Can I configure these dimmers manually by leveraging the REST API?

Kane610, 2018-10-18:
I have the same issue on my one dimmer. It worked with an old firmware, gave it to a colleague to update firmware. Re-added it to deCONZ and now I can't select it in Phoscon. It does send events over web socket though.

ErnstEeldert, 2018-10-22:
Not sure what fixed it, but I seem to be able to edit them again. An upgrade and restart of deconz has happened in the mean time, but I don't know if this had anything to with it?

closed 2018-10-22 by ErnstEeldert

manup, 2018-10-22:
There were some fixes related to end-devices in general might be a side effect of this.