dresden-elektronik/deconz-rest-plugin #879: After Beta 41 no sensor is working

heitec11, 2018-10-18:
After installing the newest version all my sensors are unreachable.
Pairing the sensor after gateway reset or sensor delete ia also not possible.

manup, 2018-10-18:
After starting deCONZ they should be reachable again as soon the next signal is received.
Note: for Xiaomi sensors this can take up 1 hour.

>Pairing the sensor after gateway reset or sensor delete is also not possible.

Do you mean a gateway powercycle? A reset of the gateway from Phoscon App Gateway > Settings would break the the complete network.

wvuyk, 2018-10-18:
@manup is it not possible the report the last known status instead of marking sensors unreachable at the start of deCONZ? This was this way before 2.0.5.36 or so I think? It is now first reporting unreachable and thus triggering events in my plugin once a first report comes in, which in fact report that last known status most often. Because of that events can trigger in my plugin that did not need to be triggered...

manup, 2018-10-18:
>@manup is it not possible to report the last known status instead of marking sensors unreachable at the start of deCONZ?

Yes that will come back in a improved version, it was buggy in the past, I had sensors shown as available which weren't powered for months.

>This was this way before 2.0.5.36 or so I think? It is now first reporting unreachable and thus triggering events in my plugin once a first report comes in, which in fact report that last known status most often. Because of that events can trigger in my plugin that did not need to be triggered...

Understand, I'll see to bring it back as soon as possible.

manup, 2018-10-18:
>Yes, I have 3 different Xiaomi sensors. But nobody of them is reachable until now, which meams 4 hours waiting time. See response.txt
>To delete all 3 sensors and try a new pairing was also not possible. The gateway (Phoscon App & Conbee) only count the 3 minutes without any results.

Did you reset the devices while sensor search is active by pressing the buttons until the LED blinks?

heitec11, 2018-10-21:
I did go back to Ver 39, but still no communication with the sensors. Also it is not possible to add /pear a new Xiaomi sensor. The only thing which is not back to the old version is the firmware (26270500).
How can I downgrad the firmware?

heitec11, 2018-10-22:
With Beta 42 now one of my three Xiaomi sensors are back.
[response_42.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2501020/response_42.txt)

heitec11, 2018-10-22:
With Beta 42 one of my three Xiaomi sensors are back.
[response_42.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2501030/response_42.txt)

manup, 2018-10-31:
Closing this issue to keep discussion in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/912

closed 2018-10-31 by manup