dresden-elektronik/deconz-rest-plugin #881: Support for Osram Smart+ Motion Sensor

SwoopX, 2018-10-18:
Hi,

It would be great if those could be supported as well.

[image]

[image]

Let me know if anything else might be required.

Thanks!

manup, 2018-10-18:
I have this sensor running in my home for two weeks now.
Currently it needs some manual fiddling to get it working, the plan is to add automated implementation in the next versions.

SwoopX, 2018-10-18:
Sounds great, thanks. I'll put them aside for the time being.

fabsfuchs, 2018-10-20:
Hey could you please describe the "manual fiddling" a little bit further? I added the sensor and its reporting changes in the presence. I can see this through the deCONZ GUI. When I am trying to get the sensor data through REST with a request. The answer is always the following:

`{
    "config": {
        "battery": 100,
        "duration": 60,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "ae3fc0e6c983647d214c2489b8b97e11",
    "manufacturername": "CentraLite",
    "modelid": "Motion Sensor-A",
    "name": "Bewegungsmelder",
    "state": {
        "lastupdated": "none",
        "lowbattery": null,
        "presence": null,
        "tampered": null
    },
    "swversion": "\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd\ufffd",
    "type": "ZHAPresence",
    "uniqueid": "00:0d:6f:00:0e:05:89:73-01-0500"
}`

So the interesting values are always set with null.

manup, 2018-10-21:
The most important step is configuring the ConBee as CIE in the motion sensor

* Open IAZ Zone cluster in *Cluster Info* Panel
* Double click on the `IAS_CIE_Address` attribute to open the attribute editor
* Write the mac address of your ConBee/RaspBee in attribute

[image]

This should be enough to make the motion sensor work.

I have also created bindings of the following clusters to the gateway via the Binding Dropbox:

* IAS Zone cluster (might not be necessary)
* Temperature cluster
* Power configuration cluster

[image]

fabsfuchs, 2018-10-21:
Thank you very much for the fast response. After the CIE configuration it works.

I am a little bit confused about the reporting of the presence value (unoccupied). I think there are two values to influence this behaviour. The first is the "PIR Occupied to Unoccupied Delay" and the second the "config.duration" value of the sensor. 

For testing purposes I set the "PIR Occupied to Unoccupied Delay" to value 2 and the "config.duration" via REST API to the value 60 (default I think).
The lower outputs are received from the websocket:

**Unoccupied to Occupied:**
`
{ e: 'changed',
  id: '3',
  r: 'sensors',
  state: 
   { lastupdated: '2018-10-21T19:17:49',
     lowbattery: false,
     presence: true,
     tampered: true },
  t: 'event' }
`
**Occupied to Unoccupied:**
`
{ e: 'changed',
  id: '3',
  r: 'sensors',
  state: 
   { lastupdated: '2018-10-21T19:18:21',
     lowbattery: false,
     presence: false,
     tampered: true },
  t: 'event' }
presence = false
`

The difference is 32s. When I observe this in the deCONZ GUI I can see the the Occupancy value being 0x01 for just 2s. Exactly how I set it up. Why takes it 32s to report it via the websocket?

derBroBro, 2019-01-14:
Are there any options to connect the sensor just by the rest module?

manup, 2019-01-15:
Currently the GUI is needed to do the missing steps as described above.

Looking forward to add these steps to the automatic join process, as well as the temperature cluster setup, in the next two releases.

manup, 2019-01-22:
The sensor is supported in 2.05.57 and added in the wiki supported devices section.

closed 2019-01-22 by manup

Jasonthefirst, 2019-02-20:
I am not sure if the update added the sensor. I am not able to pair the Osram smart+ motion Sensor to my deconz gateway. I am currently on 2.05.59 and did a reset of the sensor. When searching for a sensor device (others) it won't find the osram sensor.

manup, 2019-02-20:
The following sequence is needed to join the sensor:

* Pull out battery
* Start sensor search in Phoscon App
* Press the sensor button and keep it pressed
* Insert battery
* Release the button

After 5–30 seconds the sensor should be added.

tas-group-01, 2019-02-21:
I added the Sensor to the deconz phoscon app with the latest version. What about the „????????“ within the Data Code Line?
I also don‘t get Temperatur Values updated within the Phoscon App. It displays always 20 C. Is there a hardware, software or config Problem with updating the Temperatur Values?

quotschmacher, 2019-06-27:
Are there any plans for a configuration of the sensor via the deConz GUI?

manup, 2019-06-27:
What do you mean by configuration? You can use the sensor in the Motion sensor control of the Phoscon App. https://phoscon.de/en/app/doc#group-edit-sensors-en

quotschmacher, 2019-06-27:
The package says something about configuration via app. As I previous had a Tradfri motion sensor I thought one could configure something like a Lux-level or a switching duration.

manup, 2019-06-27:
The sensor doesn't provide light measurement only motion trigger and temperature measurement.
The duration can be configured as described in the link above.

thaefner, 2019-11-11:
@manup What does the duration config do? Does it control the detection delay, like 300 is 5 min? Mine is set on 90 by default, same as the battery, so i am a bit confused. It can be updated and set correctly, but it does not seem to change anything.

thiroxx, 2020-03-10:
Any chance to get rid of the tampered flag?

peer69, 2020-04-08:
> Any chance to get rid of the tampered flag?

This annoys me as well.

7wells, 2020-09-19:
> The following sequence is needed to join the sensor:
>     * Pull out battery
>     * Start sensor search in Phoscon App
>     * Press the sensor button and keep it pressed
>     * Insert battery
>     * Release the button
> 
> After 5–30 seconds the sensor should be added.

Hello, I followed these steps and can see the Osram Smart+ Motion Sensor in Deconz - see screenshot:
https://workupload.com/file/CpmxJ3L6ZKk

In the right window (no screenshot), it appears without any connections/lines (but joined to the network). I don't know whether sensors should be shown with connecting lines or not but would like to mention it.

In Phoscon, I cannot add it as sensor and it does not appear there under sensors.

Do you have any hints for me how to get it really working? Thank you so much!

Smanar, 2020-09-19:
Sure wich one deconz version are you suing ? You need a version >= 79.
You have made the inclusion with deconz and not phoscon ? If yes, just set phoscon in permit join (add new sensor), then read again the basic attribute in deconz (need to wake up the switch in the same time)

Smanar, 2020-09-19:
It s possible this device is invisible in phoscon.
Do you have third application connected to deconz, or have you check in the API ?

7wells, 2020-09-19:
There are no 3rd party apps connected to Deconz. What exactly should I check in the API? And what do you mean by "capture from this issue with your configuration"?

Smanar, 2020-09-19:
Yep to fill the fields, you need to press the "read" button, and make the device react to wake it up. The name is not important, better to compare the values in the table.

Without third applications, you need to check direclty the API for exemple with `http://IP:PORT/api/API_KEY/sensors`

Cf https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

7wells, 2020-09-19:
Can I get the API key from Phoscon or Deconz?

Smanar, 2020-09-19:
If you haven't the "node info" panel, on the menu "channel" then check "node info". The other panel can be enabled here too.

On the previous link you have a way to obtain an API key else with curl
`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

But if you have all like the capture, no reason for your sensor don't be created, it s probably invisible and only usable with third applications, but for me it s strange, it s an osram sensor ....

7wells, 2020-09-19:
Do you know of other cases of invisible devices and if/how this was fixed?

What could be a reason for a device like a sensor from a known manufacturer (like Osram) to not be visible on Phoscon but visible on Deconz?

Which 3rd party app would you recommend? I have Android devices at home but would like to primarily use the switches. The normal-sized Osram Switch works nicely, but the mini switch does not (a different topic, though).

7wells, 2020-09-19:
Has any other owner of an Osram Smart+ Motion Sensor found a way to get it working in Phoscon? How?

If I had to allow a 3rd party app access to my Phoscon gateway, how secure would this be, and which one would be recommendable?

Smanar, 2020-09-19:
>What could be a reason for a device like a sensor from a known manufacturer (like Osram) to not be visible on Phoscon but visible on Deconz?

Good question ^^, the phoscon dev have perhaps not included it yet, but as you said, it s a big manufacture, so wait for some other return, perhaps the problem is from deconz.

But if you can't see the device in "sensor" part in API, the problem is from deconz, and using third will change nothing.

I think deconz is supported by all third app possible ^^, depend of what you need, something light, something complete, something easy to use, power user, ect ....

But first, try to see if the device is realy invisible, or wait for some days for a return from an other user that have the device.

7wells, 2020-09-20:
I have checked with `http://IP:PORT/api/API_KEY/sensors` and am unsure what I should expect. I see there a "Dailight" entry, which (I assume) is the virtual Phoscon sensor for timers following daylight. And then there are 4 switches (which I find peculiar, as I thought only sensors should appear when using the trailing `/sensors` in the link. The number of 4 switches is correct, as I have 2 normal-sized (4 buttons) and 2 mini (3 buttons) Osram Smart+ Switches.

But there is not the Osram Smart+ Motion Sensor that appears in Deconz but not in Phoscon. So if it does not appear with the aforementioned method `http://IP:PORT/api/API_KEY/sensors`, what exactly does this mean?

And yes, I have a similar problem with the 3-buttons mini switches: They do appear both in Deconz and also in Phoscon under switches, but they are not listed when I try to apply a light to a switch - then only the 2 normal-sized (4 buttons) switches are listed for selection. Maybe it is better to have a separate issue about this, though.

Smanar, 2020-09-20:
OK so it mean propably the API don't support it. But it s strange, it seem it have worked before ...

Just can you compare, the model ID, the IEE address (need to start by 0x000DEF) and the manufacture code (need to be 0x104E)

After that, as your device is already in deconz.
- Set phoscon in permit join (add new sensor)
- In deconz, read basic attribute (the same thing you have already done to have the model ID, the "read" button in cluster 0x0000 with making the device awake.

It will trigger an inclusion.

But if after that, the entry is still not created in the sensor part in the API, I m sorry but I think we will need to use log.

Remark:
There is only "sensor" and "light"section, so switch are too in "sensor"

7wells, 2020-09-20:
Deconz shows for the Motion Sensor correctly `Manufacturer Code: 0x104e`, but `IEEE: 0x000d6fxxxxxxxxxx`, i.e. this starts not with `0x000DEF`. What does this mean? Where can I find the `Model ID`? Or did you mean another place (not Deconz) for these parameters?

7wells, 2020-09-20:
The "Cluster Info" tab in Deconz is always empty, even why I click on any object in the right-side window. Is this normal?
Thanks also for your remark about the "sensor" and "lights" sections.

7wells, 2020-09-20:
> After that, as your device is already in deconz.
>
>     * Set phoscon in permit join (add new sensor)
>
>     * In deconz, read basic attribute (the same thing you have already done to have the model ID, [...]
Well, the sensor is shown in Deconz, and there is no change in comparison to yesterday:
https://workupload.com/file/CpmxJ3L6ZKk

It still does not appear in Phoscon, though.

7wells, 2020-09-20:
RESOLVED! 👍 

I have manually updated both the Deconz FW and Phoscon according to this guide on my RaspBee II module:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

With these files:
http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26650700.bin.GCF
http://deconz.dresden-elektronik.de/raspbian/beta/deconz-latest-beta.deb

Then I physically powered off all lights. Actually, I assume that this was the trick and vaguely remember to have read somewhere that for pairing the Osram Smart+ Motion Sensor to Phoscon, all devices within 10 metres around the Phoscon gateway should be physically powered off before inserting the battery into the sensor (holding it close to the gateway) and while pressing the little button on the side of that sensor and releasing it after the battery is inserted. Obviously, other powered devices/lights that are connected to the Phoscon gateway seem to disturb a correct pairing of that sensor (at least that is my guess).

Now it is of course not 100% clear whether also the update described above was really needed, and I am really sorry about this. But at least every user/owner of this sensor should be able to get it working. Maybe you first want to try it with your current firmware and Phoscon version and only update of that fails.

BTW, the sensor immediately appeared in Deconz with the name `Bewegungsmelder` (English: motion sensor) in the "Node Info" tab. Before, this was always empty.

7wells, 2020-09-20:
There is still a minor issue, i.e. the manufacturer is not shown as Osram (but it should), but as CentraLite, and the version number is not correctly shown (question marks instead). See screenshots:
https://workupload.com/archive/frGDMgZq

Can this also be resolved, please? Again, thanks for your/everybody's help! :-)

peer69, 2020-09-20:
> There is still a minor issue, i.e. the manufacturer is not shown as Osram (but it should), but as CentraLite, and the version number is not correctly shown (question marks instead). See screenshots: [...]

Same here. Not really an issue for me just some visual imperfection. 
I have included all osram sensors like any other sensor. I did not turn off any lights or moved them close to the gateway prior to the inclusion. Nobodies what might have caused your issue but the osram sensors are working great for me, even better than the philips hue ones I have.

7wells, 2020-09-20:
Yeah, maybe there was something "special" in my case, e.g. the problem sitting in fron of the PC/gateway. ;-)

Does the Osram Smart+ Mini Switch work for you (in case you have it)? See there:
https://github.com/dresden-elektronik/phoscon-app-beta/issues/64#issuecomment-695767666

Smanar, 2020-09-20:
Lol, this device seem working since the version 57, So I haven't asked wich one version you had ...

The IEE was my fault too, you are right, was 0x000d6f.

For "CentraLite", it s normal, it s the value send by the device itself, so deconz use the same.

7wells, 2020-09-22:
Sorry, but I am not sure if I understood you right: Do you mean that the Osram Smart+ **Mini Switch** works for you?
I know that there are other threads* for it, but wanted to be sure what you meant. *E.g. there:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2763#issue-616507995
https://github.com/dresden-elektronik/phoscon-app-beta/issues/64#issue-568358368

Smanar, 2020-09-22:
Yep it works, perhaps not in phoscon, but it works.

7wells, 2020-09-22:
That's nice to know! :) Would you mind sharing a few hints how I can accomplish this? Sorry if the answer is obvious. :blush:

Smanar, 2020-09-22:
From discord, working 2/3 days ago

```
    1. First, remove & reinsert the battery, 
    2. open the add sensor window in phoscon
    3. then button 2 & 3 to reset, button 1 & 3 till blue led blinks...will go solid red in a second
    4. after that button 1 & 2 for three seconds.. till the blue light blinks...
    5. press some buttons on the remote
    6. phoscon should display ready
    7. no device will show in phoscon, but you will see the remote listed in deconz
```

7wells, 2020-09-22:
But the steps you have shared are to show the mini switch in deconz, where I can see it already. The problem is that the switch does also show up under switches in phoscon, but not under switches in phoscon where a switch should be selected for a group of lights.

And the text you have quoted is for the 4-buttons switch, which works already fine for me.

Now I am still curious to know how people really use the mini switch if it cannot be assigned to a group of lights in phoscon.

Do you or others use a 3rd party app that works with deconz to get them paired?

Smanar, 2020-09-22:
Oups sorry, I have cumulated 2 users.

For the group feature, have you tested the old web app ?

On my side I never "pair" device with each other, all is managed by the third app, and I haven't this device to test, so I can't say you if the working mode is normal or not with phoscon.

7wells, 2020-09-22:
Thanks for the hint about the old app, which I will try. And I might also try the Hue Essentials app. Finally, I would like to check if I can use my Logitech Harmony remote control (with hub).

7wells, 2020-09-22:
Wow, the Logitech Harmony recognises all lights and plugs out-of-the-box, and I can switch them on/off, and in/decrease the lights.

7wells, 2020-09-22:
Sorry, this thread here is actually for the Osram Smart+ Motion Sensor.

For issues about the Osram Smart+ Switch Mini, please let us continue there:
https://github.com/dresden-elektronik/phoscon-app-beta/issues/64