dresden-elektronik/deconz-rest-plugin #883: [Request Device Support ] Legrand Netatmo device

Smanar, 2018-10-18:
Hi, I have a new device who works strangely, All seems good, but the led on the device stay red (go off on normal full commissioning) and on deconz it go to yellow then red.
It's a device from the netatmo gamme from Legrand https://www.netatmo.com/partners/legrand.
I m making my test with the switch without neutral, see as light and not switch (I think it's normal because it control light on/off+ dimmer)

[image]
[image]
[image]

IDK if there is a special protection or not, I can send a complete sniff make with ubiqua from the power on to the final led change make with the offciel centrale. Here a sample
[image]

Smanar, 2018-10-20:
I have make a second try with sniffing with conbee key, and there is a lot of missing request.
[image]

Missing:
Request key
Transport key B
Verify key
Confirm key

Mgmt permit join request is not send by the device too after the key echange.

manup, 2018-10-20:
Looks like the communication and security is alright since there is some exchange of messages on ZDP level.

But the switch leaves the network for some reason, so I guess it waited for specific commands or configuration (binding?). Question is which :)

Smanar, 2018-10-20:
Yep, but on the first "sniff" (this one with the official central) there is only classic request, so I think the specific waiting command isn't an exotic command like a manufactured one.
But not using the 2 keys echange can be the problem, and it seems it s not implemented in this version yet. ATM Deconz only use a light key echange protocol.

Smanar, 2018-10-21:
I m comparing the 2 captures and I can say
- the device leave 15 s after the association request on ConBee

- On the both capture we have
-- device announce
-- Node descriptor request+Response

- Only on the working
-- Basic read attribute request+response
-- Match  descriptor request+response
-- IEE address + response
-- OTA upgrade request+response
-- A message permit join request ( ? but it s from device to central so useless I think)
-- Add group/scene
-- Bind request+response (20s after the association request so I have stoped here)

- Only on the no working
-- Active Endpoint request+response
-- Management lqi request+response
-- Simple descriptor request+response

On the layer NWK, on the working mode We have
link status centrale > broadcast
Route request 0x270F centrale > broadcast
Route request 0x270F all devices> broadcast

I will some tries with more request with conbee forcing some commands like ota or attribute read.

manup, 2018-10-21:
>-- Add group/scene
-- Bind request+response (20s after the association request so I have stoped here)

These are the ones I meant :) the switch might wait for this configuration to be done.
Can you please provide the sniffed details of these requests?

Smanar, 2018-10-21:
Hu !
I m seing something, the device have the adress BD86, so it's the device itself who ask for "basic read attribute" (Request n 159) to the device router, not the coordinator. And this request is make in the first trame, so perhpas the blocking request is just between the commissioning frame (Request n 96 ) and this one.
In fact in normal mode the device is more active.

I have others devices ikea bulb, usb device, there is a way to force the installation to work like the original one ? In fact the central have a buildin device inside, it s 'not possible to use the original central without the router, the prb is perhaps from here.

I can send you the captured packet in ubiqua format if you have a public email ? But there is too much differences betwen the working mode with and without router.

Smanar, 2018-10-21:
Yeah ^^.
I have make a try with an ikea bulb as router, and It works ^^, at least the on/off, not the dimmer but on normal usage you need to enable this option with application.

I will post a sniff soon, but it 's random, it works 1 time on 5.

Smanar, 2018-10-21:
[image]

The sniff during a working process with Deconz. I can 't explain what happen, the device try 1 time, leave , retry and it works.
But on the same sniff, I have a working try and a not working.

Smanar, 2018-10-21:
Now I need your help, this is the missing cluster/attribute I need to control it

```
Frame Information: (53 bytes)
	Packet ID: 13
	Protocol: ZigBee
	Date: 2018/10/21
	Timestamp: 15:31:03.061918
	Time Delta: 2.172160
	Channel: 11
	Length: 53
	RSSI: -90 dBm
	Source: COM9: NXP JN51xx / Rainforest Toucan@1000000
	Layer: ZCL
	Status: Decrypted
MAC Header: (9 bytes)
	Frame Control: 0x8861
		···· ···· ···· ·001 = Frame Type: [0x1] Data
		···· ···· ···· 0··· = Security Enabled: [0x0] No
		···· ···· ···0 ···· = Frame Pending: [0x0] No
		···· ···· ··1· ···· = Acknowledgement Request: [0x1] Yes
		···· ···· ·1·· ···· = Intra-PAN: [0x1] Yes
		···· ··00 0··· ···· = Reserved: 0x0
		···· 10·· ···· ···· = Destination Addr Mode: [0x2] 16-bit Short Address
		··00 ···· ···· ···· = Reserved: 0x0
		10·· ···· ···· ···· = Source Addr Mode: [0x2] 16-bit Short Address
	Sequence Number: 14
	Destination PAN ID: 0xDAAD
	Destination Address: 0x2EBE
	Source Address: 0x0000
MAC Payload: (42 bytes)
	NWK Header: 0x081400002EBE0248
		Frame Control: 0x0248
			···· ···· ···· ··00 = Frame Type: [0x0] Data
			···· ···· ··00 10·· = Protocol Version: [0x2] ZigBee Pro
			···· ···· 01·· ···· = Route Discovery: [0x1] Enabled
			···· ···0 ···· ···· = Multicast Flag: [0x0] Unicast or Broadcast
			···· ··1· ···· ···· = Security Enabled: [0x1] Yes
			···· ·0·· ···· ···· = Source Route Included: [0x0] No
			···· 0··· ···· ···· = Destination IEEE Address Included: [0x0] No
			···0 ···· ···· ···· = Source IEEE Address Included: [0x0] No
			··0· ···· ···· ···· = End Device Initiator: [0x0] No
			00·· ···· ···· ···· = Reserved: 0x0
		Destination Address: 0x2EBE
		Source Address: 0x0000
		Radius: 0x14
		Sequence Number: 8
	NWK Aux Header: (14 bytes)
		Network Security Control: 0x28
			···· ·000 = Network Security Level: [0x0] None
			···0 1··· = Key NWK ID: [0x1] Network Key
			··1· ···· = Extended Nonce: [0x1] Yes
			00·· ···· = Reserved: 0x0
		NWK Frame Counter: 1572954
		Source Address: 00:04:74:00:00:18:32:58
		NWK Key Sequence Number: 0
	NWK Payload: (16 bytes)
		APS Header: 0x03010104FC010100
			Frame Control: 0x00
				···· ··00 = Frame Type: [0x0] Data
				···· 00·· = Delivery Mode: [0x0] Normal Unicast Delivery
				···0 ···· = Acknowledgement Format: 0x0
					Format: [0x0] Data Frame
				··0· ···· = Security Enabled: [0x0] No
				·0·· ···· = Acknowledgement Request: 0x0
					Request: [0x0] No
				0··· ···· = Extended Header Present: [0x0] No
			Destination Endpoint: 0x01
			Cluster ID: [0xFC01] Private Cluster
			Profile ID: [0x0104] ZigBee Home Automation
[... 92 lines cut ...]
			Profile ID: [0x0104] ZigBee Home Automation
			Source Endpoint: 0x01
			APS Counter: 23
		APS Payload: 0x01100001022F10
			ZCL Header: 0x022F10
				Frame Control: 0x10
					···· ··00 = Frame Type: [0x0] Command Acts Across the Entire Profile
					···· ·0·· = Manufacturer Specific: [0x0] Manufacturer Code Not Included in the ZCL Frame
					···· 0··· = Direction: [0x0] From Client to Server
					···1 ···· = Disable Default Response: [0x1] Yes
					000· ···· = Reserved: 0x0
				Transaction Sequence Number: 47
				General Command Frame: [0x02] Write Attributes
			ZCL Payload: 0x01100001
				Attribute 0: 0x01100001
					Attribute ID: [0x0001] Reserved
					Data Type: [0x10] Boolean
					Unknown Data Type: 0x01
	NWK MIC: 0xEBD68619
MAC Footer: 0xB231
	Frame Check Sequence: 0xB231
```

Data 01 > enable led in dark
Data 00 > disable led in dark

manup, 2018-10-23:
I've checked the code, this datatype is not yet supported in the deCONZ GUI, I'll add it in the next version.

Smanar, 2018-10-23:
Lol, I have spend 1 hour trying all datatypes.
I think this device can work "out of the box" with somes litles hacks. I will wait for next version to continue my tests.

BTW thx a lot for answers :)

Smanar, 2018-10-26:
Hu, just for information I m seing on "about deCONZ" option, I m using this one from the website, it s the V2_05_20
And from https://www.dresden-elektronik.de/deconz/win/ I m seing there is more in date version. Do you want I make tries with this version or better to wait for version > V2_05_42

Smanar, 2018-10-29:
Hello, I m trying to use the last version 2.05.44, and I can't edit the general.xml file. I m trying to put my code on lot of place (and domain) but the cluster stay "Unknow".
Even if I try to replace a working cluster (eg 0019) by mine (fc01), it disable the OTA cluster (now unknow) but not enable mine.

Smanar, 2018-10-29:
Hi, I m making tries with the version 2.05.44, it's better but when I try to write some data, it's always "0x0000" on sniff. Do I need to wait for a future version, or it's another problem ?

manup, 2018-11-03:
Cool, can you please submit a PR for the general.xml addition?

>Hi, I m making tries with the version 2.05.44, it's better but when I try to write some data (datatype = dat16, other are working), it's always "0x0000" on sniff (or 0x0100, the last read value).

Do you mean the write attribute isn't filled with the data provided in Cluster Info Panel?

Smanar, 2018-11-03:
Exactly, I can put any values in the editbox, if I press write, I have the message "sucessfull", but the ConBee always send the same value at the device, and if I press the "read button" it come back to the old value again.
The problem is not from the device, because I have "snif" the request and it rly send the same value whatever is the value in the editbox.

Sure I will make the PR, but I m waiting for the GUI is working again, because without this correction, impossible to enable dimmer without deconz-cli-plugin, so my addition can't work.

Smanar, 2018-11-14:
Just tested 2.05.47, bug still here, GUI don't support DAT16.
And I haven't checked but it seem there is exactly the same prb for int24 cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/812

manup, 2018-11-14:
Yes sorry hasn't made it yet, looking forward to fix it in the next release around Friday.

Smanar, 2018-11-15:
No problems, I will make tries on futures versions.
But this github is for only the rest plugin or all the deCONZ application ? Because it's a bug in the GUI, nothing to see with the plugin.

Smanar, 2018-11-19:
It works, thx again for your app and support :).

So to resume, It s a switch with wire, but without neutral (zero line) with on/off AND dimmer.

- if you use it "out of the box" it works without problem but I don't think dimmer will work.
- if you update its firmware with the gateway, it will be harder to pair, you need to use another device as router to make it possible, I have used an ikea bulb.
- You need to use the attribute in special cluster to enable dimmer

After that, the device is see as bulb (not a switch) so easier for integration, it works without problem in phoscon too.

closed 2018-11-21 by Smanar

Smanar, 2019-06-02:
Ok so just for information the values I m using for attribute report.

For on/off : I have set 1 - 300
For level control : I have set 1 - 600 - 0

WhistleMaster, 2019-08-31:
So are all the Legrand connected switch series are now supported with this change ?

Why is it seen as a light instead of a switch ?

Smanar, 2019-08-31:
Only the wired one, the batteries ones not.

Yes, it's a light, not a switch, switch send command, light receive command, this one receive command to enable/disable power.

I haven't tried the micro module, but it's the only one, that work out of the box (and without firmware update) on other solution, so I think it will works too, but I haven't confirmation yet.

But Take care :

***********************************************
YOU NEED GATEWAY TO UPDATE THE DEVICE FIRMWARE TO ENABLE DIMMER.
***************************************************

Else you will have only on/off.

WhistleMaster, 2019-08-31:
Thanks for your insight on that !

> I haven't tried to add the manual part in the API code to make it automatic, because I have realy few return for this device.

I'm not sure to understand. What do you mean by adding the manual part in the API ?

Smanar, 2019-08-31:
2 things:
1 - the binding, with it you will have state return, on every manual action on the switch (else you need pooling)
2 - Set the attribute report, else the gateway will spam the switch for attribute.

These 2 actions can be include in the API code, to make them automatic during pairing.

WhistleMaster, 2019-08-31:
I see. How do you set those two manually then ?

Smanar, 2019-08-31:
For bind:
- Go In "Bind dropbox panel"
- Drag and drop endpoint 01 from Node to "Source" square.
- Drag and drop endpoint 01 from gateway to "Destination" square.
- Press "Bind"

For attribute setting
- Go in cluster 0006 "on/off"
- Go on attribute, the table on bottom
- Select attribute 0x0000
- Press Read config
- Set new config
- Press Write config

ATM I m using this config
For on/off : I have set 1 - 300
For level control : I have set 1 - 600 - 0

For information, the Legrand plug are working, but without consumption return, and this is not possible without changing code (need whitelist), so I m waiting for that to change the switch code in same time. but:
1 - No one have ask for it, and I don't like spend time for nothing.
2 - I need someone with the deconz GUI and the device to make test (I haven't one)

WhistleMaster, 2019-08-31:
Thanks for the clarification !

> 1 - No one have ask for it, and I don't like spend time for nothing.

Well, I'm interested to have the Legrand switches fully integrated with Deconz !

Smanar, 2019-08-31:
Lol, I was speaking for the plug, for the switch you can do that manually, only 1 time, and it's not an obligation, the switch can work without it.
You are the first one who ask me something about this manufacture and I m on other deconz projects too.

WhistleMaster, 2019-08-31:
I got you ! I thought you were asking for the switches.

I'm using Home Assistant (HA) with Deconz, so without doing the binding and the attribute setting, would the switch work correctly with HA (states reporting, etc.) ?

I'm interested by this brand because of the price tag and also because it's a well know brand.

Smanar, 2019-08-31:
Depend of home automation application for bind, attribute is just a bonus, like I have said Phiips hue are spamming too.

Yeah it's a well know brand, with realy serious hardware and support (for quality it's another story), it's the world n 1 in switch and plug with 20% of the world market, but no-one ask for this kind of device ^^. Perhpas because of price (and because you need the gateway too).

You already have the device or not ?
Have you tried to make the binding yourself ?

I will take a look on the code tomorrow, give me some time to see, if I can make something fast.

tocks53, 2019-09-09:
Hello,

I wish to install dooxies in my house.
I would like to know if the following modules will be compatible with deconz.

600087.600085.600081, 600698

For 600087 will there be the opening in percentage of the shutters?

Thank you for the work done in the integration of Legrand products.

Smanar, 2019-09-09:
Was not so hard, all was working naturally, the harder part was sniffing the packet to find the special attribute to enable the dimmer.
And like I have said before, the integration is not full yet, there is some part that need to be manually.

6 000 81 : yes it's the one I have (but remember,you need the gateway to enable dimmer)
6 006 98 : Yes, but I don't think you have consumption report, I haven't the device to make tries but for that I think you need to gateway to update the firwmare (like for the dimmer) and put this plug in whitelist in the API code.

For other not, I have never see a poject with netatmo battery device, the wired shutter switch can work, but the battery one, I think not, but I haven't tried it.

If you wana try the dimmer switch or the plug, I can make modification on the code in "blind mode" with your help for tests, I have already take a look on the code, will not be hard, but I can't test it on my side.
But for batteries devices, I can not guarantee anything, and I m not optimistic.

And no, there isn't percentage of the shutters, it's just up/down/stop.

Xylomid, 2019-09-11:
Hey man, just letting you know that we appreciate your work and you're not the only one interested in having Legrand hardware integrated in Deconz.

Do you know if i can, with only a wireless legrand dimmer switch (no hub), trigger a deconz_event in home assistant ? Thanks

Smanar, 2019-09-11:
wireless legrand dimmer switch ?
There is a wireless dimmer ? Wireless switch are just for on/off ?

Xylomid, 2019-09-11:
My bad, wireless are only on/off. That's the one i'm talking about

Smanar, 2019-09-11:
Ha, so I think no.

No-one have already tried on deconz, but on other zigbee project, nobody has ever succeeded to include a netatmo battery device. I haven't tried on my side too, always for the same reason, price ^^.

BTW, for information, if someone want to help me for better Netatmo integration (like consumption  for plug), I will be on holiday in some days and will leave my country. So I will be back in 15 days.

Smanar, 2019-09-28:
So still no-one ?
Ok, so if someone want a better integration, I need someone with the hardware (and the centrale for the plug, I don't think it's usefull for the wired shutters switches).

xmillies, 2019-10-01:
For the moment, I have tried only the preinstalled SD-card image on Raspberry Pi :

Raspbian Buster Desktop
"Stabiles Image mit Desktop basierend auf Debian Buster."
deCONZ 2.05.69
Version 9-2019
released 08/09/2019

If I remember I see the plug (not with last firmware). I will check as soon as possible.

Smanar, 2019-10-01:
Ok, on my side I have a starting code > https://github.com/Smanar/deconz-rest-plugin
Do you know how to compile it ? I can send you my compiled file but it s perhaps plateform dependant (and I m still on Strech).

So on this version, the plug will be on whitelist, so the values will not be good, but you will see values on websocket return.
I don't know wich one value the plug return, and I assume the model id is "plug.Celiane.Legrand", I think it will be good for you but not for dooxie users.

To compare, you can use this issue > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/812

The code will do the binding itself, but you can test params too if my modification don't working

For information, on the code I have used
if (lightNode->manufacturerCode() == VENDOR_LEGRAND) > I m sure this one will be good, If i m right your plug MAC start with 000474.....
sensor->modelId() == QLatin1String("plug.Celiane.Legrand") > for this one I m not sure at 100%, it's just deduction.

xmillies, 2019-10-02:
You are on RPI debian strech ?
I 'll try on the preinstalled SD-card image on Raspberry Pi (=Buster).

Smanar, 2019-10-02:
Yep, and I have compiled the file, I can send it if you can't compile it yourself, but compilation are platform dependant so not sure my version will work on your system (logicaly yes, compilation on older system work on newer).
Just tell me if you need the file ?

xmillies, 2019-10-03:
I'll try with Buster.

Smanar, 2019-10-03:
It's not a problem, just tell me if you can compile the file yourself or if you need I send you a compiled one to try.

xmillies, 2019-10-04:
The compilation has worked. I have switch the plugin.
I 'll try to do test this week-end.

Smanar, 2019-10-04:
Nice, if I m right you will receive data (even with bad value) on websocket event.
If not, I can be wrong at
- Your MAC adress don't start by 000474.
- Your model id is not "plug.Celiane.Legrand" (case sensitive I think)
- I have not finished the automatic atribute reporting setting (can be done manualy for tests).
- Need more setting for attribute reporting.

Smanar, 2019-10-06:
BTW, tell me too if you have problem to include them, I know it's possible, because I know user that have already succed with that, but ATM I have someone in PM that have problem to include them after firmware update by the gateway (like my problem when I have started this issue).

xmillies, 2019-10-07:
Hi , here some screenshots with an plug Legrand  677 7.5

xmillies, 2019-10-07:
The second plug (I think with firmware updated)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

xmillies, 2019-10-07:
In Phoscon we see the 2 firmwares.
[image]
[image]
[image]

xmillies, 2019-10-07:
On previous screen, you can see **active power** value.
- 2 for phone charger
- 14 for a laptop and his charger

xmillies, 2019-10-07:
For the unit 2, on the Legrand app I can see version 32d(ecimal) = 20h(exadecimal).
That what we can see on last screenshot.

Smanar, 2019-10-07:
Ok so good news, all seem standard, I don't think legrand use his proprietary cluster for plug.

Bad new now, I have used "plug.Celiane.Legrand" for model id, I have take this value on other site, but on your screen the good value is "Connected outlet", so IDK for binding but attributes setting have failed for sure.
So i have made 2 modification on code
https://github.com/Smanar/deconz-rest-plugin/commit/8a74cbd08a099e5f7f714f0391b91bd3d4ed3c1d#diff-70a87389db32b8f306683c8362093bfc
https://github.com/Smanar/deconz-rest-plugin/commit/524bf4ffa6af9490c651b5b29742e6cae0088ce8#diff-c941fc017347f7109bf6036ba9505576
Sorry, you need to recompile, and remake inclusion for the api reconfigure the plug.

After that, if you use domoticz, just enable logging "debug info only", and you will see new field in plug json.

If not.
- Take a look if setting attribute reporting have worked, in the cluster "electrical mesurement", double clic on the attribute "050B Active power", it will popup the attribute reporting window, press the read button. If I m right you will have 1/300 as values.
- If yes the problem was the binding.

xmillies, 2019-10-07:
I 'll try this evening if I can. No problem to recompile it. So simple.
For the moment I just use Raspbee SD Image.
I 'll probaly use later Home Assistant (or Domoticz). 

To make the test. I have just recompile, switch plugin. Restart raspbee ?
No need to setup again plugs ?

xmillies, 2019-10-07:
Do you think we will get power summation in cluster or elsewhere as in Legrand App (Home +) ?
In Legrand App we could have consumption by day , by month.
or for last 30 days by plug or items.

Smanar, 2019-10-07:
You need to re-include them, for the API remake the plug configuration.
If you have problem for inclusion, this hardware is realy hard to include, there is a trick to test
>>"Next, open the network from the old web app, or search for devices from Phoscon. Then in the Node Info panel, read the Basic cluster attributes. That should trigger the creation of the /lights resource."

BTW, if you don't use home automation application, IDK where we can read the JSON return, log are realy verbose to read, I have a plugin in firefox to connect to a websocket, but the easier way is using a home automation application (if you need it later), you can take the one you prefer, deconz work on all of them ^^ (Home assistant is a realy good one for me too)

>>Do you think we will get power summation in cluster or elsewhere as in Legrand App (Home +) ?
In Legrand App we could have consumption by day , by month.
or for last 30 days by plug or items.
And it's for that you need an home automation application, this plug just return the active power (I don't see summation in cluster), so it's the application that memorise values and make table for days or month. You need one to replace the Legrand one.

In all case, alls applications will use the websocket return to work, so to test if the modifications are working you need to read the websocket information, with an home automation app, simple app, plugin in browser.

Smanar, 2019-10-07:
You have another way to check if it works, if you dont want to install application yet.
Just ask API for sensors, with something like 
`http://deconz_api/api/your_aoi_key/sensors`

xmillies, 2019-10-10:
I'll try this evening. 

Good news : I get today another Legrand item :-) 

Legrand - 064879 - Sortie de câble connectée

Smanar, 2019-10-10:
Ha yep, realy nice device, I think i will buy it too, for my "dry towel", I just need to check for dimension, the box is realy small.

xmillies, 2019-10-11:
Some picture with "sortie de câble".
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Is there another way to contact you (PM, ...) ?

[645 more replies]