dresden-elektronik/deconz-rest-plugin #885: Missing switch editor?

jariivanoff, 2018-10-19:
Hi!
I installed the released deconz-2.05.20-qt5.deb, on a Ubunut 18.04 server and got it working, but then I read about the switch editor that would make it possible to actually use the remotes I got fully with all the buttons (IKEA Tradfri mainly). So I upgraded to deconz-dev-2.05.39 expecting to get it since  2.05.39  was marked as the latest release here on github. No switch editor, so I thought I'd go a bit wild and crazy and upgrade to 2.05.41 but still same result... no switch editor. So, now I am wondering if I have to make some config editing somewhere?
[image]

manup, 2018-10-19:
Looks like the cached version, to force cache reload append a bogus query string starting with a `?` in url bar like this:

    index.html?1538983867000

jariivanoff, 2018-10-19:
Ahh.... thanks! That solved it!

closed 2019-01-17 by manup