dresden-elektronik/deconz-rest-plugin #887: Almost no devices are meshing 2.05.41 and 26270500

ACrazyConcept, 2018-10-19:
Is it on purpose that 2.05.41 and 26270500 causes almost no devices to mesh?
Only one device out of 68 has more than one connection.
Things seem to be working but it used to look very different.

ACrazyConcept, 2018-10-20:
> I would suggest restart deCONZ, meshing should work normally maybe some queues are full?

I have rebooted and restarted several times. It also takes a long time for any green connection lines to appear.
What are these queues?

Filialen, 2018-10-24:
I have the same problem. 2.05.42 and 0x26280500.
When I first tried .42 all lights were gone in Phoscon and I had to do a restore...
And when I tried to downgrade, both DeCONZ and firmware, it still was very sluggish.
Like if something broke the database?
It does mesh but it is very unreliable and very slow!

donnib, 2018-10-24:
@manup any ideas?

pipiche38, 2018-10-24:
Excuse me for a stupid question, but is the meshing the business of the Device itself. How can DeConz could interact and disable it ?

manup, 2018-10-24:
Working on it since  few days, the problem is deep nested and needs more testing and verification the stack. Until it's solved I recommend reverting back to the latest working version:

    sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x261f0500.bin.GCF

donnib, 2018-10-24:
@manup so i downgraded but the ui shows very few lines and behaves weirdly ui wise. I am running latest deconz and restapi from master.

wvuyk, 2018-10-24:
@manup, how do windows users do this for a Conbee?

manup, 2018-10-24:
@donnib give it a while it should settle after some minutes (I reckon your network was >40 devices).

@wvuyk I put something up tomorrow, did it happen in your setup as well?

I need to change my test networks at work, they are >90 lights and work very stable but contain mostly FLS devices which all have power amplifiers, guess this hides such issues.

donnib, 2018-10-25:
@manup Yes you are right, i have many devices, i have 56 devices. Even after 10hours it sill does not show lines. Be aware that i am still running 2.05.42 deconz and latest master of the rest api.

manup, 2018-10-25:
That took a while...
Today I've installed a new mixed test network with around 40 routers from IKEA, OSRAM, Innr, Philips, Paulmann and Paul Neuhaus and 3 end-devices.

Then I've checked the firmware versions 0x261f0500 — 0x26280500.

The biggest issue was from 0x26240500 onwards:
* Networks with more than 26 routers stopped to send link status messages (needed for routing). In my home network the bug never occurred because I have only 16 routers.
* It seems some RAM memory was corrupted at NWK layer
* The stack itself did work but weird things were going on 

#### Fixed version
Firmware version 0x262b0500 should fix all of that, it will be part of the next release 2.05.43 in one or two days.

It's running in 4 test networks including the new mixed one now with no issues.

https://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x262b0500.bin.GCF

#### Linux

    sudo GCFFlasher_internal -f deCONZ_Rpi_0x262b0500.bin.GCF

#### Windows

    GCFFlasher -d 0 -f deCONZ_Rpi_0x262b0500.bin.GCF

There are still some open issues with routing which will be improved in further versions, like that routes in larger networks are too volatile and should be more sticky.

donnib, 2018-10-25:
@manup so i installed the firmware and still the lines in deconz are not shown although i can use the REST api and use the lights. Do i need to do a complete power recycle ? There was a command to reset the Raspbee but i forgot what it was, i can try that if it helps. What could be the problem since no lines what so ever are shown ?

Filialen, 2018-10-25:
@donnib The reset command is sudo GCFFlasher_internal -r

@manup  But it is still an issue here too. Still very slow and unresponsive...
I do get lines in deCONZ, but it is painfully slow. And some routers do not get any lines at all. (At least not for quite some time...)
I've got 40 routers, IKEA and Hue, and 12 end-points, IKEA, Hue and Xiaomi.

Is there something else in deCONZ that could have been broken?
What I did when I installed 0x26280500 and it felt broken, was a restore from a backup about a week old.
Since then all firmwares behaves like this...

manup, 2018-10-25:
@donnib is query routers activated in your deCONZ?

[image]

manup, 2018-10-25:
@Filialen this might be a different case

How long is the new firmware running?

I'm not sure if it's the case here but one thing that you can check is the **NWK Update ID**

[image]

Currently the backup won't detect if the Update ID from backup is lower when the otherwise active Update UI used by the network.

For example:
* If you create a backup while Update ID is 1
* Then change channel, the Update ID in the network becomes 2 (there could be also other reasons)
* When restore the backup with Update ID 1, this might cause problems since the still active network expects 2 or higher

This will be handled automatically in future versions, but currently the only way to fix this is to
raise the Update ID, Save + Leave + Join.

Filialen, 2018-10-25:
@manup my Update ID is 5. How do I know what to raise it to?

donnib, 2018-10-26:
@manup yes i have. 

I have some of the lights off but i see the same as @Filialen some are online and have no lines and none of the mesh either. See screenshot :

[image]

[image]

manup, 2018-10-26:
Can you please check the zll.db file with sqlitebrowser, here the table zbconf for modified configurations.
We also need to check the Update ID as mentioned above.

manup, 2018-10-26:
>@manup my Update ID is 5. How do I know what to raise it to?

Currently you can only guess, try, 6, 7, 8 .... it just needs to be higher.

donnib, 2018-10-26:
@manup i am not sure what to look for
[deconz.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2518397/deconz.zip)

In regards to Update ID, it was 4 and i increased to 5 but no changes.

manup, 2018-10-26:
Damn, same problem as in the other network yesterday

[image]

Please do the following:
* Open deCONZ network settings
* Set `APS Use Ext PAN ID` to all zeros
* Set `NWK Update ID` to 7
* Save
* Leave
* Join

[image]

You may have to power-cycle the lights afterwards if they don't get it on their own (not sure).

I'll add something to detect and fix the issue in code.

donnib, 2018-10-26:
@manup Right after i did above lines started to show in the UI between nodes e.g meshing was starting to work so i guess that solved it, thank you very much.

wvuyk, 2018-10-27:
@manup,

Trying here to get my raspbee updated witht he new firmware as Phoscom is also not updating this firmware anymore. Is it an issue when you have both RaspBee and ConBee active?

Anyway, when I try the command you posted, is get this:

`pi@raspberrypi:~ $ sudo GCFFlasher_internal -f deCONZ_Rpi_0x262b0500.bin.GCF
rmmod: ERROR: Module ftdi_sio is not currently loaded
rmmod: ERROR: Module usbserial is not currently loaded
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
no device specified use RaspBee (/dev/ttyAMA0)
using firmware file: deCONZ_Rpi_0x262b0500.bin.GCF 
error opening firmware file deCONZ_Rpi_0x262b0500.bin.GCF
pi@raspberrypi:~ $ 

`
I am a real nitwit in this area and don't really get all meanings of errors here. The only thing I am guessing is that there in indeed no USB, as this is an raspbee, with an internal card. 

I was guessing I will need to start from the directory of GCFFlasher_internal, so I did a search on the raspbee and found many of these files in /usr/bin/X11/X11/[many of X11!!]. so this did not help either to solve it? See image for this.

Then I decided to do the same for the Windows/ConBee machine. GCFFlasher was easy to determine and I performed the command
`GCFFlasher -d 0 -f deCONZ_Rpi_0x262b0500.bin.GCF` as admin by hand. I did this, knowing the version 2.05.43 is not yet available for Windows. Downloaded the firmware and placed it in both the firmware directory of my installation as well as the C:\Users\[user]\AppData\Local\dresden-elektronik\deCONZ\firmware
But also here I am getting an error:

`E:\deCONZ\bin>GCFFlasher -d 0 -f deCONZ_Rpi_0x262b0500.bin.GCF
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26
using firmware file: deCONZ_Rpi_0x262b0500.bin.GCF
error opening firmware file deCONZ_Rpi_0x262b0500.bin.GCF`

I am now guessing this is also happening for Phoscon under the hood? What can I do to get the firmware update actuall applied?

Wim

manup, 2018-10-27:
@wvuyk yes the deCONZ must be closed prior flashing

>E:\deCONZ\bin>GCFFlasher -d 0 -f deCONZ_Rpi_0x262b0500.bin.GCF GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26 using firmware file: deCONZ_Rpi_0x262b0500.bin.GCF error opening firmware file deCONZ_Rpi_0x262b0500.bin.GCF

Either the full path to firmware must provided or you just put it in the same directory as GCFFlasher is.

manup, 2018-10-27:
>It worked immediatly for the raspbee, but it did not for the conBee. As it appears, the same windows machine is also holding a Zwave stick and a usb port reader for my smart meter. Had to remove them all, adjusting the device number did not work here..... Is that maybe the reason for Phoscon not working on the ConBee?

Yes this might be possible. You may check the output of `GCFFlasher -l` to list all devices if the others are also FTDI based the "right" one needs be selected.

>And what is the matter with the repeating X11 direcotries on the RaspBee? Or should I just stop worrying here?

I don't know but it is some OS specif stuff, shouldn't be important on the matter.

Filialen, 2018-10-28:
@manup Just leaving a last comment in this thread...
Everything is working as usual again! :)
Turned out to be my normal 2.4 Ghz network that suddenly (after 1.5 year) decided to interfere with my Zigbee net! Locked it down and everything instantly worked smoothly again.
So there really were no problem, that I know of in my case, with the firmware!
Anyway, big thanks for your help! You're doing a fantastic job!
It's fun to live on the edge! :)