dresden-elektronik/deconz-rest-plugin #888: Quirky+GE Tapt switch support in Dimmer/Level Control Mode

jsb5151, 2018-10-20:
I have a few Quirky+GE Tapt switches and they work just fine with deCONZ (2.05.40 with RaspBee), even though they all show up as lights.  These switches have 2 endpoints; one is a ZHA Dimmer Switch and the other one is a ZHA On/Off output.  Normally, only the On/Off output endpoint is used and the buttons control the load, but there is a way to reconfigure the switch to operate as a dimmer for smart bulbs (e.g. keypresses do not control the load).

I've configured one of my switches to do just that (zcl change attribute command on cluster FC20, mfg code 0x5488, attribute 0, u8 value 2 instead of 0), but I'm not getting anything on the WebSocket port, the deCONZ logs or the CLI interface when I press the buttons on the switch.  I used a ConBee to sniff the ZigBee traffic and I can see the switch is sending Level Control messages (output cluster 8) as well as On/Off messages (output cluster 6) on endpoint 1.    It seems that deCONZ does not interpret these messages... is there a way to get this working at all?

I want to be able to interpret these keypresses to perform certain actions, similar to a Lutron Connected Bulb Remote, for example pressing a button will trigger an action in Home Assistant.

Screenshot of the clusters:

[image]

Thanks in advance.

ebaauw, 2018-10-20:
That looks pretty straightforward, actually.  The switch needs to be whitelisted, so a `/sensors` ZHASwitch resource will be created for endpoint 1, next to the `/lights` resource for endpoint 2.  And we need a buttonmap, mapping the commands the switch sends on buttonpresses to `buttonevent` values.  Please provide the info as described in the Wiki, 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

jsb5151, 2018-10-20:
See screenshot in the 1st post for the list of available clusters on both endpoints.

Basic Cluster Attributes (both endpoints are the same):
[image]

Node Info:
[image]

The switch operation mode can be changed as described above by changing attribute 0x0000 in cluster 0xFC20 on endpoint 1 with manufacturer-specific code 0x5488, as described above.  Default value "0" works like a standard on/off switch hooked up to Mains and report their status using endpoint 2.  Changing that attribute value to "2" makes it operate as a smart switch/keypad; the buttons no longer turn on/off the load, and start reporting Level Control & On/Off messages from endpoint 1.  Other modes are available (e.g. upper button toggles load and lower button is smart, and vice versa); I will document them once we get this working.

I can share Wireshark/ZShark traces if required at a later time; I need to find the transport key first :-)

(btw is there a way to check the transport key from the deCONZ GUI?)

jsb5151, 2018-10-20:
One additional comment: my test switch is configured in mode 2, but I can still operate the load as a light in deCONZ/HASS which is exactly what I want.  I originally configured that switch so friends & family wouldn't be able to electrically turn off my Hue lights, but I could still toggle/reset the switch load from Home Assistant or deCONZ if need be.  So supporting both the sensor and the load at the same time would be great!

jsb5151, 2018-11-07:
Any news on this?  Do you need any further information, traces, debugs?

jsb5151, 2019-01-15:
Testing a modified version here; will submit a PR if this works as expected... it's dirty at the moment but it should work.

closed 2019-09-19 by stale[bot]

aque0us, 2020-02-02:
a

aque0us, 2020-02-02:
I keep getting an error that 0 is an invalid attribute when I attempt to change the load, any advice?