dresden-elektronik/deconz-rest-plugin #889: Help requested connecting two Philips Hue dimmers

Paul-dH, 2018-10-20:
I'd like to ask for some help connecting my two Hue Dimmers to Deconz. I've been trying to connect them for over a week now but It wont succeed...

The steps I've tried:
- Started a search for a Hue switch in the Phoscon App.
- Hold reset button for 10 secconds and place the dimmer next to the ConBee.
> Nothing is found and the search times out.

- Started a search for a Hue switch in the Phoscon App.
- Hold the ON button for a while, once the led turns red and place it nest to the ConBee.
> Nothing is found and the search times out.

- Started a search for a Hue switch in the Phoscon App.
- Hold reset button for a seccond and release when it turns orange.
> Nothing is found and the search times out.

Then I'd tried to connect the dimmers back to the Philips Hue Bridge. The bridge found them and I could use them to turn of a spare Hue Light. After a hour or so I removed the devices and tried the above steps again to connect them to Deconz. No luck...

What other steps can I take to connect the dimmers?

reprz, 2018-10-23:
I´m using 2.05.39 and just realized that the UI won´t update firmware. I´ll try the recommended method for firmware update and try pairing the dimmer again.

My Conbee is attached straight in Raspberry USB-connection. Might have to try the extension method also.

Paul-dH, 2018-10-23:
I'm using 2.05.40 and my setup consists of a ASUS Tinker Board with Raspbian installed. I'm not yet able to run the Gui trough VNC but all webpanels work (Phoscon and the old one). I found somewhere that the Deconz GUI and service can not run headless and provide a GUI at the same time.

I'll also try to upgrade first..

Paul-dH, 2018-10-23:
I've upgraded to 2.05.42 but I'm still unable to join my Hue Dimmers, are there some other steps that I can take or provide log info?

manup, 2018-10-23:
Which firmware version do you have installed?
It's visible in Phoscon > Menu > Settings > Gateway.

Paul-dH, 2018-10-23:
The firmware is:  26280500

reprz, 2018-10-23:
Well that went well. I upgraded to 2.05.40 (Hassio has difficult upgrade method) and reinstalled Deconz add-on for Hassio. Now my dimmer switch shows in Deconz and light also.

Paul-dH, 2018-10-23:
@reprz And you followed the onscreen directions (10 seccond hold of the reset button on the back)? For me this didn't work. Is there any other way to to reset the device to factory?

reprz, 2018-10-23:
@Paul-dH Yes, I managed to pair the dimmer with default 10 second method. It just started to work after I re-installed Deconz. I think the 10 second method resets the dimmer and other methods just either reconnect or do something else. I have tried pushing all the 4 buttons for 10 seconds (should also reset) and push the reset button only once (starts to seek). What does your dimmer switch do after you push the reset 10 seconds? Will it start to blink orange couple of times and then Green led turns on?

Paul-dH, 2018-10-23:
After 10 seconds the green led lids op, after this it is red/orange for about 3 seconds and then off. I will try it again in a half hour or so. Still busy at work :)

Paul-dH, 2018-11-02:
Hi, I've updated to the latest version but still unable to connect the dimmers to Deconz.

Any help is appreciated :)

manup, 2018-11-02:
Do you have a usb extension cable? It might help, there were several cases there signal was suboptimal when ConBee was directly connected to the PC or Raspberry Pi and pairing of devices was very difficult.

Paul-dH, 2018-11-02:
The ConBee is connected with a 40cm extension cable, I had WiFi interference earlier and found this fix also. But thanks for the suggestion.

closed 2019-03-09 by stale[bot]

bturrie, 2019-06-30:
I am having this issue on June 29,2019. phoscon cannot find/pair my hue dimmer with the 10 second method. I'm running firmware 26420700

manup, 2019-06-30:
This is the initial shipping firmware which has some issues. Please update the firmware to a recent version 0x26490700 and deCONZ to version 2.05.66.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

bturrie, 2019-07-05:
As  it turns out, I'm running deconz in docker using the marthoc download version. When I look in the docker logs I see two lines that seem to indicate an issue. The first line says GW sd-card image does not exist: /root/.local/share/dresden-elektronik/deconz/gw-version. Not sure if that's in docker or the linux root. Also I don't see anything like the lines that provide the file name I need to follow the proceedure for marthoc. . It also says GW firmware not supported on headless x86 linux server That sounds like I need to build a new system or just give up and buy a raspberry pi.