dresden-elektronik/deconz-rest-plugin #890: Aqara Wall Switch not reporting App/Rest-API

juzta, 2018-10-21:
Hello all,

I got the Aqara Wall Switch Wireless 2 buttons connected to Deconz but it is not showing up in the Rest-API or WebApp.

Windows 10
Version 2.05.20 / 4/8/2018
Firmware 261F0500

Am I missing something here? 

Thereby, I see that there are new versions of deCONZ and FW updates, but these are only for ARM. Is there some sort of time frame when this is being released for x64?

[image]
[image]
[image]

Thanks!

juzta, 2018-10-21:
After clicking a few times on the button I noticed that it has updated the cluster info tab with some additional info.

[image]

ebaauw, 2018-10-21:
This seems to be a newer model of the `lumi.sensor_86sw`, and needs to be whitelisted explicitly.

I doubt whether the endpoints have been read in full; you're running an older version of deCONZ (and of the ConBee firmware).  Did more clusters appear after button presses (left, right, both)?

Could you run deCONZ with `--dbg-info=1 --dbg-aps=2` flags and check what commands teh switch sends when pressing/releasing/holding the buttons?  If it's like the `lumi.sensor_86sw`, it would send a _Report Attributes_ commands for the _OnOff_ cluster on press, from endpoint 0x01 for the left button, 0x02 for the right button and 0x03 for both buttons.

juzta, 2018-10-21:
Stupid question, how can I run Windows version of deCONZ with --dbg-info=1 --dbg-aps=2? Adding them to the target of the shortcut?

"C:\Users\Server\AppData\Local\deCONZ\bin\deCONZ.exe --dbg-info=1 --dbg-aps=2" seems to change nothing?

ebaauw, 2018-10-21:
> Stupid question, how can I run Windows version of deCONZ with --dbg-info=1 --dbg-aps=2?
> And where ccan I check what commands have been sent?

Sorry, I have no clue about Windows.

> I tried to connect a new Xiaomi Aqara wall switch too.

This is a main-powered wired in-wall switch (by the looks of it a `lumi.ctrl_ln`?), not a battery-powered wireless wall switch.  It's exposed as a light, to control the wired light(s).  If it is indeed a `lumi.ctrl_ln` switch, there should also be three `/sensors` resources, a ZHASwitch, a ZHAConsumption, and a ZHAPower.

> Can me please someone explane how to to install the deconz-rest-plugin-master

The REST API plugin is distributed with deCONZ.  This repository is needed only, if you want to make changes to the plugin and/or compile it yourself.  It is only supported on Raspbian, see https://github.com/dresden-elektronik/deconz-rest-plugin#supported-platforms.

The latest Windows beta of deCONZ (incl. the REST API plugin), v2.05.40, can be downloaded from https://www.dresden-elektronik.de/deconz/win/

juzta, 2018-10-21:
Thanks, atleast I have a newer version of the deCONZ SW and REST API. Was not able to find this URL somewhere else.

Sadly it didn't change that it is still not showing up in the REST API or PWA.

juzta, 2018-10-21:
Actually the end points have been changed for the device

[image]

ebaauw, 2018-10-21:
> Sadly it didn't change that it is still not showing up in the REST API or PWA.

That was to be expected; the switch will have to be whitelisted, before it will show.

> Actually the end points have been changed for the device

That's more like it, but still no _OnOff_ clusters.  And no _Multistate Input_ on endpoint 0x03.  Could you check in the deCONZ GUI if the _Present Value_ attribute 0x0055 of the _Multistate Input_ cluster changes on button presses?

juzta, 2018-10-21:
I connected it to the Xiaomi GW I still had to see if there was maybe some FW updates for the switch. Removed the GW again from socket and tried reconnecting it to deCONZ. Now I'm having troubles that deCONZ is not able to pair with it anymore? Tried adding sensor, switch, open network.. standing 10cm away from the stick.

Removed the the node ofcourse from deCONZ but everytime I restart deCONZ the note is there again, but without any connection or lines to one of the coordinators or routers.

Why is the device not being totally removed and can I add it again?

ebaauw, 2018-10-21:
Did it update the firmware?  You need to factory reset the switch, I guess by holding one or both buttons for 10 seconds, until the LED blinks reds.  Make sure to open the network (search) from Phoscon, not from the GUI.  Try reading the _Basic_ cluster attributes while pressing/releasing a button every second or so, to make sure the switch is awake.

Rolf56, 2018-10-21:
Thank you. The latest Windows beta of deCONZ (incl. the REST API plugin), v2.05.40 shows the device as lumi.ctrl_In2.aq1 as two lamps now. The relays can be switch from the WebApp by clicking on the bulb. Scenes do not work and there are no /sensors resources shown.

ebaauw, 2018-10-21:
Phoscon probably doesn't yet support the `/sensors` resources.  You should be able to see them using the REST API.  Check the mac address in the `uniqueid` to match the resources.

If they don't show, search for devices from Phoscon, then read the _Basic_ cluster attributes in the deCONZ GUI.  That should trigger the creation of the `/sensors` resources.

juzta, 2018-10-21:
Thanks for your clear ebaauw. 

To be sure, Rolf56 is talking about some other problem with an other type. I'm using the Wireless version.

I seem to not be able to remove this node from deCONZ. I tried deleting is 100 times, rebooting, etc.. its unresponsive to any action.

[image]

any other methods to clear an entry like this? Because I'm also not able to reconnect the switch to deCONZ now. Tried factory resetting multiple times.

deCONZ is now on
Version 2.05.40 / 10/8/2018
Firmware 26240500

juzta, 2018-10-21:
Oke I hard deleted the record 0x8D85 from the zll.db with DB Browser in table devices. It is gone from the GUI now after restarting but still can't seem to reconnect the switch to deCONZ with this latest SW+FW version

juzta, 2018-10-21:
Thanks, just a new version I see. Now on 
Version 2.05.42 / 10/21/2018
Firmware 26280500

I purged it myself from the DB with DBBrowser. It never appeared in the REST-API so I'm not sure if I can even delete it from there?

juzta, 2018-10-22:
I'm still unable to connect the device to deCONZ. Removed everything from deCONZ to start with fresh setup. 10sec hardreset, LEDs start blinking like they should but it is not pairing. Replaced batteries twice to see if that helps.. nothing..

What can I do differently?

manup, 2018-10-22:
I reckon you use ConBee? If the device doesn't even show up in deCONZ you may try to use a USB extension cable so that the ConBee is not directly connected to the PC, in some cases this caused issues due some interference.

juzta, 2018-10-25:
> That was to be expected; the switch will have to be whitelisted, before it will show.
>
>
> That's more like it, but still no _OnOff_ clusters. And no _Multistate Input_ on endpoint 0x03. Could you [...]

So after a super tiresome proces of trying to reset this switch I finally succeded with the following method.

Holding one of the keys until all 3 LEDs starting to burn, at that exact moment pull out of the battery. Of this I was able to pair it again. Maybe luck.. maybe a condensator that finally decharged.

So within the 2.05.42 version I have the following end points. As you can see also on the 03.
[image]

So.. checking this 0x0055 value in the Multistate input cluster.

Left button gives a change to this on value 0x01 from 0 to 1 or the other way around. But not like that you click and it goes to 1 and another click it is 0 again.. 

Right is 0x02.. same behaviour

0x03 changes when pressing left and right at the same time.. from 1 to 2 or other way around.

Not sure where to go from here... :-)

ebaauw, 2018-10-25:
Ah, it’s a stateful switch.  Probably, when used with the Xiaomi gateway, first press/release is to turn on associated lights, second is to turn them off again.  We could create different buttonevents for these (effectively exposing 3 _On_ and 3 _Off_ buttons; buttonevent values 1002, ..., 6002), but it probably makes more sense to map both _On_ and _Off_ to the same buttonevent (resulting in 3 _Toggle_ buttons; buttonevent values 1002, ..., 3002).  That the endpoint corrsponds to the buttons (1 = left, 2 = right, 3 = both) is quite common for the Xioami switches.

juzta, 2018-10-25:
3 Toggle buttons makes most sense yes. The rest of the "intelligence" can be done by Home-Assistant imo. Making the Xiaomi switches have the same logic makes also most sense.

So how to proceed? I also just fetched some Wireshark information but not sure if this adds any value. There is clear communication for that 0xAF21 source when clicking on the buttons.

ebaauw, 2018-10-26:
Yes, the switch sends _Report Attributes_ commands for the _Multistate Input_ clusters.  The GUI already uses these to display the attribute value.  We’ll simply have to whitelist the model, create a buttonMap, and link the model to it.  I’ll see if I can cook up a PR.

juzta, 2018-10-26:
Thanks! You are a winner!

veeceeoh, 2018-11-19:
Sorry to raise this thread from the dead, but I believe the new 2018 revision of the 2-button Aqara Wall Switch (WXKG02LM) with ZigBee product ID `lumi.remote.b286acn01` may actually support single-press, double-press, and hold on each of the two buttons.

I don't yet own one of the new WXKG02LMs, and haven't received any reports from SmartThings or Hubitat users with one, but in my searching I see that [the Homey driver for the WXKG02LM](https://github.com/TedTolboom/com.xiaomi-mi-zigbee/blob/master/drivers/remote.b286acn01/device.js) supports single-/double-press and hold. 

Also [this announcement article on Xiaomi Today](https://www.xiaomitoday.com/aqara-wireless-switch-enchanced-version/) for the new 2018 revision Aqara Wall Switch models seems to confirm single-press, double-press, and hold support by saying "the dual-key version has seven different functions".

If I understand the Homey driver code correctly, the data needed to identify single-press, double-press, and hold is also included in Multistate Input attribute reports.

manup, 2018-11-24:
Thanks for the information, please continue discussion in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/811

Closing this issue so we have just one to track progress on the matter.

closed 2018-11-24 by manup