dresden-elektronik/deconz-rest-plugin #891: GE Zigbee Smart Dimmer Switch (45857GE)

wayneflo, 2018-10-21:
GE has an excellent smart dimmer switch called the GE Smart Dimmer (45857GE). This is a powered light switch that replaces a standard wall switch. I would appreciate you adding it to your supported device list. Here is the basic info. Let me know if you would like any other screen shots.

[image]

[image]

[image]

[image]

ebaauw, 2018-10-21:
Please provide a screen shot of the _Node info_ panel, see 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

It should already be exposed as a dimmable light, to control the wired light(s).  To expose the _Simple metering_ cluster as ZHAConsumption, we need to whitelist the dimmer.  Does it report power (_Instantaneous demand_) in 0.1W and the lifetime consumption _Current summation delivered_ on 0.1 Wh, as the _Divisor_ would suggest?

I expect endpoint 0x02 is used to other smart lights (that are not wired to the dimmer).  The _OnOff_ and/or _Level_ client clusters probably need to be bound to a group for that to work.  If you can find out what commands are sent, we could expose that endpoint as a ZHASwitch.

ebaauw, 2018-10-21:
The plug should also already be exposed as a `/lights` resource.  For the rest, it looks very similar to the dimmer - we can probably share most of the code.

closed 2019-04-24 by stale[bot]