dresden-elektronik/deconz-rest-plugin #894: Scene transition firmware issues in IKEA trådfri lights

lbschenkel, 2018-10-22:
I have done some experimentation on my IKEA lights. This is my setup:

- deCONZ 2.05.42, firmware 0x26280500
- TRÅDFRI bulb GU10 WS 400lm (003.182.92), firmware 1.2.217
- TRÅDFRI bulb E14 opal 400lm (703.182.84), firmware 1.2.217
- TRÅDFRI bulb E27 CWS opal 600lm (703.389.51), firmware 1.3.002

As far as I'm aware, this is the latest firmware available for each light.

I have noticed the following behaviour: if you create a scene with a transition, and later recall that scene, during the period of that transition the light will fail to respond to new commands (from the REST perspective, not sure what is happening at the Zigbee layer), and the state in deCONZ will be out of sync with the state of the light.

For example, let's say I recall a scene with a 10 minute transition. During this period the light will appear "hung" and the following commands will **not** work on the light (but will affect the deCONZ state):
- setting `on` to either `true` or `false`
- setting `bri` to any value, with or without `transitiontime` (not even 0)
- setting `bri_inc` with or without `transitiontime` (but see caveat below)

But to my surprise, I noticed that the following **do** work and "unfreeze" the light, making it able to accept commands again:

- recalling a different scene
- setting `ct` (but for IKEA lights deCONZ sends `xy` to work around a different bug, so not sure what `ct` would actually do)
- setting `bri_inc` to 0 (I believe that this gets translated to a "stop" command)

My testing was with direct commands, but I did a quick check and it looks like group commands have the exact same effect on the light as the direct ones. 

I got aware of this when I tried to implement a wake-up light by transitioning between two scenes with a 30 minute transition time, and then the lamp "hangs" and I tried different things until I figured out that some commands actually managed to make it respond again. 

Curiously, the lights don't freeze in the same way when not recalling scenes. If I send `bri=255 transitiontime=6000` (or `bri_inc`) the light does respond to subsequent commands during the transition time. I have seen some reported issues that seem related (but don't explicitly mention scenes): https://github.com/dresden-elektronik/deconz-rest-plugin/issues/450, https://github.com/dresden-elektronik/deconz-rest-plugin/issues/384, https://github.com/dresden-elektronik/deconz-rest-plugin/issues/66 

I wonder if with the knowledge above, this firmware issue could be worked around on the deCONZ side? For example, if the last command sent to the light was a scene recall any new command could be preceded by a "stop" command to make sure the light will respond. Just brainstorming here. Depending on what is deemed an acceptable solution, I might be able to submit a PR with the implementation (if not too complex, I'm a beginner in this codebase with limited Zigbee knowledge).

lbschenkel, 2019-02-19:
Hi @manup, any comments on the above?

manup, 2019-02-19:
Sorry for the late response, great findings about the transition time.

> I wonder if with the knowledge above, this firmware issue could be worked around on the deCONZ side? For example, if the last command sent to the light was a scene recall any new command could be [...]

In theory yes but this would open the window for new issues which are hard to handle reliably. There are already networks with > 50 IKEA lights out there, I'm afraid this approach doesn't scale well.

I think this issue with your discoveries should be forwarded to IKEA developers so they can recreate and hopefully fix the transition time bugs (shouldn't be that hard).

Part of the IKEA Tradfri team is active on Reddit, might be a good way to get the message through:

https://www.reddit.com/r/tradfri/

lbschenkel, 2019-02-24:
As suggested, reported in https://www.reddit.com/r/tradfri/comments/au903n/firmware_bugs_in_ikea_bulbs/

manup, 2019-02-28:
Nice very good writeup hope it will reach the devs.

There seems to be new firmware available now:
https://www.reddit.com/r/tradfri/comments/avqc73/new_firmware_out_this_day_ota_is_coming/

The changelogs don't mention anything related to the problems mentioned here, but might be worth a try.

[image]

ebaauw, 2019-03-01:
Looks like new firmware for the colour and for the white spectrum bulbs.  Upgrading...

lbschenkel, 2019-03-01:
It looks like the new official firmware is simply the test firmware which got promoted to official (however, the binaries are not exactly the same even though the versions are). The only exception that I can see is that for `0x11C1` the release version (`12214572`) is still behind the test one (`12222572`).

I had previously upgraded all my devices to the newest test firmware, and as far as I can see there has been no perceptible difference in behaviour regarding those bugs. Actually I didn't notice any change of behaviour at all.

closed 2019-07-06 by stale[bot]

pvizeli, 2019-11-01:
Can we reopen this issue?

Phontana, 2019-11-01:
Yes, we should reopen the issue since the scene transition issues are still present.