dresden-elektronik/deconz-rest-plugin #895: Issue setting bri and color (ct/xy) simultaneously in IKEA trådfri lights

lbschenkel, 2018-10-22:
Please check https://github.com/dresden-elektronik/deconz-rest-plugin/issues/894 for reference about the setup. I'm reporting here what it looks like a different firmware bug.

In IKEA lights, if you set `bri` and color (`ct` or `xy`) simultaneously in the same REST call, the light changes to the requested configuration for a fraction of a second but then goes back to the previous `bri` and only the color "sticks". Setting it separately works. Strangely, if you set both and also specify `transitiontime` of `0` or `1`, then it works.

My understanding is that setting `bri` and `xy` is not possible within the same Zigbee message, so deCONZ must be sending two when they're specified simultaneously. The fact that a transition of 0 works makes me speculate that the light has a firmware bug in which when it gets two messages in quick succession, the second overwrites the state of the first when the light is still transitioning from the first message. Setting a shorter transition than the light default (which seems to be a bit over 100ms) works because the light finishes changing state before the next message and the race condition is avoided.    

As with the other issue, I'm reporting this to check if it's possible to implement some sort of workaround in the deCONZ side to address this firmware bug (at least partially) and hopefully help getting one implemented.

lbschenkel, 2019-02-24:
As suggested in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/894, posted on Reddit: https://www.reddit.com/r/tradfri/comments/au903n/firmware_bugs_in_ikea_bulbs/

manup, 2019-06-24:
To my knowledge the bug still isn't fixed. Hopefully the new device revisions which will be released in August (?) have improved firmware.

closed 2019-10-29 by stale[bot]