dresden-elektronik/deconz-rest-plugin #901: Orvibo Smoke Sensor

alonalon, 2018-10-23:
Hi,
Would appreciate if someone could look into add support for [Orvibo Smoke Sensor.](https://www.orvibo.com/en/product/smoke_sensor.html)

I run everything on headless Linux server, so can't provide any cluster info unfortunately. 
But I think it's very much like [Trust smoke detector](https://www.trust.com/en/product/71197-zigbee-smoke-detector-zsdr-850) & [Heiman HS1SA Smoke detector](http://www.heimantech.com/product/96.html)

Maybe all those three devices can share share most of the code?

jonatanhost, 2018-10-24:
I have the same device connected but not recognized. Here is a screenshot of node info.

[image]

jonatanhost, 2018-12-19:
@manup Do you guys need more information from us of this smoke sensor to implement it?

manup, 2018-12-19:
>@manup Do you guys need more information from us of this smoke sensor to implement it?

Some more screenshots as described here would be useful, especially `model identifier` from basic cluster is needed.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

alonalon, 2019-01-14:
@jonatanhost is it possible for you to get the info manup asks for?

jonatanhost, 2019-01-21:
> > @manup Do you guys need more information from us of this smoke sensor to implement it?
> 
> Some more screenshots as described here would be useful, especially `model identifier` from basic cluster is needed.
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

I don't know how to get those attributes to show

alonalon, 2019-02-06:
[image]

think i got it, does this help @manup ?

jonatanhost, 2019-03-27:
@manup 

Do you guys need any more information about the device or is this enough?

jonatanhost, 2019-05-02:
@manup Hi, do you have any status on this? Have two sensor myself of this I would like to use as not just dumb smoke sensors :)

manup, 2019-05-02:
Hi, sorry for the late reply. Yes one more thing, please provide a screenshot of the IAS Zone cluster after reading the attributes.

The modelid looks a bit random so it would be helpful if we can pin point the sensor via its IAS cluster attributes.

alonalon, 2019-05-02:
[image]

[image]

Got two different result for same device, the values changed during pairing process in deconz, don't know which is right

alonalon, 2019-06-26:
@manup any news about this? :)

alonalon, 2019-08-27:
#1763 adds support for this, closing

closed 2019-08-27 by alonalon

arnonel, 2019-09-09:
Could we add SF21 smoke sensor please

mn26826, 2019-09-17:
I opened a new issue for this device in #1875 .