dresden-elektronik/deconz-rest-plugin #904: Hue bulbs not being discovered anymore

roamerP, 2018-10-25:
I have a couple of Phillips Hue bulbs and all have been working fine with the Raspberry Pi and Home Assistant with Conbee. But yesterday evening everything stopped working, and today I have trying to solve the problem with no luck. deCONZ webApp can't discover the Phillips bulbs anymore and can't reset the bulbs with webApp that supports Touchlink.

I have Deconz version 2.05.42.

manup, 2018-10-26:
Might be related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/887

roamerP, 2018-10-29:
Hi thanks for the tip. I have updated the firware referred in the thread above and I have installed the deCONZ software on my windows 10 laptop, when I put the conbee stick in a usb port the software connects to the conbee but it does not connect to network. I have read the conbee manual and it says that deCONZ automatically connects to the ConBee USB Dongle, but in my case no, I have red text saying Not in Network. 

[image]
[image]

When the conbee is connected to my rasperry the Phoscon app looks like this and I can't connect any lamps

[image]
[image]

Any ideas on what is going on

manup, 2018-10-29:
Configuration looks ok, does it work when you press Join?

roamerP, 2018-10-29:
No nothing happens. What to do?

Regards Ronny

manup, 2018-10-29:
That's odd it should show `Connecting` and then `In Network`.

You may try to install deCONZ 2.05.44 and upgrade firmware.
If that doesn't work can you please send me the zll.db file to check if there are any issues in the configuration?

roamerP, 2018-10-30:
Hi, I istalled deCONZ 2.05.44 and it found a firmware update but the firmware updater is stuck on 100% and it says "Firmware update is running, please wait". It has doing that for about 15 minutes now. Is it safe to close the program or what to do?

Regards Ronny

manup, 2018-10-30:
The .dat file also contains the zll.db file, if you can send it to mpi@dresden-elektronik.de I can have a look.

manup, 2018-10-31:
I've checked the zll.db file it looks very empty, beside the ConBee device there is zero data stored about other nodes like lights or sensors. I guess that's  because your network database file is on the Raspberry Pi, so I assume the backup is from the Win10 machine?

closed 2019-03-08 by stale[bot]