dresden-elektronik/deconz-rest-plugin #905: Aqara water sensor suddenly reports water for no reason

mdescher, 2018-10-26:
I added an Aqara water sensor some time ago. Not visible in the Phoscon app but in the API. It reported **"water" : false** as expected for a few days. Suddenly it switched to **"water" : true** although it never got wet. I played around with it and put it into water, dried it, etc. but nothing changed. I eventually removed it from deCONZ, added it again and somehow managed to made it work again. It reported **"water" : false** when dry and **"water" : true** when wet. Then I put it in its place again and after a few days it switched to **"water" : true** again and still keeps reporting this although it never got wet. Does this sound like a problem with the sensor? Does anybody else see such a behaviour?

ebaauw, 2018-10-26:
Did you check the sensor's battery?  I haven't experienced this with my leak sensor, but my IKEA Trådfri motion sensor started reporting motion falsely, when the batteries began to run dry.

mdescher, 2018-10-26:
Good point, I will try a new one later when I am at home.

timbru31, 2018-11-05:
I’ve had some false alarms, too. I’ve checked the battery, too, but it was full. 
So far (2 weeks?) I had no false alarms.

martikainen87, 2018-12-17:
I also have a sensor that's showing **water: true** constantly, have tried to replace the battery but no change.
Pushing the sensor updates the rest api but it still think it's wet.

jantje88, 2018-12-18:
Got the same problem with the leak sensor.

maffee, 2019-01-16:
Same problem here. Could this have anything to do with it?

https://github.com/Koenkk/zigbee2mqtt/issues/448

maffee, 2019-01-16:
Great, thanks!

ebaauw, 2019-01-16:
I installed one of these last April - I haven't seen any false positives during these 9 months.  I do receive the special reports.

Not happy with this change - I check `state.lastupdated` (manually) to verify that the sensor is still connected.

Can we please double-check if this is related to a certain firmware version of the sensor.  Or at least continue to update `state.lastupdated` on receiving the report?

manup, 2019-01-16:
>Not happy with this change - I check state.lastupdated (manually) to verify that the sensor is still connected.

Sorry I thought the IAS reports are also sent periodically? I haven't checked this though.

>Can we please double-check if this is related to a certain firmware version of the sensor. Or at least continue to update state.lastupdated on receiving the report?

Sure `state.lastupdated` can be refreshed safely.

My sensor didn't have false reports either, it has version 20170721.

wvuyk, 2019-01-16:
No false positives here either, also version 20170721

ebaauw, 2019-01-16:
> Sorry I thought the IAS reports are also sent periodically?

Nope, only Xiaomi special reports in the log.

Mine reports `swversion` 3000-0001 - I probably paired it before `swversion` would show the date.  Will fetch the sensor and squeeze it to read the attributes this weekend.

manup, 2019-01-16:
>Mine reports swversion 3000-0001

Normally even when the former version 3000-XXXX is known the datecode should be queried after the special report.

Does the sensor still work with a recent timestamp?

ebaauw, 2019-01-16:
```
{
  "config": {
    "battery": 98,
    "on": true,
    "reachable": true,
    "temperature": 1300
  },
  "ep": 1,
  "etag": "a51ac85379e2362ec9ecf26eecf6e7a7",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_wleak.aq1",
  "name": "Boiler Leak",
  "state": {
    "lastupdated": "2019-01-16T21:43:22",
    "lowbattery": false,
    "tampered": false,
    "water": false
  },
  "swversion": "3000-0001",
  "type": "ZHAWater",
  "uniqueid": "00:15:8d:00:02:33:f2:90-01-0500"
}
```

dxxxm, 2019-01-17:
I can confirm "false" water alarm. Happened in the past 12 months a couple of times. 
current deconz   2.05.55
e.g. now:
```
{
  "config": {
    "battery": 95,
    "on": true,
    "reachable": true,
    "temperature": 2100
  },
  "ep": 1,
  "etag": "a02b2bfc34ef5bfe50d41e362dcc9f48",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_wleak.aq1",
  "name": "Water Sensor",
  "state": {
    "lastupdated": "2019-01-16T11:24:55",
    "lowbattery": false,
    "tampered": false,
    "water": true
  },
  "swversion": "20170721",
  "type": "ZHAWater",
  "uniqueid": "00:15:8d:00:02:11:b4:59-01-0500"
```

ebaauw, 2019-01-17:
After manually reading the _Basic_ cluster attributes, while squeezing the sensor, `swversion` has changed to 20170721.  I even managed to read the power descriptor.  While messing with the sensor, tested it as well.  It reports leak/no more leak just  as it's supposed to.

[image]

mdescher, 2019-01-17:
I did not try the new versions yet, but the constant false alarms (they always occurred a week or two after a reset) made me remove the sensor from my network since those false alarms rendered it useless. Will try with the latest version soon.

maffee, 2019-01-17:
Is there a way to update it without the Xiaomi Gateway? I assume that is the intended way.

dxxxm, 2019-01-17:
To winnow down false alarms I have put to ea Xiaomi a Heiman water leak sensor. The later less sleek but never reported a false alarm.

ebaauw, 2019-01-19:
@dxxxm @VVW44 @mdescher @maffee @martikainen87 could you please check the _Model Identifier_, _Date Code_, and _SW Build ID_ of your sensor (see my screenshot above)?  You'll have to briefly squeeze the sensor every other second while reading the attributes from the deCONZ GUI.  It might take several attempts.  If you run headless, please check the `/sensors` resource, that should report the _Model Identifier_ under `modelid` and the _Date Code_ (or _SW Build ID_) under `swversion`.

martikainen87, 2019-01-19:
Sorry @ebaauw, I removed my sensor from deconz and haven't been able to add it back again =/ Had some issues before with the same sensor and now I've mostly given up :P Cant even reset it by holding the button for "unlimited" time.

ebaauw, 2019-01-19:
It should blink when holding the button for 5 to 10 seconds, see #263.  Did you double-check the battery?

manup, 2019-01-19:
Currently testing some general IAS Zone improvements, turns out also the Xiaomi water sensor can be configured just fine. By writing the IAZ CIE address and sending the enroll response.

Reading the attributes back it looks properly.

[image]

[image]

I'll keep the sniffer running, and hope for some periodically IAS reports.

ebaauw, 2019-01-20:
Thanks, guys.  Unfortunately, it doesn't look like versions reporting false positives vs versions that don't.

manup, 2019-01-20:
To the ones with false positives, please check if the issue remains with latest version:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_57

dxxxm, 2019-01-21:
Well, neither a spontaneous self-healing nor worsening ...
Got water sensors on two different locations. Ea site had one "aq1" reporting a false "wet" state. (same swversion, modelid)
Both sites upgraded to 2.05.57 this morning.
After the upgrade situation unchanged, both "aq1" kling on to {"water" : true}.
The local one changed its mind after a deliberate "wetting/drying"-cycle. Done the "brain wash" multiple times in the past. Doesn't last long. (> many days)
As of writing the one on the remote place is still on { "water": true}.
Please s. below. All of the three water sensors located in one spot.  (Scoring)
In fact I suspect flakey hardware. Had other lumi.aq1 ones dying slowly after initially working ok-ish. e.g. becoming increasingly unresponsive, blue led constantly on etc
And it's not the battery. Multiple times replaced to boil down on issue. Not the reason.
```
"16": {
    "config": {
      "battery": 100,
      "on": true,
      "reachable": true,
      "temperature": 1800
    },
    "ep": 1,
    "etag": "77748a46f344d8f272b9776e46ff444a",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_wleak.aq1",
    "name": "lumi.sensor_wleak.aq1_16",
    "state": {
      "lastupdated": "2019-01-21T13:57:28",
      "lowbattery": false,
      "tampered": false,
      "water": false
    },
    "swversion": "20170721",
    "type": "ZHAWater",
    "uniqueid": "00:15:8d:00:02:33:4f:83-01-0500"
  },
  "17": {
    "config": {
      "battery": 100,
      "on": true,
      "reachable": true,
      "temperature": 1900
    },
    "ep": 1,
    "etag": "a85ab6eb72a81114dbfd1502c4c8ad57",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_wleak.aq1",
    "name": "lumi.sensor_wleak.aq1_17",
    "state": {
      "lastupdated": "2019-01-21T14:19:23",
      "lowbattery": false,
      "tampered": false,
      "water": true
    },
    "swversion": "20170721",
    "type": "ZHAWater",
    "uniqueid": "00:15:8d:00:02:13:8f:22-01-0500"
  },
{
  "config": {
    "battery": 14,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "dda709f57bebc95777c9337a0df57760",
  "manufacturername": "Heiman",
  "modelid": "WATER_TPV14",
  "name": "WATER_TPV14",
  "state": {
    "lastupdated": "2019-01-17T23:12:01",
    "lowbattery": true,
    "tampered": false,
    "water": false
  },
  "swversion": "20160311",
  "type": "ZHAWater",
  "uniqueid": "00:50:43:c9:53:29:12:7f-01-0500"
}
```

Ltek, 2019-02-01:
Guys, any verdict on these devices... still having false triggers? Worth getting or too much trouble?

mdescher, 2019-02-01:
My sensor remains in "dry" state since the 2.5.57 update. For me the issue seems to be fixed.

maffee, 2019-02-01:
Mine was "wet" at the time of the update and has stayed wet. I'm unable to 'reset' the state from HA so I will get the sensor, put it in water, dry it and see if that helps. It helped the last time. It's just that the sensor is in such a difficult place that I haven't had the energy to get out... :)

Will report back.

mdescher, 2019-02-01:
That's what I did of course, mine did not reset to dry automatically by updating but I didn't expect that either.

manup, 2019-02-01:
>That's what I did of course, mine did not reset to dry automatically by updating but I didn't expect that either.

Likely this is needed since the sensor only sends proper *changed value* reports. The hourly reports which also contain a water value are not used anymore to update the value.

closed 2019-06-08 by stale[bot]