dresden-elektronik/deconz-rest-plugin #908: Ubuntu (x86_64) build of 2.05.43 missing

jurriaan, 2018-10-27:
https://www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.43-qt5.deb is not there.

jurriaan, 2018-10-27:
@manup http://www.dresden-elektronik.de/rpi/deconz-firmware/ is also missing the latest firmware files

manup, 2018-10-27:
Hi, Ubuntu version will be up tomorrow with some fixes for current config limbo.

>@manup http://www.dresden-elektronik.de/rpi/deconz-firmware/ is also missing the latest firmware files

Nope, all there but alpha numeric ordering :)
Please check the *Last modified* sort indicator.

MattL0, 2018-10-28:
Firmware : http://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x262d0500.bin.GCF

jurriaan, 2018-10-28:
@manup ah, oops :) ordering was the problem here 😅. Thanks!

closed 2018-10-30 by jurriaan