dresden-elektronik/deconz-rest-plugin #911: Bug deleting/adding Aqara switch

etique57, 2018-10-29:
Hello,

I'm having many problems since a few versions (currently on 2.05.43) adding or removing my Aqara switches (86sw2), since I started to use the Switch/sensor.

I don't really know how to debug it since I don't understand how it works (I recentfly discovered this CLIP-ON-OFF mechanism) and working with the REST API only is quite cumbersome. And I'm not able to get a consistent behavior either.

So I'll try to describe what happened, in the order that it happened:

1. added a new Osram light, works fine from the Phoscon app. Just after that, the Aqara switch (86sw2) stopped working
2. tried to remove & add again the key assignments from the Phoscon app, but it didn't work
3. removed the switch from the Phoscon app (but I think I forgot to remove 1 key assignment before)
4. paired the switch using the Switch interface but it didn't work (no green "success", no light blinking)
5. checked the REST API, the switch was actually added, but using OSRAM metadata (I guess from the light I added in 1., I noticed that in the past already).
6. Button events are not properly registered in the REST API, and the switch doesn't appear in the Phoscon app
7. deleted the switched from the REST API, and added the switch again from the Phoscon app, this time it worked (green button, blinking lights). Checked in the REST API that buttons events are properly registered
8. Found some rules in the REST API that were not deleted (though I deleted them from the interface)

In a nutshell:

- I think the delete routine is buggy, especially when it comes to key assignements. Maybe it deletes everything with the sensor number, while the rules are refering the uniqueid of the sensor.
- I also think the CLIPs are not properly cleaned.

heitec11, 2018-10-29:
After I had upgrade from Ver 2.05.39 I lost the completly communication with all of my Xiaomi sensors. Also new sensors are not able to pair. It looks like a big bug.

manup, 2018-10-29:
Can you please try 2.05.44 and the included firmware 0x262d0500 it fixes some really bad issues of the previous versions.

heitec11, 2018-10-29:
I have tested 2.05.44 and 0x262d0500 which are also not working 14 hours ago.

manup, 2018-10-30:
@heitec11 sorry for the trouble, can you please provide more details of your setup, screenshot of deCONZ network settings and deCONZ node view. If you can create a backup and send it to me via mpi@dresden-elektronik.de that would be helpful as well.

flavio20002, 2018-11-01:
I have 2.05.44 with Firmware 261F0500 and I use 10 Aqara switches and sensors without any trouble. I don't think I will ever the Firmware, if everything works...

etique57, 2018-11-05:
So, I tried again with the v2.05.45 and latest firmware.

I don't know if the new version did something, but:
- the switch gets correctly added
- I can create a rule which is properly mapped to the switch uniqueid
- the ON-OFF_TOGGLE flag sensor is properly added as well
- phoscon detects the proper switch event (1002, 2002, 3002) (the sensor gets updated)

But the rule itself is never triggered.

No clue what's happening

closed 2019-03-12 by stale[bot]