dresden-elektronik/deconz-rest-plugin #915: Unrecognized Aqara Wall Switch reporting on cluster 12

wassfila, 2018-11-01:
I got a new aqara switch 0x00158D00028CAD58 that I could not pair with phoscon, then I realized that it's actually paired with deconz and when I press it reports to deconz on cluster 12. It does not even has a cluster 6 (On Off).
Actually I had a previous aqara switch 0x00158D00024B3E45 pairing with phoscon displayed there, and reporting in the webapi. That old switch is reporting on cluster 6 (Of Off) that the new one does not have.
For the new one, I could not read any attribute even pressing read and switch click many times.

Should this be handled as a new device request or is there a change required to deconz to handle the cluster 12. I'm not sure how this relates to the special handling of Xiomi devices, I started reading the of the rest 

plugin code but still a bit lost in what is specified in the general.xml, what is coded in deconz and what is xiomi custom.

[image]

[image]

[image]

extracts from logs

log with "deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1"

`15:26:08:318 ZCL attribute report 0x00158D00028CAD58 for cluster 0x0012, ep 0x01
15:26:08:398 ZDP status = 0x00 -> SUCCESS
`

log with more options
`13:03:36:254 APS-DATA.indication srcAddr: 0x00158d00028cad58, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0012, lqi: 255, rssi: -16
13:03:36:254 	asdu: 18440a5500210100
13:03:36:254 ZCL attribute report 0x00158D00028CAD58 for cluster 0x0012, ep 0x01
13:03:36:254 	payload: 5500210100
13:03:36:254 APS-DATA.indication from child 0x3254
13:03:36:254 Delay APS request diff = 182
13:03:36:254 Node data 0x00158d00028cad58 profileId: 0x0104, clusterId: 0x0012
13:03:36:291 Node 0x00158D00028CAD58 is known by 1 neighbors, last seen 0 s
13:03:36:430 Delay APS request diff = 357
13:03:36:770 Node 0x00158D0001C95CAA is known by 0 neighbors, last seen 0 s
13:03:37:036 ZCL read cluster: 0x0000, attribute: 0x0000
13:03:37:036 ZCL read cluster: 0x0000, attribute: 0x0001
13:03:37:036 ZCL read cluster: 0x0000, attribute: 0x0002
13:03:37:036 ZCL read cluster: 0x0000, attribute: 0x0003
`

[request_device_support.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2538888/request_device_support.log)

If needed the log with more options
[OpenSwitch_full.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2538894/OpenSwitch_full.log)

I would like to give it a try to add the support for this switch myself, for that any hints would be appreciated.

ebaauw, 2018-11-01:
What’s the _Model Identifier_ in the _Basic_ cluster?  Same as in #890?

wassfila, 2018-11-01:
I tried a dozen times to click on read attributes, then keep waking up the switch by pressing it, did not manage so far.
It is likely that it is the #890 because we have the same screenshots where the On Off cluster 6 is not there.
I will give it more tries for the read attributes and I will go again through the details of #890 , From the name of the issue I wasn't sure if it's a new device but now that you mentioned I see more similarities.
Note I'm using deconz-dev-2.05.44.deb

wassfila, 2018-11-01:
Additional info:
- this is a battery powered single switch, I do not have a Xiomi gateway so never updated its firmware.
[image]

- I saw in the debug that it is well reporting on the Multistate Input cluster 12, and in the screenshot the Present value updated to 1, but contrary to #890 the Number of states is 0 not 6.
[image]

So if I understood #890 right, same here, the model would have to be whitelisted and a buttonmap created.
I'll dig now in the deconz plugin to see if I can understand how to edit that.

ebaauw, 2018-11-01:
> #890 has .b286 instead of .b186 not sure if that changes everything.
> this is a battery powered single switch

Probably the difference single vs dual switch.  Typically they share the same firmware, and they would be exposed as the same, as a ZHASwitch sensor, but with fewer buttonevent values.

wassfila, 2018-11-02:
OK, great.

manup, 2018-11-02:
The smart thing groovy device handler contains some useful information

https://github.com/bspranger/Xiaomi/blob/a7ecc57eddac0d9d8a8f7a429a3808b4324b05c7/devicetypes/bspranger/xiaomi-aqara-button.src/xiaomi-aqara-button.groovy

To get the model identifier I would suggest start sensor search and reset the switch by pressing both buttons for 10 (?) seconds, it should then rejoin and provide the information in basic cluster.

wassfila, 2018-11-02:
@manup , I managed to get the model identifier already, it is : lumi.remote.b186acn01
as the 1 in .b186 stands for a single switch as visible in the picture, so long press of 10 sec has worked.
Thank you for the reference, I'll have a look at that.

carloscae, 2018-11-03:
I also got single and double switch Aqara Remotes and I'm facing the same issue. It's the new international version (this is the new version that is supposed to work with their new homekit bridge). The previous version I could pair to Phoscom with no issues.

unfortunately I have no access to the rest api as I'm running it from Hassio.

carloscae, 2018-11-03:
My devices are:
Aqara Wireless Remote Switch (Single Rocker) - WXKG03LM
Aqara Wireless Remote Switch (Double Rocker) - WXKG02LM

wassfila, 2018-11-03:
@manup , @carloscae ,
note that the reference WXKG03LM was already supported with another model id, the novelty here is the new model id lumi.remote.b186acn01. Basically Xiaomi has multiple model ids for the same reference WXKG03LM and they don't work the same.

ebaauw, 2018-11-03:
Or rather, they use the same reference for different products.  Maybe it's actually the same hardware, with different firmware versions (which is  _not_ upgradable over the air).  Anyways, we best use the _Model Identifier_ to refer to these devices.  Unfortunately, it's a bit of a wheel of fortune which one you'll get when ordering one.

zilluss, 2018-11-04:
I confirmed that the switch worked if you just copy the cases for lumi.remote.b1acn01 and replace it with lumi.remote.b186acn01. Now I can use the switch with the button event 1002.

carloscae, 2018-11-04:
Guess I’ll need a windows pc to try that out. :/

manup, 2018-11-09:
Related wiki page https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG03LM-2018

manup, 2018-11-24:
Please continue discussion in #811

Closing this issue so we have just one to track progress on the matter.

closed 2018-11-24 by manup