dresden-elektronik/deconz-rest-plugin #916: Missing attributes after paring - is this fixed?

jo-me, 2018-11-02:
There was an issue on headless systems that, e.g. sensors were not created after pairing (basic cluster?) and there was no reliable way to fix this (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/654)

So for my Innr SP120 plug, the switch part was created but not the power consumption sensors. 

Has this issue been addressed in the latest firmware / deconz version (2.05.44)?

Thank you

manup, 2018-11-02:
There were some changes in the recent versions to improve integration of sensors and switches (ongoing) the Philips hue dimmer for example should now pair more reliable.

You may check 2.05.45 and firmware 0x262e0500 out (create a backup first).

manup, 2019-01-17:
Closing this for now.

closed 2019-01-17 by manup