dresden-elektronik/deconz-rest-plugin #917: Pairing Aqara Smart Wireless Switch WXKG11LM

saxn-paule, 2018-11-02:
Hello,

your list of supported devices contains the Aqara Smart Wireless Switch WXKG11LM.

Unfortunately I'm no able to pair it in the Phoscon App.

In the switches category there is no matching manufacturer. Trying to pair it via "Sensors --> Other" it isn't recognized.

Produkt ConBee
Version 2.05.23 / 24.4.2018
Firmware 261F0500

Some advices?

manup, 2018-11-02:
This might be WXKG11LM Rev.2 which identifies itself with model identifier
`lumi.remote.b1acn01` can you confirm that?

I don't have the switch yet which makes testing a bit cumbersome, there is already code for it active but it seems not to work properly.

saxn-paule, 2018-11-02:
Where can I see this information? I have a Conbee connected to a RasPi3 and installed deconz in headless mode.

manup, 2018-11-02:
You may try a recent version (make sure to create a backup first), the switch might be visible in the REST-API, but not yet the Phoscon App switch editor.

https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages

saxn-paule, 2018-11-02:
Thank you very much.

With Version 2.05.45 / 2.11.2018 and Firmware 262E0500
the switch could be found as sensor and is available via the API as lumi.remote.b1acn01

Will there be a frontend representation in Phoscon App in the near future too?

saxn-paule, 2018-11-02:
On press the button provides the value "1002". Maybe this is helpful.

manup, 2018-11-02:
Ok that means API works correctly.

> Will there be a frontend representation in Phoscon App in the near future too?

Yes the plan is to support all Xiaomi switches as soon as possible, I hope to receive the missing ones in the next 2 weeks to complete integration.

saxn-paule, 2018-11-02:
That are great news :D

I'm waiting for my wall switches too.

I think this issue could be closed. 

Thanks for the support.

ACrazyConcept, 2018-11-03:
I also have 2 WXKG12LM i believe the model is. They do not come up in Phoscon, but they do in deCONZ and they work. Except one does not send the long press events. But maybe that is the device and not deCONZ's fault.

saxn-paule, 2018-11-03:
Cool, I didn't know about the doubleclick feature yet. Works for me (1004) . Thank you.

ACrazyConcept, 2018-11-04:
@manup 
Sorry, I was wrong. It is also the WXKG11LM (b1acn01) I have. It has single double and hold/release like listed in that link.

Elliott32224, 2018-11-19:
It's been over a couple of weeks, so I thought I would check. Anything happening with the fix for this? As for others, I have Phoscon issues with WXKG11LM (b1acn01). Thanks. Elliott

manup, 2018-11-20:
In the last days more details were collected and documented to be used as base to complete the implementation (see the linked pages in supported devices wiki) hopefully my switches arrive soon to simplify testing.

manup, 2018-12-02:
Good catch, yes I didn't have it on the radar, will be added too.

Still a bit confused about the 1000/1002
As I understand it the switch supports:

short press: 1000
short release: 1002

But the switch doesn't support  hold (1001) hold release (1003)?

esseki, 2018-12-02:
> As I understand it the switch supports:
short press: 1000
short release: 1002
But the switch doesn't support hold (1001) hold release (1003)

Exactly. That's a kind of weird behavior from this first version of the Xiaomi button. But it supports up to 4 clicks ;)

esseki, 2018-12-10:
Hi @manup,

immediately after I saw the release of 2.05.50 I updated my setup.
But I could not find the way how to use the switch editor with my Xiaomi round buttons.
Here is what I tried : 

- for the buttons that were already included in my setup before 2.05.50 : I tried the switch editor for one of my groups. The problem is that I don't see my Xiaomi round button in the "select switch" screen (when my Ikea and Hue switch appear). See the screenshot: 

[image]

- Then I tried adding a new button. When  I went to "switches / Add a new switch" I did not see the brand Xiaomi. I tried to add a button without any success. I then went to "sensors / Add a new sensor", I could add my Xiaomi button but it does not appear in the list of sensors (it did not appear either before 2.05.50). It does not appear in the list of switches either. It shows in the Rest API "/api/apikey/sensors/" but that's it. I can not find this new button in the switch editor for a group. 

What am I missing ?

manup, 2018-12-10:
The round button – WXKG01LM is not supported yet, only the switches listed in the release notes.
I've ordered two round buttons via Gearbest, hopefully they arrive soon to complete integration.

saxn-paule, 2018-12-19:
Today my second charge of WXKG11LM arrived. If I could trust the prints on package and button, it is the 2018 version.

In phoscon it's detected as the old version (no Rev2.0 behind model).

The problem is, that I only get 1002 and 1004 events. No tripple, quad or long press events at all.

Manufacturing date is 10/2018

The upper one is a Rev 2.0 from my last order, the lower one is the new one from yesterday. As you could see, the glue pad is also much thinner.

[image]

saxn-paule, 2018-12-19:
Okay, forget about it. I pressed the button too short. Its a rev 1 and tripple/quad press works.

So it seems to be luck wich version you get.

manup, 2018-12-19:
Interesting, which firmware version do they show?

>So it seems to be luck wich version you get.

Sadly yes, it's hard to tell from the shop descriptions which version it is.

saxn-paule, 2018-12-19:
> Interesting, which firmware version do they show?
How could I see this? Running deconz (2.05.50 / 8.12.2018 / 262F0500) in headless mode on a Raspi 3.

manup, 2018-12-19:
In the *Phoscon App > Menu > Switches > Switch Details*

[image]

I guess the date code as version is only queried and shown in newer deCONZ versions. On older versions usually all Xiaomi devices report 3000-0001 as version which isn't helpful.

saxn-paule, 2018-12-19:
There is no version shown on all my Aqara devices:

[image]

[image]

[image]

[image]

manup, 2018-12-19:
Jep, this is only queried from the devices in newer versions 2.05.51 and 2.05.52. It may take up to one hour, until the version shows up.

saxn-paule, 2018-12-19:
Well, then lets update to the .52 version ;)

manup, 2018-12-19:
And backup first ;)

saxn-paule, 2018-12-19:
Always :)

saxn-paule, 2018-12-19:
My first order is version "20180525". The ones from yesterday "20161128"

So I definitely got version 1. :(

Xento, 2018-12-21:
I try to pair a WXKG11LM but it doesn't work.
I'm using this versions:
2.05.52 / 16.12.2018
262F0500

saxn-paule, 2018-12-21:
You need to pair it as a sensor. Press the small button and hold it for some seconds. Then press the big button several times. Worked for me with all 8 buttons.

Mr-REM-Esq, 2019-02-16:
I try to pair a WXKG11LM as a sensor but it doesn't work. 
I'm using this versions:
Version 2.05.59 / 14.02.2019
Firmware 262F0500

manup, 2019-02-17:
How do you pair the device?

Usually:
* Start sensor search
* Press the small button on the side until the led blinks

Mr-REM-Esq, 2019-02-17:
Sure. I did it the way you described it. Start sensor search, then press the small button on the side until the led blinks. It blinked 7-10 times and went out. Connection did not happen.
Before that, I successfully connected two Aqara temperature sensors.
The same button successfully worked with an earlier version of deconz.

ademobile, 2019-02-25:
With my issues with IKEA sensors stopping working (events being forwarded #1200), I thought in the meantime to help remove the grief I am getting over this, I ordered one of these switches. It pairs, can be seen by Phoscon and GUI, however I am not getting any events from it. Also if I assign it to a group, it does not switch it. I seems to be the same issues as above. Here is what I can see:

[image]
[image]
[image]
[image]

So I am still stuck, and I am using the same versions as above

ademobile, 2019-02-27:
As I need to get this switch work, I spent all morning, removing and adding it again, as I felt that it must work to get as far as I did. After a few attempts it started to work! Don't ask what I did (the only thing different was I pressed the middle button over and over whilst waiting for it to be 'seen')

alinelena, 2019-03-12:
I do not seem to see how can one configure hold/long release events in switch editor. I see the events in the log.

Mr-REM-Esq, 2019-03-15:
Unable to pair WXKG11LM in version 2.05.60 too.

haakonstorm, 2019-03-18:
WXKG11LM here. Latest version of everything. HASSIO, RaspBee, RPI3B+.
Most of the Aqara devices I got are now _more or less_ running OK, either directly or via REST.
The devices I can't get to join are these simple buttons.
Not sure about the following, did anyone get any 100% info:
1. How to reset
2. How to set in pairing mode

rb2k, 2019-06-30:
Same issue here. I was able to Pair WXKG12LM switches just fine (as switches), but the Xaomi Mija WXKG01LM doesn't seem to work as a switch or a sensor.

closed 2019-11-04 by stale[bot]

Franco757, 2022-01-06:
Hi,

I am trying to pair an Aquara mini switch WXKG11LM with my ConBee II (ver 2.13.04 12/18/2021, Firmware 26660700) attached to a RPI3B+ running home asssitant 2021.12.8 but not successful. 

I am able to pair and use Aqara temperature/door/water/motion sensors without any issue but not the mini switch.

Any thoughts?
Thanks

Mimiix, 2022-01-06:
@Franco757 please search this git as it's a known issue.

Franco757, 2022-01-06:
I have and the suggestions do not work.

Mimiix, 2022-01-06:
It's a known issue that it broke with the latest stable. Fix will be in next beta.

insalada, 2022-01-11:
I have the same problem as some users above. All my aqara devices are perfectly recognized and working via Phoscon, except with aqara mini switch (WXKG11LM). I have tried everything, as a switch and as a sensor, all possible combinations, but no success so far. I am in the last version of Home Assistant and using conbee II. Any suggestion will be welcome. Thanks.

Franco757, 2022-01-12:
@Mimiix this case has been closed and merged with #5647.
#5647 is listed as closed with a modificaton in the CPP file.
What are the users running Phoscon via HA running on a PI + ConBee II supposed to do with that modified CPP file?
I doubt we have to recompile the software ourselves.
Thanks

Mimiix, 2022-01-12:
Wait until home assistant updates the addon. That only happen on stable releases so that can take a while unfortunately

insalada, 2022-01-12:
@Mimiix thanks. Will this fix be included in the next stable release?

Mimiix, 2022-01-12:
If it's in next beta's, it will be in next stable.

simonfxr, 2022-01-26:
Any idea when there will be a beta release including this change?

jankryh, 2022-02-09:
the problem still remains deConz version 2.14.1

bumpyflea, 2022-04-12:
Hi @Mimiix 

I'm also using deConz version 2.14.01 / 2/6/2022 on firmware 26720700, and still am unable to pair any of my WXKG11LM switches. 

Have tried pairing through both the phoscon UI + the deconz UI,  but no success. 

Is there anything else I (or we, as seems like Jankryh and others are still having the same issue) should be doing to get it working?   

I have seen this new node appearing in the Deconz UI which doesn't how any model number or manufacturer. 
[image]

Mimiix, 2022-04-12:
Looks like yours isn't paired fully as the clusters are missing.

This might help out : 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

bumpyflea, 2022-04-13:
@Mimiix  - you are a legend!!! thanks heaps for your help!. That worked beautifully.. 
Thanks again for the quick response!!