dresden-elektronik/deconz-rest-plugin #925: Window Covering - REST API State Does Not Report Tilt Percentage (=sat)?

felixstorm, 2018-11-04:
I'll direct this to @ma-ca as he was so kind to implement the window covering functionality.

I have tried several variants to get tilt percentage reported in the state of the emulated light, but never succeeded. I even tried to completely delete zll.db, add the ubisys J1 again, calibrate it by putting `{"windowcoveringtype" : 8}` to `/api/DEADBEEF/sensors/23/config`, but never managed to get `sat` reported.
Lift position is being reported as `bri` correctly, even putting `sat` does work (blinds move to intended tilt position) and in the GUI the cluster attributes for CurrentPositionLiftPercentage and CurrentPositionTiltPercentage are updated in real time while the blinds move, but `sat` is still missing when querying state.

@ma-ca - what would be the intended procedure to make a ubisys J1 report it's tilt percentage as `sat` in it's state?

Thanks,
Felix

ma-ca, 2018-11-04:
The intended procedure was to read attribute `0x0000` WindowCoveringType and then add `sat` only if Tilt has been configured.

But unfortunately, this seems not to work. Perhaps the attribute value has not been read when the ubisys J1 is being added as light node.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/3ebf00f010f591a19dca1fe123b221ca7c52d38a/light_node.cpp#L528-L565

I have added a commit to my latest PR #919 that adds `sat` during calibration if the type is Tilt.

felixstorm, 2018-11-04:
I have finally managed to get deCONZ to compile and run on WSL (Windows Subsystem for Linux) and therefore could also work on the code a bit. Coding on a Raspberry Pi is just not realistic ;-)

I have about 15 ubisys J1 running for more than a year now and have observed that there are certain situations when a J1 does not know current positions at all (lift and/or tilt) and then reports `0xff` instead of a valid percentage. This happens for example after a power outage, if LiftToTiltTransitionSteps have not been configured or if the J1 cannot detect positions at all (e.g. in case when the cover is connected via relays - I have one that runs on 24 volts dc).
Therefore I extended handleWindowCoveringClusterIndication so that it will add or remove sat and bri based on whether the J1 actually reports valid positions or not: https://github.com/felixstorm/deconz-rest-plugin/commit/0f677c876f84d9f51b185a47c9ff34809762262e
I have also removed the logic in setHaEndpoint as this is not needed anymore then.

What do you think about it, @ma-ca?

ma-ca, 2018-11-05:
> Therefore I extended handleWindowCoveringClusterIndication so that it will add or remove sat and bri based on whether the J1 actually reports valid positions or not

I am not sure, if dynamically adding and removing `sat` and `bri` is good. I am using HomeKit with homebridge-hue and don't know what happens, if `bri` is added and removed.

@ebaauw has `lumi.curtain` and is also using Window Covering.

felixstorm, 2018-11-08:
@ma-ca - you might be right about the dynamic attributes with HomeKit and other software. But I think we should still try to cover the following two points somehow:
* catch invalid attribute reports (above 100%, e.g. 0xff)
* allow to manually disable `bri` (and maybe also `sat`) independently of WindowCoveringType for covers that are unable to report it's lift percentage at all (like my example where the cover is controlled by relays)

If you have a suggestion on what implementation would be best I am happy to try to create an additional PR if you would like me to.

ebaauw, 2018-11-08:
Indeed, dynamically adding or removing attributes doesn't seem like a good idea, but I wouldn't mind if you only do it on initial setup and/or calibration (after which, you would restart homebridge).  You might want to set `state.bri` to `null` when the value is 0xff.

felixstorm, 2018-11-09:
Setting `state.bri` or `state.sat` to `null` on invalid attribute reports sounds like a good idea to me.

`pydeconz` (used in Home Assistant) will most likely returns this as Python `None` which could then be handled in the layers above to signal that there is no valid state available.

closed 2019-03-16 by stale[bot]