dresden-elektronik/deconz-rest-plugin #928: State of the nation.

olemr, 2018-11-06:
Hello.
Coming from 2.05.40 / 26240500 will there be any known issues going to 2.05.45 / 262e0500 ?
Have the Ikea light/HUE1 dropout issues been improved?

I see some have experienced the need to re-add Ikea lights. This will be a hassel for me as there are 16 bulbs in 8 facade enclosures. Does restoring a backup work if this should happen?

apeinspace, 2018-11-07:
Hi,
I cant comment to your setup, because I mainly run tradfri lights and quite a lot of Xiaomi sensors - But I wish I never made the upgrade away from .39 to .42 and .44

If your setup works - do nothing. Grab a glass of wine and take the time to read the open issues ...
Cheers

assistcyber, 2018-11-07:
As per the above - it's awful right now, hardly anything zigbee wise works in the house at the moment and I wish I could go back. Perhaps I can, but there's no easy answer to that which I can find on here.

I have door and motion sensors which drop on and off the network at will, and so I'm having to disable most automations because they're so unreliable.

It has made me question reliance on it all really...

christmasjumper, 2018-11-07:
Agree entirely with the above - hardly anything in my house connected to automation works reliably at the moment - it's a nightmare. I'm slowly trying to disable automations because they're all so unreliable with sensors dropping on and off at will....

olemr, 2018-11-07:
OK. The devil you know, right? :-)
I'm content with power-cycle my 16 facade bulbs once a week for now. Takes only a few secs.
Only snag is that when power comes back on, Trådfri goes to full brightness. Wish there was a 'remember last state' possibility.
My HUE1 has left the building permanently. Rapid pressing, nor 4x hold-down will get it  back.
The most trustworthy device on the Conbee/Deconz net is my mi/aqara cubes. Haven't missed a single gesture, ever!

apeinspace, 2018-11-17:
... just to update to the development on my side - and maybe off topic in here.

Out of frustration with the deConz versions from .39 to .44  I finally got started on a project I had been planning for some time. Testing out the [zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started) Project. I did order the required hardware (+/-20€)  a while back from aliexpress to play with it in the wintertime. They maintain a hassio container as well.

Long story short - I did end up moving almost all my equipment (Xiaomi, Osram an IKEA) to it and stuff just works. A bonus is the graphviz network map you can pull via mqtt - which gave me back insights into my Zigbee mesh network I had lost with moving to the deConz hassio container. Learned a lot about Zigbee (optional) and mqtt on the way.

Pairing Xiaomi switches can be a pain and take patience. But I got all my equipment to pair - 2 of them I wasn't able (or didn't have the patience) to pair before to deConz/ConBee.

I am happy to read that things seem better with deConz now in .45 and .47 - I will probably return in the future.

closed 2019-03-24 by stale[bot]