dresden-elektronik/deconz-rest-plugin #929: Busch Jäger Light Link switches only show up in Deconz but not in Phoscon App

xeetsh, 2018-11-06:
Hello,

i have five Busch Jäger Light Link switches with different switch and relay types (1, 2 and 4 buttons, dimmers and relays). They all show up fine in the Deconz app and are fully controllable but visible and/or controllable in the Phoscon App. The Pairing Process over the "Add new switch" button does nothing. I've already reinstalled the image and reseted the switches without success. Other devices like the Osram Lightify plug work fine.

I'm on the deCONZ 2.05.44 Raspberry Pi image with the Raspbee module firmware 0x26D0500

[image]

[image]

Please feel free to ask if you need any more information. Thank you.

heitec11, 2018-11-07:
I had a similar situation. I could fix it by reset the node under deConz. After then I could use the Phoscon App to add the device.

xeetsh, 2018-11-10:
Thank you, this kinda worked for me.

To add the switches to the Phoscon app I had to activate the „Search for switches“ function. While the app was searching I had to reset the switch in Deconz. After this procedure the Switch was visible and controllable via Phoscon.

This semms like a bug to me.

closed 2019-03-17 by stale[bot]

vandalon, 2019-04-23:
I'm seeing this issue with icasa 8s as well.

manup, 2019-04-23:
The iCasa switches are not yet supported in the Phoscon App and switch editor, but they are on the todo list.

rendgeor, 2019-09-04:
In the last release 8 days ago it was announced that they are supported.
Whatś the difference between phoscon and web app under the hood?