dresden-elektronik/deconz-rest-plugin #931: Improve Change log Documentation

GadgetUK, 2018-11-07:
Whilst changes are clearly documented under the Github Releases section

https://github.com/dresden-elektronik/deconz-rest-plugin/releases

The main Deconz Website change log remains significantly out of date.

Github releases also lag the downloadable packages from the downloads website, and finally there is no indication what version "deconz-latest.deb" actually is.

wmn79, 2019-01-07:
Wouldn't it be possible to change the changelog link in the Phoscon Gateway page (under the firmware update button) to the above github releases page? That would make more sense than linking it to the current really outdated changelog page for the webapp and deCONZ.

closed 2019-05-14 by stale[bot]

manup, 2019-05-15:
Status: A better change log is on the todo list. It will likely be based on machine readable RSS sources from which online version for the web page and Github will be created.