dresden-elektronik/deconz-rest-plugin #933: Mesching changed with  0x262f0500 and 2.05.49?

wvuyk, 2018-11-08:
After upgrading, the deCONZ view chaged a lot here. Most lights ony have one connection in this view, and some have very many coonctions. Also one sensor seems to have changed in a router? I thin this is not looking well if this is correct? No more meshing?

[image]

manup, 2018-11-08:
The meshing is fine under the hood, but sometimes it happens that neighbor tables aren't queried (used to display the relations between nodes) when the queue is busy with oder zdp requests. This bug exists for some versions now and is under investigation, as workaround when you restart deCONZ it starts with fresh queues.

>Also one sensor seems to have changed in a router?

Which one? Can you please post a screenshot of the Node Info Panel when it's selected?

wvuyk, 2018-11-08:
It is a IKEA motion sensor... Very surprising!

[image]

wvuyk, 2018-11-08:
A restart did not change the visual display... never seen this before... but indeed all lights keep working, so I will just trust it is ok for now. 

Andother observation, this IKEA motion sensor seems not to be connected to anty other router of gateway. But then, some other sensors are also not visibly connected, so maybe that means nothing....

ebaauw, 2018-11-08:
It's probably just a glitch reading the node descriptor.  I had some issues with my `lumi.ctrl_ln2` wired switch showing the wrong clusters in the fingerprint.  I deleted the node and all resources.  On re-detecting the node, the GUI wouldn't show the endpoints, then it showed it as an end-device.  Deleted it again.  Upon next discovery, it showed correctly.

MattL0, 2018-11-08:
where are 0x262f0500 and 2.05.49?

wvuyk, 2018-11-13:
The new version 2.0.47 is looking better again :-)

[image]

closed 2018-11-13 by wvuyk