dresden-elektronik/deconz-rest-plugin #935: Problems with upgrade to 2.05.46

Pigweed, 2018-11-09:
I just performed this upgrade and had some issues that I could use a hand with rectifying.
I upgraded from 2.5.26 to 2.5.46. I have a Raspberry Pi 3 B+ with the image I had downloaded from the dresden-elektronik website.

First, during install deCONZ, these steps seemed to do very little or nothing (super fast, with only a couple lines returned:
 sudo apt update
 sudo apt install -f
I skipped: 'Install deCONZ development package (optional)'.
Then, during 'Get and compile the plugin', qmake and make both failed saying file not found.
I fished around the issues and found this article: Development setup - doc ammendment #388.
I did this and things looked a lot better: 
sudo apt install libqt5websockets5-dev libqt5serialport5-dev qt5-qmake qt5-default
qmake ran error free but make -j1 through a fatal error about sqlite3 directory not found.

I did the firmware update in Phoscon and I have this:
Version 2.05.46 / 5/3/2018
Firmware 262F0500

But, features from the release notes are missing. There is no switch editor as is detailed for: V2_05_34. 
Most everything seems to be working as it was before but appearing not to have updates.

What should I do next? 
Thanks.

manup, 2018-11-09:
Looks like the update worked but the older version of the app is still in browser cache.

Refreshing the cache will be done automatically in future releases, as a workaround just append a bogus query string to the URL.

For example:

    http://<raspberryip>[:<port>]/pwa/index.html?12345

Like:

    http://192.168.2.4/pwa/index.html?12345

This will force the browser to update the page.

Pigweed, 2018-11-09:
Thanks for the response. This does not seem to be the issue. I am still not getting the new features.

manup, 2018-11-09:
It can only be browser cache related since the Phoscon App is part of the package.
Note: the new switch editor is located under Groups > Edit.

Which browser are you using and what is the URL?

As alternative you may try http://dresden-elektronik.de/pwa which also has the latest version of the app.

[image]

Pigweed, 2018-11-09:
You are right. It seems to be a problem with Chrome not reloading properly. On my Windows computer I cannot seem to get Chrome to refresh the app.
I'll sort it out here.
Thank you for the help.

Pigweed, 2018-11-09:
I had to clear the cookies manually for it. Good to go.

closed 2018-11-09 by Pigweed