dresden-elektronik/deconz-rest-plugin #940: v2.05.46 / 0x262F0500 APS request queue corruption?

ebaauw, 2018-11-11:
Running v2.05.46 with 0x262F0500 since this morning.  After a while, the log is spammed with APS request timeout messages.  It looks like the corresponding APS request isn't removed from the queue.

In the log below, the corresponding request is to a Trådfri dimmable light bulb, but I saw these messages before, for different nodes and clusters.  They only stopped after restarting deCONZ (I did _not_ reset the RaspBee).

It looks like deCONZ tries to send an APS ACK (or default response?) on receiving an attribute report from the bulb (I do have _APS Acks_ enabled in the GUI):
```
Nov 11 11:25:09 pi2 deCONZ[25767]: 11:25:09:434 APS-DATA.indication srcAddr: 0x000b57fffe3249aa, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 191, rssi: -67
Nov 11 11:25:09 pi2 deCONZ[25767]: 11:25:09:434 	asdu: 08420a00001000
Nov 11 11:25:09 pi2 deCONZ[25767]: 11:25:09:435 ZCL attribute report 0x000B57FFFE3249AA for cluster 0x0006, ep 0x01
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:435 APS-DATA.request id: 78, addrmode: 0x03, addr: 0x000b57fffe3249aa, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 11 len: 5 tx.options 0x04
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:436 Delay APS request diff = 71
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:487 APS-DATA.indication srcAddr: 0x000b57fffe3249aa, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 191, rssi: -67
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:487 	asdu: 08430a000020fe
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:488 ZCL attribute report 0x000B57FFFE3249AA for cluster 0x0008, ep 0x01
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:488 APS-DATA.request id: 82, addrmode: 0x03, addr: 0x000b57fffe3249aa, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x01 queue: 12 len: 5 tx.options 0x04
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:488 Delay APS request diff = 124
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:489 Delay APS request diff = 124
```
The ACK (response?) to the _OnOff_ cluster is confirmed:
```
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:602 APS-DATA.confirm id: 78, status: 0x00 SUCCESS
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:602 APS-DATA.confirm request id: 78 -> erase from queue
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:607 APS-DATA.confirm id: 0, status: 0x00 SUCCESS
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:607 APS-DATA.confirm id: 0, status: 0x00, match: 0
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:608 APS-DATA.confirm id: 0, status: 0x00 SUCCESS
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:608 APS-DATA.confirm id: 0, status: 0x00, match: 0
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:612 APS-DATA.confirm id: 0, status: 0x00 SUCCESS
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:612 APS-DATA.confirm id: 0, status: 0x00, match: 0
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:615 APS-DATA.confirm id: 0, status: 0x00 SUCCESS
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:615 APS-DATA.confirm id: 0, status: 0x00, match: 0
Nov 11 11:25:10 pi2 deCONZ[25767]: 11:25:09:633 aps request id: 78 finished, erase from queue
```
I'm not sure if the response for id 0 is related.  I see more of these messages throughout the log.  If I understand correctly, the confirm is generated by the RaspBee firmware after it has sent the request to the network?

The second ACK (reponse?) times out, but is not removed from the queue:
```
Nov 11 11:26:09 pi2 deCONZ[25767]: 11:26:09:009 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:26:09 pi2 deCONZ[25767]: 11:26:09:088 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:26:09 pi2 deCONZ[25767]: 11:26:09:168 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
...
Nov 11 11:32:33 pi2 deCONZ[25767]: 11:32:33:017 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:33 pi2 deCONZ[25767]: 11:32:33:097 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:33 pi2 deCONZ[25767]: 11:32:33:176 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
```

When the APS request id is recycled, the confirmation of the new request seems to be handled OK:
```
Nov 11 11:32:33 pi2 deCONZ[25767]: 11:32:33:257 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:33 pi2 deCONZ[25767]: 11:32:33:337 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:416 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:551 poll node 00:17:88:01:10:4f:10:e6-0b
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:553 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:577 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:584 APS-DATA.request id: 82, addrmode: 0x03, addr: 0x00178801104f10e6, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 2 len: 5 tx.options 0x00
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:642 APS-DATA.confirm id: 82, status: 0x00 SUCCESS
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:642 APS-DATA.confirm id: 82, no match [1]
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:642 APS-DATA.confirm request id: 82 -> erase from queue
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:656 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:657 aps request id: 82 finished, erase from queue
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:736 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:768 APS-DATA.indication srcAddr: 0x00178801104f10e6, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -51
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:768 	asdu: 1860010000001000
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:816 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:897 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 11:32:35 pi2 deCONZ[25767]: 11:32:33:976 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
```

manup, 2018-11-11:
>I'm not sure if the response for id 0 is related. I see more of these messages throughout the log. If I understand correctly, the confirm is generated by the RaspBee firmware after it has sent the request to the network?

Correct, the deCONZ core will wait until a confirm is received with latest firmware this should take up to 8 seconds when a route exists. It can take longer when route establishment is needed. There is also one case when no confirm is received — if device is power-cycled after APS request is send and before confirm is created.

deCONZ core should throw too long waiting uncomfirmed APS requests out of the queue and yield a artificial APS confirm to keep other  sub systems like REST plugin going.

But I need to check the code if it is working and maybe adjust max timeout (which could be as long as 90 minutes based on older firmware).

Also not that all APS ACKs are handled by the firmware, deCONZ only sends the command with APS ACK enabled flags.

ebaauw, 2018-11-11:
> There is also one case when no confirm is received — if device is power-cycled after APS request is send and before confirm is created.

Do you the gateway device?  As in power cycling (resetting) the RaspBee while deCONZ remains running?  That would be a neat trick.  Maybe deCONZ keeps running when removing and re-inserting the ConBee?  Or do you mean the device to which the request is sent?  Then the APS-DATA.confirm is raised only after an ACK or default response was received from the device?  I cannot find this in the deCONZ serial protocol documentation.

> deCONZ core should throw too long waiting uncomfirmed APS requests out of the queue and yield a artificial APS confirm to keep other sub systems like REST plugin going.
But I need to check the code if it is working and maybe adjust max timeout (which could be as long as 90 minutes based on older firmware).

Indeed, after well over 90 minutes:
```
Nov 11 12:59:10 pi2 deCONZ[25767]: 12:59:08:873 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 12:59:10 pi2 deCONZ[25767]: 12:59:08:952 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 12:59:10 pi2 deCONZ[25767]: 12:59:09:033 aps request id: 82 prf: 0x0104 cl: 0x0008 timeout (confirmed: 0) to 0x000B57FFFE3249AA (0xB352)
Nov 11 12:59:10 pi2 deCONZ[25767]: 12:59:09:112 aps request id: 82 failed, erase from queue
```

I now have seven of these requests:
```
Nov 11 18:51:49 pi2 deCONZ[16191]: 18:51:49:776 aps request id: 7 prf: 0x0104 cl: 0x0406 timeout (confirmed: 0) to 0x0017880102027F90 (0x81A9)
Nov 11 18:51:49 pi2 deCONZ[16191]: 18:51:49:776 aps request id: 176 prf: 0x0000 cl: 0x0031 timeout (confirmed: 0) to 0x00178801103C9D0A (0x510A)
Nov 11 18:51:49 pi2 deCONZ[16191]: 18:51:49:776 aps request id: 72 prf: 0x0104 cl: 0x0406 timeout (confirmed: 0) to 0x0017880102027F90 (0x81A9)
Nov 11 18:51:49 pi2 deCONZ[16191]: 18:51:49:776 aps request id: 63 prf: 0x0104 cl: 0x0702 timeout (confirmed: 0) to 0x00158D0002179269 (0xB59C)
Nov 11 18:51:49 pi2 deCONZ[16191]: 18:51:49:776 aps request id: 26 prf: 0x0104 cl: 0x0406 timeout (confirmed: 0) to 0x001788010200A85B (0xC5F0)
Nov 11 18:51:50 pi2 deCONZ[16191]: 18:51:49:776 aps request id: 42 prf: 0x0104 cl: 0x0000 timeout (confirmed: 0) to 0x0017880100F04B13 (0x2BEC)
Nov 11 18:51:50 pi2 deCONZ[16191]: 18:51:49:776 aps request id: 193 prf: 0x0104 cl: 0x0406 timeout (confirmed: 0) to 0x0017880102027F1A (0xC9BA)
```

manup, 2018-11-11:
>Do you the gateway device? As in power cycling (resetting) the RaspBee while deCONZ remains running? 

Yes exactly this, 

> Maybe deCONZ keeps running when removing and re-inserting the ConBee?

It does, and it will just reestablish the serial port connection.

>Or do you mean the device to which the request is sent? Then the APS-DATA.confirm is raised only after an ACK or default response was received from the device? I cannot find this in the deCONZ serial protocol documentation.

The APS confirm is generated for every APS request by the firmware, when the device (light, sensor,..) is not powered a APS confirm with non success status will be raised.

>Indeed, after well over 90 minutes:

This is the real problem here, since with the newer firmware a confirm should no longer take when 1 minute, I'll lower this timeout in core.

I need also to check that confirms are always issued as long as the APS requests are send during the firmware lifecycle, it would otherwise be a bug.

The firmware can also by rebooted by deCONZ core when it detects unusual conditions, I need to check that the queues are also cleaned up properly in this case.

closed 2018-11-21 by ebaauw