dresden-elektronik/deconz-rest-plugin #941: Philips hue motion does not want to mesh

donnib, 2018-11-11:
Hi,
I have one Philps hue motion sensor which seems it doesn't like to mesh thru IKEA lights. It only makes one line which goes to the Raspbee and never makes mesh lines thru other routers. Do you also see this ?

/donnib

donnib, 2018-11-11:
@ebaauw yes but I see no other line from it and that's what I woul expect since the signal is not good to the gateway I would expect to have better connection to the lights which is very close to it

wvuyk, 2018-11-11:
I remeber in the past there have been issues with Philips Motion sensors connecting to IKEA lights, which would prevent the sensors sending updates. Maybe it is now somehow preveted to have Philips sensors connecting?

wvuyk, 2018-11-11:
Also check #649 , maybe this gives a clue?

manup, 2018-11-11:
And also related issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/940 this should behave better in the next version.

donnib, 2018-11-12:
@manup should i understand your comment that this is a known problem ?

manup, 2018-11-12:
>I have one Philps hue motion sensor which seems it doesn't like to mesh thru IKEA lights. It only makes one line which goes to the Raspbee and never makes mesh lines thru other routers. 

It's the choice of the sensor to chose the parent node which might be the coordinator. Does this setup cause problems or is the sensor working otherwise?

donnib, 2018-11-12:
@manup well it causes problems because i don't get updates from it and i pressume it's because of the range to the coordinator. As mentioned there are two ikea lights which are 30cm from it so it should use them because it has better signal.

allofmex, 2019-03-10:
@donnib Could you solve the hue motion sensor issues? 
I'm experiencing some problems too, sounds similar.

- Sensor works stable for some time (some hours)
- then permanently looses connection
`"AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network."`
- All other 17 devices keep working (incl. hue rwl021 dimmer switch)
- I have to reset cc2531 to make it work again for some hours

Sensors straight network link to coordinator is week, but have bulbs nearby that have better link (2 of them are Ikea).

closed 2019-07-15 by stale[bot]