dresden-elektronik/deconz-rest-plugin #943: Websocket events sometimes contain config instead of state

oscar-b, 2018-11-12:
The websocket events for my Hue Motion (id 4 is type `ZHAPresence`) sometimes doesn't have the state in the message:

```json
{  
   "config": {  
      "alert": "none",
      "battery": 100,
      "delay": 122,
      "ledindication": false,
      "on": true,
      "reachable": true,
      "sensitivity": 2,
      "sensitivitymax": 2,
      "usertest": false
   },
   "e": "changed",
   "id": "4",
   "r": "sensors",
   "t": "event",
   "uniqueid": "00:17:88:01:02:13:41:66-02-0406"
}
```

Proper event:

```json
{  
   "e": "changed",
   "id": "4",
   "r": "sensors",
   "state": {  
      "lastupdated": "2018-11-12T07:33:42",
      "presence": true
   },
   "t": "event",
   "uniqueid": "00:17:88:01:02:13:41:66-02-0406"
}
```

Sensor from `/sensors`

```json
    "4": {
        "config": {
            "alert": "none",
            "battery": 100,
            "delay": 122,
            "ledindication": false,
            "on": true,
            "pending": [],
            "reachable": true,
            "sensitivity": 2,
            "sensitivitymax": 2,
            "usertest": false
        },
        "ep": 2,
        "etag": "b5c9e7242bf4e60031ea9ab7f19999f5",
        "manufacturername": "Philips",
        "modelid": "SML001",
        "name": "Motion Uppe",
        "state": {
            "lastupdated": "2018-11-11T08:26:03",
            "presence": true
        },
        "swversion": "6.1.0.18912",
        "type": "ZHAPresence",
        "uniqueid": "00:17:88:01:02:13:41:66-02-0406"
    },
```

Is this expected behavior? It seems like I'm missing a real motion event when this happen, but I'm not sure yet.

manup, 2018-11-12:
>Is this expected behavior?

Yes there are separated events for config and state.

oscar-b, 2018-11-12:
Allright, I'll handle it when receiving, thanks!

closed 2018-11-12 by oscar-b

ebaauw, 2018-11-12:
The sensor sends an attribute report message per cluster, which deCONZ maps to the appropriate attribute(s):
- _Occupancy Sensing_ to `state.presence` of the ZHAPresence resource;
- _Illuminance Measurement_ to `state.lightlevel` of the ZHALightLevel resource;
- _Temperature Measurement_ to `state.temperature` of the ZHATemperature resource;
- _Power Configuration_ to `config.battery` of all three resources;
- _Basic_ to `config.sensitivity` of the ZHAPresence resource and `config.usertest` of all three resources.

deCONZ updates `state.lastupdated` when the corresponding report is received, and it computes `state.lux`, `state.dark`, and `state.daylight` from `state.lightlevel`.

Each attribute report results in a web socket notification.  The reports for the clusters corresponding to `state` attributes are sent every 5 minutes (and when the value changes significantly); the reports for the `config` attributes are sent every 2 hours (and when the value changes).  So you would expect (at least) three `state` web socket notifications per 5 minutes, and (at least) two `config` web socket notifications per 2 hours.

Clear `websocketnotifyall` through the API (`ph put /config '{"websocketnotifyall": false}'`), to receive only the changed attributes in the notification, instead of all `state` or `config` attributes.  I think this suppresses the `config` notifications when `config.battery` or `config.sensitivity` and `config.usertest` haven't changed.  You'll still receive all `state` notifications, since `state.lastupdated` has changed.

oscar-b, 2018-11-12:
Thanks @ebaauw very much appreciated! 🍻

Trying to set up the lighting automations in node-red, seems like a really nice systems but there are less ready nodes for lighting automation than I thought. Might try to do some deCONZ nodes with your ph tool :)