dresden-elektronik/deconz-rest-plugin #945: How to handle ZigBee Gledopto RGB-CCT Controller

saxn-paule, 2018-11-13:
I bought such a controller: https://www.ebay.de/itm/RGB-CCT-RGBW-WW-CW-Steuergerät-Controller-ZigBee-kompatibel-LED-Lichtband-12-24V/123410333065

I connected a RGB Stripe (V+, R, G, B) and a 12V LED spot (V+, C) to the controller.

Via Phoscon I’m able to control the different channels.

RGB is controlled via Color, Brightness and Saturation.
CCT is controlled via Brightness and WhiteLevel (warm/cold).

Unfortunately it's not possible to turn both channels on at the same time. When I enable the spot, the stripe is toggled off and the way around.

In addition, the controller goes offline after some minutes/seconds and isn't reachable any longer. Only by powering the controller off and on again it's reachable again.

Do you have any hints?

ebaauw, 2018-11-13:
I used to have loaners to integrate it in deCONZ and homebridge-hue, but I returned them, so I can no longer test them.  See https://github.com/ebaauw/homebridge-hue/issues/244.

Which one do you have?  If memory serves:
- The RGB-CCT is for a 5-channel light strip, with R, G, B, WW, CW, and V+ connections.  It has a single ZigBee extended color light endpoint, which deCONZ exposes as a single `/lights` resource.  It switches between RGB and WW/CW output, depending on the _Color Control_ command;
- The RGBW is for a 4 channel RGBW light strip, with R, G, B, W, and V+ .  It has two ZigBee endpoints, an extented (!) color light for RGB and a dimmable light for W.  deCONZ exposes it as two `/lights` resources, which are controlled independently.

You might want to double-check that you don't send a y-value of 0 to the extended color light.  The GLEDOPTO has the same (I assume divide by 0) bug as IKEA: the firmware hangs until a power cycle.

saxn-paule, 2018-11-13:
I have the RGB-CCT version. If I understand correctly, the RGBW version is the best solution for my setup?

I want to have two "devices" that can be controlled independently.

ebaauw, 2018-11-13:
The RGBW does offer two independent controls, so functionally: yes.

I only tested it with a single light strip - I don't know how happy it will be with different loads (strip vs spot) on the different channels.  I have heard of the dresden FLS-PP lp ballast being used to drive simultaneously an RGB lightstrip and a separate monochrome lightstrip.

saxn-paule, 2018-11-14:
Okay, thank you for your answer. I ordered a RGBW controller and let you know the result when I was able to test it.

shadowtsw, 2018-11-16:
Since version  2.05.46 a new connected RGBW GLEDOPTO is shown as light with 3 endpoints ?! Never had this issue before, I already running 3 controller´s RGBW and one RGB-CCT.

2 of these endpoints are the same to control RGB.

Any idea ?

saxn-paule, 2018-11-22:
My RGBW GLEDOPTO arrived today. It's only detected as one endpoint and one unknown device.

Here you can see the RGB-CCT (008) and the RGBW (007) Controller:
[image]

Klagio, 2019-01-24:
> * The RGB-CCT is for a 5-channel light strip, with R, G, B, WW, CW, and V+ connections.  It has a single ZigBee extended color light endpoint, which deCONZ exposes as a single `/lights` resource. [...]

Ohhh, I thought it would be the opposite: the RGB CCT with 5 channel has less commands then the RGBW with 4 channels??? How strange. 

I bought the RGBCCT model so I could use 5 channels: RGB cold white and warm white ...

shadowtsw, 2019-01-24:
> Ohhh, I thought it would be the opposite: the RGB CCT with 5 channel has less commands then the RGBW with 4 channels??? How strange.
>
> I bought the RGBCCT model so I could use 5 channels: RGB, cold [...]

I guess most people misunderstand the behaviour of an cct controller. What you are searching for is RGBWWCW and not RGBCCT. CCT means color temperature control and will be mixed to the required kelvin by ww and cw.

Klagio, 2019-01-24:
I see, but now I ordered 6 pin led to go with the CCT driver, if I want only cold white, or only warm white, how can I control it with saturation and RGB???

shadowtsw, 2019-01-24:
usually there must be at least 3 sliders (color, brightness, saturation), please post a screenshot

shadowtsw, 2019-01-24:
as I am at work, I've tried to find a picture. Please see
[image]

shadowtsw, 2019-01-24:
you need to switch to color temperature, then 2 sliders appear to control the white channels (brightness and temperature).... changes on this will turn off the rgb channel afaik

shadowtsw, 2019-01-24:
but as I understand correctly.... its not an problem, this is the way the cct controller works...... you also cannot control the hue lights with rgb and cct at the same time.... this results in crazy colors xD

shadowtsw, 2019-01-24:
for example, because of this I just work with RGBWW controllers.... I'm able to mix white with the rgb channel and then turn on WW...... my advantage is to use all LEDs on my rgbww stripe with nearly the same result.... for this I have a HUE table 
http://planetpixelemporium.com/tutorialpages/light.html .... works perfect for me

saxn-paule, 2019-01-24:
I did a quick test. I also have a GLC-007 (RGBW) controller here. It's the same. The RGB LEDs are used to control the temperature of the white color.

So its not possible to control two different devices (RGB + W) with this controller.

saxn-paule, 2019-01-24:
I also did a test with the GL-009 WW controller.
You could attach two lights to it, but you aren't able to control them independently.

Klagio, 2019-01-24:
> usually there must be at least 3 sliders (color, brightness, saturation), please post a screenshot

Ahh ok.

So to recapitulate with RGB CCT  (008 from Gledopto) you get 
Brightness+Color+Saturation
when click in color temperature I get
Brightness+ Temperature (cold/warm)

But, I also have a dresden FLS-pp and that has:
Brightness+Color+Saturation+White
when click in color temperature I get
Brightness+ Temperature (cold/warm)+ White

Basically the dresden (whuich has if I am not mistaken one pinout less) is better then Gledopto RGB CCT?

shadowtsw, 2019-01-24:
in another thread I‘ve posted that new versions of the gledopto devices don’t work proper anymore.... I have 3 old devices which can be controlled independently.... but to be honest, my last use of this setup was 1 month ago.... I will test it tomorrow, when I‘m back at home

shadowtsw, 2019-01-24:
regarding FLS-pp I don’t know, I just have GLEDOPTO

shadowtsw, 2019-01-24:
especially for you I was at home for testing...... everything runs well, see
ww
[image]

rgb (green)
[image]

rgb+ww
[image]

shadowtsw, 2019-01-24:
> I did a quick test. I also have a GL-007 (RGBW) controller here. It's the same. The RGB LEDs are used to control the temperature of the white color.
> 
> So its not possible to control two different devices (RGB + W) with this controller.

rgbw has no color temperature ?! (even through rgb control if it appears)
its just a dimmable light

shadowtsw, 2019-01-24:
> I also did a test with the GL-009 WW controller.
> You could attach two lights to it, but you aren't able to control them independently.

i don't know what gl-009 exactly is, but if its the ww/cw controller it works also fine, because its an cct-controller, you cannot set up each channel to 100%, because it adjusts the color temperature..... please correct if I'm wrong (regarding model gl-009 would be cw/ww)

shadowtsw, 2019-01-24:
ok, I understand..... and you have 2 lights added in phoscon for your w/w controller ?

these are my endpoints for rgbww
[image]
[image]

shadowtsw, 2019-01-24:
> The 009 is only a dimmer controller for two LEDs. Ports: V+ / W / V+ / W
> 
> https://de.gearbest.com/diy-parts-and-tools/pp_009893773954.html

costs less on aliexpress ;)

saxn-paule, 2019-01-24:
Was just an example. Got them for 15$

shadowtsw, 2019-01-24:
ok, I just use the rgbww setup in my whole house and it fits all needs. I also use it just for white although I have the rgbww stripe connected..... the price was similar for all parts compared to any other combination.... why not..... but I have some controllers with firmware 1.0.3 which triggers many problems..... but every controller has at least 2 endpoints as mentioned above.... one controller actually shows 3 endpoints (in addition one channel for Home Automation) .... but with the next updates hopefully it will be fixed

shadowtsw, 2019-01-24:
> My RGBW GLEDOPTO arrived today. It's only detected as one endpoint and one unknown device.
> 
> Here you can see the RGB-CCT (008) and the RGBW (007) Controller:
> [image]

but you also have 2 endpoints or why light 8 was not listed ?

saxn-paule, 2019-01-24:
Yes, all my GLEDOPTOS (007/008/009) have two endpoints. All have firmware 1.0.3

shadowtsw, 2019-01-24:
then you need to be patient :D

shadowtsw, 2019-01-24:
I also control all lights via node red usually, but controlling unknown will cause crashes instantly

saxn-paule, 2019-01-24:
Yes, because of that I only control the named ones and ignore the unknown.

shadowtsw, 2019-01-24:
we have the same problem, but as manup says, it could be fixed in the next versions

dmalene, 2019-04-23:
I believe the 2ID RGBW Controller on [AliExpress ](https://www.aliexpress.com/item/GLEDOPTO-ZIGBEE-Led-Controller-RGB-CCT-WW-CW-zigbee-controller-LED-DC12-24V-LED-strip-controller/32858603964.html?aff_platform=link-c-tool&cpt=1556032808880&onelink_item_from=32858603964&onelink_thrd=0.0&onelink_page_from=ITEM_DETAIL&onelink_item_to=32858603964&pvid=57ecbb01-fc69-4c1a-8a86-05ce42b2a5eb&onelink_duration=0.767095&sk=SUaOvsy&aff_trace_key=ee668c2b5f7a4b4d9213c6fbf91c3f94-1556032808880-09533-SUaOvsy&onelink_status=noneresult&scm=1007.22893.125779.0&terminal_id=b8553613810e4e98b4ca8fd1811503b1&onelink_page_to=ITEM_DETAIL)will allow you to control each light individually. It's a new model.

saxn-paule, 2019-04-23:
Sounds good. Just ordered two of them RGBW and RGB-CCT.

saxn-paule, 2019-05-06:
They arrived today and are named GL-C-007 and 008. Very disappointing.

If that wouldn't be enough, they don't even pair with deconz :(

closed 2019-09-10 by stale[bot]

airens, 2019-10-16:
Can anybody tell me if its possible to turn on\off light strip smoothly using this controller and deconz?

guny74, 2021-01-29:
Hello,

I have a GL-MC-001s (Multicolor) RGB-CCT Mini Controller (RGB + WW +CW).
It appears in DECONZ as one light.

I have two problems: 

1. I want to have a "scene" OFF, but the strip (RGB leds) "glow" always a little bit. Only after via a real "off" command it is out.
2. I also want a "scene" "warmWhite" where only the the WW/CW Leds on. I change ist in deconz, save the scene, als seems fine.
   But selecting the "scene" the RGB Leds are on with a "cyan" color.

  What can I do ?

Thanks

Regards