dresden-elektronik/deconz-rest-plugin #946: Help Pairing Samsung Leak Sensor

sktaylortrash, 2018-11-14:
Apologies if this is not the place to ask. Has anyone successfully paired the new SmartThings leak sensor. I can't get anything to show up in deconz. It pairs instantly with a SmartThings hub of course, but despite repeated tries no joy with deCONZ.

ebaauw, 2018-11-14:
I don’t think that is yet supported by deCONZ.  Please see 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

ebaauw, 2018-11-14:
Until support is added, the sensor won't show in the REST API, and therefore not in Phoscon nor in the Web app.  You need to search for it in one of these apps, and then look for the node in the deCONZ GUI on the server.

sktaylortrash, 2018-11-14:
Okay well I guess I'll just keep trying because it doesn't show up in the deCONZ gui on the server. That's where I was looking for it

ebaauw, 2018-11-14:
Is it a ZIgBee device?

sktaylortrash, 2018-11-14:
It says it is in the manual and on the box

ebaauw, 2018-11-14:
Ok, then the trick is to figure out how to reset it, so it wants to join the network.  Make sure you run a recent version of deCONZ.  You might want to power down all your ZigBee routers (lights), and keep the sensor in range of the RaspBee or ConBee, to force a direct connection.  Make sure there's still room for new end devices on the RaspBee or ConBee (recent firmware versions allow 32, older versions only 10).

sktaylortrash, 2018-11-14:
So I did update to the newest release versions I could find.  
Version 2.05.47 / 11/13/2018
Firmware 262F0500

Powered down everything but the Pi with the RaspBee and I only have 2 endpoints according to deCONZ which makes sense as I only have 1 motion sensor and 1 remote.   

The reset procedure is hold the button for 5 seconds until the red light flashes. 
I tried 15-20 times to no avail, yet it works over and over again with an older smartthings hub.  
I tried with the sensor next to the Pi and a few feet away in case that made a difference as well

Is it possibly because the device is ZigBee 3.0 the ConBee page lists itself as compatible but the RaspBee page makes no mention of it

manup, 2018-11-14:
Just to check, did you also put deCONZ in sensor search mode?

*Phoscon App > Menu > Sensors > Add new sensor > Other*

manup, 2018-11-14:
>Is it possibly because the device is ZigBee 3.0 the ConBee page lists itself as compatible but the RaspBee page makes no mention of it

They both share the same software and firmware, so it doesn't make a difference, the sensor should at least join the network.

sktaylortrash, 2018-11-14:
I tried both from thePhoscon as described and also from the old WebApp and just opening the network neither was successful

sktaylortrash, 2018-11-14:
Okay so I unboxed a different unit and it showed up immediately so here the pics. No idea what I did different
[image]
[image]

[image]

Hope I did that right

ebaauw, 2018-11-15:
Cool.  Can you also post screenshots of the _IAS Zone_ and _Temperature measurement_ clusters, _after_ reading the attributes.  You need to wake the sensor (by briefly pressing the button) before the attributes can be read.

sktaylortrash, 2019-03-21:
This issue should not be closed as the sensor still is not supported

closed 2019-07-26 by stale[bot]

srg74, 2019-08-06:
Sensor is paired and show proper values in the beginning and suddenly became unavailable. After pairing again working for some time and disappear again.

sktaylortrash, 2019-08-06:
> Sensor is paired and show proper values in the beginning and suddenly became unavailable. After pairing again working for some time and disappear again.

I'm not seeing that issue

srg74, 2019-08-06:
> > Sensor is paired and show proper values in the beginning and suddenly became unavailable. After pairing again working for some time and disappear again.
> 
> I'm not seeing that issue

[image]

MrHollowPS, 2020-01-11:
Sorry to dig this up. I'm trying to pair a first gen waterleak sensor, it shows up in the zigbee mesh but it doesn't get detected by the gateway in the phoscon app so i cannot use it in Home Assistant with deCONZ. Any ideeas?

jdraxgit, 2021-05-26:
> Sorry to dig this up. I'm trying to pair a first gen waterleak sensor, it shows up in the zigbee mesh but it doesn't get detected by the gateway in the phoscon app so i cannot use it in Home Assistant with deCONZ. Any ideeas?

I have the same issue, any fixes/updates?