dresden-elektronik/deconz-rest-plugin #947: Ikea Tradfri not updating / idle

iRazoR, 2018-11-14:
Hi@all,

first of all thanks for the great work at Raspbee/DeConz. It works almost perfectly since about two years but recently I installed some Ikea Tradfri Bulbs and found out that there is the ability in DeConz to Update them.

Unfortunately the Update function does not work for me. I downloaded the latest Firmware Files using the provisioned python script, enabled the OTAU Plugin, selected a Bulb and started "Query". Apparently the Query finished but no update starts, even due there are new updates available.

For Example I want to update a Ikea Tradfri GU10 Bulb.
According to Phoscon the current firmware of the Bulb is 1.2.214 but the one downloaded is 1.2.217 so it should trigger an update. Even if I choose the Update File manually and hit "update" it just changes the Progress to "Idle" and nothing happens also.

[image]

[image]

[image]

Thanks for any hints/ideas.

Cheers
David

manup, 2018-11-14:
It looks like that is not the correct firmware file albeit GU10 the ws / w matters (color temperature / dimmable only).

The firmware files are differentiated by the image type. deCONZ should pick the correct file on it's own as long as they are in the otau directory.

A few notes:

* After executing the Ikea download script, restart deCONZ so it can pickup the files
* To update a light powercycle it once, it should aks the gateway for an update and OTa should start automatically

iRazoR, 2018-11-14:
Shame on me... You are right, it is the wrong file... But that means that there is no update File for the dimmable only GU10 Bulbs (sure not a Deconz problem)?

Also learned something new: The Updates get picked up only after DeConz Restart. Is that the normal behaviour? Because usually I never Restart DeConz because is running on a (more or less) headless Raspberry.

I actually cannot check if the other Bulbs are up to date because they are switched off and I´m not at home but I have a Dimmer Switch and one Remote Control. Do you know how to upgrade them?
I read that there are usualy in Sleep mode and that the Battery might need to get replaced during/after update but how do I start the update? Do I need to wakeup the Remote first?

lbschenkel, 2018-11-14:
FYI: I had the same (or a very similar issue) some months back. I tried to upgrade a TRÅDFRI motion sensor and the upgrade simply never started (always "idle"). I'm 100% confident that it was the right file. It took many hours of frustration and multiple tries of restarting deCONZ, rejoining the device, until it eventually worked. Unfortunately, since I tried many different things, I'm not sure what exactly was the action that solved it. Since then, all my new purchases came with the latest firmware so I had no opportunity to test firmware upgrades on recent versions of deCONZ.

iRazoR, 2018-11-14:
Thanks for the hint.
For the moment I think manup was right, all the Bulbs seem to be updated.
For the Dimm Switch the detection seems to work, at least DeConz selected a File for Update for it but again it is staying on Idle. I think I will have to play a bit around with it and try to repair it. Hopefully the Update works then.

One more question: Does anybody have the Ikea Dimm Switch working in Deconz? I managed to pair it and added a Tradfri Bulb to the group of the switch but I cannot control the Brightness. When I turn the Switch I see in DeConz, that Signals are sent but nothing happens with the buld... Any idea?

manup, 2018-11-14:
>But that means that there is no update File for the dimmable only GU10 Bulbs (sure not a Deconz problem)?

Yes looks like that, the 1.2.xxx releases are the very latest stable firmware files.

>The Updates get picked up only after DeConz Restart.

The deCONZ restart is only needed once after downloading the files with the Ikea python script.

The automatic update is only activated for lights, for battery devices it is disabled since it could drain the battery a lot.

To update a battery device:
* select it in the table
* select the related update file
* click update button
* pull out the battery
* put it in again

The update should start soon after that.

iRazoR, 2018-11-14:
Perfect, thanks for the instructions. I will give it a try tomorrow.

iRazoR, 2018-11-15:
I had time to test that now and still it doesn´t update.
The Dimmer always gets detected as updateable because a Update File is selected automatically but Dimmer stucks on "Idle".
I tried the Battery Methode couple of times, rebooted and re-paired the Switch but still doesn´t work.
Also with the current firmware the Dimmer does not seem to be compatible with DeConz because I still cannot control any light with it.

Does anybody know, if I can pair the Tradfri Dimmer directly with a Hue Colored Bulb without any Bridge? That would be my alternative Solution.

[image]

iRazoR, 2018-11-15:
Ok, somehow I managed to pair the Dimmer with my Hue Bulb but the procedure is definetly not user-friendly and Im afraid it wont work anymore after a reboot... I will see.
Unfortunately the Update still didn´t work, maybe I find some more time in the weekend to find out why.

closed 2018-11-15 by iRazoR