dresden-elektronik/deconz-rest-plugin #950: Restore old State

Fl0ca, 2018-11-15:
Hey,
is it possible to develop an option, that when an power blackout passed by, all Lights return to there old State? 
Tradfri and Hue Lights are going to be active, when they had a "Soft out" and then an power loss. 

Other Systems like Homee are sending the Status again, if the "restore old State" option is activated and the Light starts communication with the Gateway after power loss. 

Would be really great if this comes to deconz too...

ebaauw, 2018-11-15:
The soon to be released new firmware for the Hue lights is supposed to fix this ridiculous behaviour.  See https://github.com/ebaauw/homebridge-hue/issues/41#issuecomment-436936220.

For powering off all lights on deCONZ startup, see https://github.com/ebaauw/homebridge-hue/issues/41#issuecomment-317155887.

For powering off all lights after a power restore, using a USB power bank as a cheap UPS for your Raspberry Pi, see https://github.com/ebaauw/homebridge-hue/issues/41#issuecomment-387698046.

Fl0ca, 2018-11-15:
I am using Tradfri, so for the Main Bulbs it has not effect, if Philips is Changing something. 
Your Workarounds are only for your homebridge-hue addon, when it is right what i read.

Would be Great, if deconz will support it out of the box, i have von apple devices so homebridge is senseless for me :/

Fl0ca, 2018-11-16:
For a non programmer it is really difficult to understand how to integrate the rules. 

Is it not possible to implement a Checkbox, like the Homee guys did? 

I'm not totally a noob, but for me it is difficult. I know some other Users, which where lucky when they get deconz running, what is with these guys? 
Implementation in the GUI would help everyone...

closed 2019-05-12 by stale[bot]