dresden-elektronik/deconz-rest-plugin #952: GLEDOPTO Zigbee RGBW shows 3 edpoints ? and no model

shadowtsw, 2018-11-16:
Since version  2.05.46 a new connected RGBW GLEDOPTO is shown as light with 3 endpoints ?! Never had this issue before, I already running 3 controller´s RGBW and one RGB-CCT.

2 of these endpoints are the same to control RGB.

Any idea ?

_Originally posted by @shadowtsw in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/945#issuecomment-439459700_

shadowtsw, 2018-11-16:
[image]
the new controller

and my old controller:
[image]

same model, from the same seller. All my "old" controller were added a month ago, the new one last week. I´ve tested again with version 2.5.47, same issue.

ebaauw, 2018-11-16:
Odd, the manufacturer and model show as unknown.  Also `GL-C-007` is a model I haven’t seen before.

Could you post a screenshot of the node and the _Basic_ cluster in the deCONZ GUI?  And the REST API resources?  See 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

shadowtsw, 2018-11-17:
Several month ago I´ve tried to access the deconz gui. As I can remember for this I have used x11, but unsuccessful.

My raspberry is running an openhabian image with deconz headless. Does it work with tight VNC server ? To start the GUI I just need the command "systemctl enable deconz-gui", right ?

shadowtsw, 2018-11-21:
Ok, for you I´ve tested with an clean image with deconz, here comes the result (light 29 is similar to my posted light 25):

[image]

[image]

[image]

[image]

ebaauw, 2018-11-21:
These are definitely a different model and/or firmware from the ones I've tested.  These had ZLL endpoint 0x0A for the dimmable light, ZLL endpoint 0x0B for the (extended) color light, and ZLL endpoint 0x0D for commissioning, _Date Code_ `20131206` and (interestingly the same) _SW Build ID_ `1.0.2`.

Apart from relocating the 0x0A endpoint to 0x0F, they seem to have added a 0x0C ZHA endpoint.  My best guess is to support ZHA-only gateways, that cannot handle the ZLL endpoints (or device type).  This seems consistent with your experience that two light resources actually provide the same RGB control.

ZLL-only gateways, like the Hue bridge, will ignore the ZHA endpoint.  I suppose the solution is simply to blacklist the ZHA endpoint in the deCONZ REST API, preventing it from creating a `/lights` resource for it.  That might actually be harder than it sounds, if the controllers don't report _Model Identifier_ or _Manufacturer Name_ in the _Basic_ cluster.  They use _Manufacturer ID_ of 0x0000 and a TI mac prefix - that combination doesn't seem specific enough to identify them as GLEDOPTO.

saxn-paule, 2018-11-22:
It's the same with the RGB-CCT (GL-C-008)

[image]

ebaauw, 2018-11-22:
Odd. The ones I tested (as well as the Floodlight I still have) return values for the _Basic_ cluster attributes on all endpoints.

@manup, we need to copy attribute values from the `/lights` resource for the endpoint that does report them to the `/lights` resource(s) of the other endpoint(s).  This is needed for the dual-key Xiaomi wired in-wall switches as well, which only have a _Basic_ cluster on the first light endpoint.  This is already done for the `/sensors` resources in `addSensorNode()`, but that doesn't translate easily to lights.

I feel like these attributes should actually be stored at the node level, rather than at the resource / endpoint / REST node level.  This way, all resources can access these values directly, instead of looping over all REST nodes with the same mac address.  However, that would require creating a subclass of `deCONZ::Node` in the REST API plugin and changing the types of `RestNodeBase.node()` and `RestNodeBase.m_node` to that subclass.  I'm not sure if we should go there.

shadowtsw, 2018-11-23:
I understand. Do I have a to do ?

saxn-paule, 2018-11-27:
Are there any updates on this?

saxn-paule, 2018-11-29:
I just tested to use a RGB Stripe and W led modules at the same time.

When RGB is touched/changed via Phoscon, the W channel is turned off.

https://static.mydealz.de/live/comments/content/xEHUp/19732857.jpg

If the W channel is turned on, the saturation of the RGB channel is gone.

https://static.mydealz.de/live/comments/content/7Op3k/19732857.jpg

What seems blue here was red before and is very cold white now. No saturation at all.

leifnel, 2018-12-16:
I have a  RGB-CCT (GL-C-008), but only RGB, no white LEDs.
Is it possible somehow to reflash it, so other systems like IKEA and Hue only sees it as RGB, turning all on for white instead of trying the non-existing white?

Or would it work to put 3 diodes from W to the RGB connections?

shadowtsw, 2019-01-06:
....would you fix this issue in later updates ? I also got a new controller 2 weeks ago and this has just 2 endpoints, but firmware 1.0.3. It was marked as unknown.

My old rgbw controller cannot control furthermore rgb and w channel seperately ? both lights now act as rgbcct with color and temperature. Before the last updates it just was an dimmable light. What happened ?

> Is it possible somehow to reflash it, so other systems like IKEA and Hue only sees it as RGB, turning all on for white instead of trying the non-existing white?
I think this is not the thread regarding flashing gledopto devices.

shadowtsw, 2019-01-23:
hello again....
I mainly work with these controllers in my house, so I would appreciate if you could add these points to the raspbee gateway.... further openhab2 crashs because its not able to handle "unknown" or NULL devices.... if you need any information, please let me know

Klagio, 2019-01-24:
> I have a RGB-CCT (GL-C-008), but only RGB, no white LEDs.

same here

shadowtsw, 2019-01-24:
> > I have a RGB-CCT (GL-C-008), but only RGB, no white LEDs.
> 
> same here. But it should be 2 white LED to be perfect

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/945

saxn-paule, 2019-05-25:
Is there any progress regarding the gledoptos?

ebaauw, 2019-05-25:
You should no longer see any unknown lights; the attributes from the _Basic_ cluster (like `modelid` and `manufacturername`) should now be copied to all endpoints, as per above commit.

The RGB+CCT based controllers and lights (with one _Extended color light_ endpoint) power either the RGB (on setting `xy`) or the CW/WW channels (on setting `ct`), see #1205.  Nothing we can do about this. The RGBW based controllers and lights (with a _Color light_ and a _Dimmable light_ endpoint) power the RGB and W channels separately, from the respective endpoint, see #1204.  These don’t support `ct`.

I’m puzzled about https://github.com/dresden-elektronik/deconz-rest-plugin/issues/952#issuecomment-443005399.  From the pictures, this is a four-channel, RGBW controller.  Either Phoscon sends some unexpected commands or the controller doesn’t like the combined load, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/945#issuecomment-438473796.   Best check by using the API directly.

I still haven’t gotten my hands on a controller with three endpoints, as shown above.  I suppose the REST API should suppress creating a `/lights` resource for the superfluous ZHA _Color dimmable light_ endpoint, but I’m not sure how best to detect this: on endpoint number, on device type, or on software version?  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/952#issuecomment-440831390.  It would help if I could borrow such controller.

The issues with the light not turning on or off when receiving a _Move to Level (with On/Off)_ command still need to be worked around in deCONZ, refactoring the handling of setting a light state (or group action), see #1111.

closed 2019-09-29 by stale[bot]