dresden-elektronik/deconz-rest-plugin #955: setting up sandbox environment - need for specific config of networks?

ErnstEeldert, 2018-11-17:
I am running a rather large network on a raspbee. To avoid disturbing the 'production' environment, I figured it would be a good idea to setup a separate sandbox environment, which I can use to play with new devices and do debugging if required. To that means, I've bought a conbee, which will run on a separate PI.

Are there any things I need to consider so that both environments do not interfere with one another? Network ids? Other specific controller settings?

Thanks!

manup, 2018-11-17:
The two networks should just work, every network has a random network id (panid, extended panid). Further it is possible (optional) to use separated channels for the two network.

ErnstEeldert, 2018-11-22:
Thanks for the clarification!

closed 2018-11-22 by ErnstEeldert