dresden-elektronik/deconz-rest-plugin #956: RPi with RaspBee and NUC with Conbee on the same network, Phoscon confused?

jantje88, 2018-11-17:
I am using a RaspBee on a RPi 3B+ and want to migrate to a NUC with Conbee.

http://192.168.1.162 = NUC
http://192.168.1.163 = Raspberry Pi

After installing the HASS and DeCONZ dockers on the NUC and opening the Phoscon app on the NUC @ http://192.168.1.162:8383/pwa/ it shows two gateways: https://i.imgur.com/kZgX4kU.png

When logging on the to Phoscon app on the NUC (http://192.168.1.162:8383), it shows the RaspBee as connected product: https://i.imgur.com/q6UkLqX.png (But: no lights, switches, sensors are visible)

Is the Phocon app confused because there are two gateways on the same network?
The Conbee is on ttyUSB1 because another USB device (USB/P1 port for DSMR smart meter) is using ttyUSB0.

My docker compose to start the DeCONZ docker on the NUC:

```
  deconz:
    image: marthoc/deconz
    container_name: deconz
    network_mode: host
    restart: always  
    volumes:
      - /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ
    devices:
      - /dev/ttyUSB1
    environment:
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - DECONZ_DEVICE=/dev/ttyUSB1
      - DECONZ_WEB_PORT=8383
      - DECONZ_WS_PORT=8444
```

DeCONZ on HASS.io on the Raspberry Pi is using other ports: https://i.imgur.com/RZ00FYG.png

Also after shutting down the Raspberry Pi with RaspBee and resetting the gateway (Phoscon-GW) on the NUC, the Phoscon app on the NUC keeps telling me a RaspBee is connected..

manup, 2018-11-22:
>When logging on the to Phoscon app on the NUC (http://192.168.1.162:8383), it shows the RaspBee as connected product: https://i.imgur.com/q6UkLqX.png (But: no lights, switches, sensors are visible)

Is it working when you directly open the Phoscon App for 192.168.1.163?

>Is the Phocon app confused because there are two gateways on the same network?
The Conbee is on ttyUSB1 because another USB device (USB/P1 port for DSMR smart meter) is using ttyUSB0.

Nope, in fact the Phoscon App can connect to multiple gateways. The REST-API requests just go to the selected gateway but the Phoscon App itself is loaded from the gateway shown in the browser url bar.

We use that a lot, in the company we have > 15 gateways (RaspBee and ConBee) :)

When logged into a ConBee gateway it should look like:

[image]

**However** Did you load a backup from the RaspBee gateway to the NUC? This could be indeed confusing since the gateway id would be cloned.

jantje88, 2018-11-22:
> Is it working when you directly open the Phoscon App for 192.168.1.163?

How do you mean this exactly? I guess you mean .162 (NUC) with portnumber? So http://192.168.1.162:8383/pwa/?

When I open Phoscon on the NUC (http://192.168.1.162:8383/pwa) it shows two gateways: https://i.imgur.com/Urv64G2.png
When opening the Gateway on the NUC (@192.168.1.162) it shows a RaspBee: https://i.imgur.com/OLZch77.png

No, I did not load a back-up from the RaspBee on the NUC..

jantje88, 2018-11-22:
> > How do you mean this exactly? I guess you mean .162 (NUC) with portnumber? So http://192.168.1.162:8383/pwa/?
> 
> Hmm strange it should indeed show 192.168.1.162 Phoscon-GW as ConBee not as RaspBee, might be an ui glitch, but it's just a product name and shouldn't have impact on functionality.

Ok, and what about firmware updates? Should this work as intented?

manup, 2018-11-22:
>Ok, and what about firmware updates? Should this work as intented?

Currently updating firmware in Docker image requires a bit of work, see:

https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

We are working on a new version of the update tool which will simplify this a lot but it will take some more time till it's ready.

jantje88, 2018-11-22:
> Currently updating firmware in Docker image requires a bit of work, see:
>
> https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware
>
> We are working on a new version of the update tool [...]

Thanks. The same process as for my RaspBee, that will be fine.

jantje88, 2018-12-02:
Not sure why and since when, but Phoscon on the NUC shows me the Conbee instead of the raspbee now :-)

closed 2018-12-02 by jantje88