dresden-elektronik/deconz-rest-plugin #959: Ikea tradfri GU10 led dont pair, help!!

PrimaryInc, 2018-11-18:
Hi, 

I got a issue with Conbee and Ikea Tradfri bulbs being unable to pair bulbs with ConBee bridge.
Environment Windows 2016 x64
ConBee USB Stick firmware 26240500
DeConz version 2.05.48
DeConz installed to C:\Program Files (x86)\deCONZ not defaul user profile path.

I have (perhaps not a good ide) installed version 2.05.48.
ConBee have paired 39 Ikea bulbes from before and had no issues pairing them.

So this week I recived 30 new bulbs to be added to ConBee. So after trying to pair them I've only been able to pair 2 bulbs. Ikea FW 1.2.214, same as all the others. 
So to the big question, wy is this happening and how can i solve it?

Thanks!!

manup, 2018-11-18:
I highly suggest to update the firmware to latest version 0x262f0500.
Note: 0x26210500 – 0x262c0500 had some nasty issues.

If that still doesn't work you might need to use the Ikea reset sequence while light search is running as shown in this video:

https://www.youtube.com/watch?v=mJm9YpPrGzk

PrimaryInc, 2018-11-18:
Hi @manup ,

Thanks fast reply, I have upgraded FW when asked by DeConz. But I guess that firmware is not in there?
I the case do you have a link to new firmware and a little guide on how?

Thanks :)

manup, 2018-11-18:
Upps, yes forgot to pack it with the Windows package.

Here it is: https://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x262f0500.bin.GCF

PrimaryInc, 2018-11-18:
Thanks @manup 

Do i Upgrade FW via DeConz app or 3de party flash prog?

manup, 2018-11-18:
The easiest would be put it in the firmware folder where the other *.GCF files are (not sure on the path you may use Windows Explorer and search for *.GCF). Then restart deCONZ.

Shortly after a update button should appear in Phoscon App > Settings > Gateway.

PrimaryInc, 2018-11-18:
@manup 

I see the oath is C:\Program Files (x86)\deCONZ\firmware under here it is 8 different FW versions, the fw version currently running is  one of the older once in there. Could it be a problem that FW is not upgrading as expected?
No update button after restart I'm afraid.

manup, 2018-11-18:
One reason for no update button is shown is when you have multiple USB devices (like ConBee and some other USB dongles) connected which are using a FTDI chip. Is that the case?

If so as a workaround you could unplug the other devices for the update.

Otherwise manual update can be done on the commandline it's a bit cumbersome on Windows.
Not to myself: small howto on how to update manually in the Wiki.

PrimaryInc, 2018-11-18:
@manup 
Yes you are right, I have RFX, z-Wave and zigee on same server. I did get the update via command so everything now seems to work again.  Thanks for your help :)

closed 2018-11-18 by PrimaryInc