dresden-elektronik/deconz-rest-plugin #964: deleted sensor still showing in Home Assistant despite no longer in deCONZ

clementx2006, 2018-11-19:
I experimented with a couple of Xiaomi sensor / switch over the weekend. 

Both the leak sensor and the button switch seemed to be a pain to pair and only showed up in deCONZ and not in Phoscon - but I guess this has already been logged as open issues - and then I decided to delete them via 'Delete Selected Nodes' in deCONZ which was successful; however I then discovered that both of them were still showing in Home Assistant. So I am wondering whether there is any way of getting rid of them? I tried to restart the HA service, the deCONZ GUI to no avail...

clementx2006, 2018-11-20:
Tried to connect through the REST API, but for some reason I am getting 'Body contains invalid JSON'? This is after unlocking the gateway...

ebaauw, 2018-11-20:
Make sure there's no invisible characters in the body.  Run it through a JSON linter, like https://jsonlint.com.

clementx2006, 2018-11-21:
seems to be working! Thanks a lot!

closed 2018-11-21 by clementx2006