dresden-elektronik/deconz-rest-plugin #965: deCONZ_Rpi_0x26290500.bin.GCF

MattL0, 2018-11-20:
is deCONZ_Rpi_0x26290500.bin.GCF the last firmware?

ebaauw, 2018-11-20:
No, see https://www.dresden-elektronik.de/rpi/deconz-firmware/?C=M;O=D.

MattL0, 2018-11-20:
this is why I ask. There is no 0x26290500 there. And last time with 0x262f0500.bin.GCF this repo was not updated.

But I saw deCONZ_Rpi_0x26290500.bin.GCF in usr/share/deCONZ/firmware/ 

so i was wondering if it was just not updated on the repo you posted, like last time

Kane610, 2018-11-21:
It is hex, 9 is lower than f so it is easy to identify if the FW is newer.

PrimaryInc, 2018-11-21:
This is the latest FW located there :)
deCONZ_Rpi_0x262f0500.bin.GCF | 07-Nov-2018 19:21 | 123K |

MattL0, 2018-11-23:
haa ok i understand now thanks!

manup, 2018-11-23:
Mystery solved, closing this for now :)

closed 2018-11-23 by manup