dresden-elektronik/deconz-rest-plugin #967: Ghost resources

ebaauw, 2018-11-21:
The REST API plugin seems to think my Xioami wired in-wall switch is also a Hue motion sensor: 
```
$ ph get /sensors/2
{
  "config": {
    "duration": 60,
    "on": true,
    "reachable": false,
    "temperature": 4100
  },
  "ep": 2,
  "etag": "1b14fe91796700dcf69f175f9e003bc0",
  "manufacturername": "LUMI",
  "modelid": "lumi.ctrl_ln2.aq1",
  "name": "Presence 2",
  "state": {
    "lastupdated": "2018-11-21T18:33:55",
    "presence": false
  },
  "swversion": "10-12-2017",
  "type": "ZHAPresence",
  "uniqueid": "00:15:8d:00:02:4a:70:18-02-0406"
}
$ ph get /sensors/3
{
  "config": {
    "offset": 0,
    "on": true,
    "reachable": false
  },
  "ep": 2,
  "etag": "1b14fe91796700dcf69f175f9e003bc0",
  "manufacturername": "LUMI",
  "modelid": "lumi.ctrl_ln2.aq1",
  "name": "Temperature 3",
  "state": {
    "lastupdated": "2018-11-21T18:33:55",
    "temperature": 4100
  },
  "swversion": "10-12-2017",
  "type": "ZHATemperature",
  "uniqueid": "00:15:8d:00:02:4a:70:18-02-0402"
}
$ ph get /sensors/4
{
  "config": {
    "on": true,
    "reachable": false,
    "temperature": 4100,
    "tholddark": 12000,
    "tholdoffset": 7000
  },
  "ep": 2,
  "etag": "1b14fe91796700dcf69f175f9e003bc0",
  "manufacturername": "LUMI",
  "modelid": "lumi.ctrl_ln2.aq1",
  "name": "LightLevel 4",
  "state": {
    "dark": true,
    "daylight": false,
    "lastupdated": "2018-11-21T18:16:32",
    "lightlevel": 0,
    "lux": 0
  },
  "swversion": "10-12-2017",
  "type": "ZHALightLevel",
  "uniqueid": "00:15:8d:00:02:4a:70:18-02-0400"
}
```
Just upgraded to .48 and these resources magically appeared.  I rebooted the Raspberry after installing .48, because the Linux kernel was upgraded by `apt update`/`apt upgrade`.  Note that deCONZ aborted a couple of times before finally starting (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/797).  I don't know if the resources were created before or after the update - I only noticed them because they showed in HomeKit after restarting homebridge-hue.  The node looks normal in the deCONZ GUI, i.e. the clusters for these resources do _not_ show.  Also the fingerprint in the `device_descriptors` table seems OK, but the table now contains 583 rows for this device (with 8 endpoints, incl. ZDP).

I had seen this once before, but figured it could have been the result of me hacking the database too rigorously.  I remedied it by removing the sensor, deleting all database resources related to it (from the `nodes`, `sensors`, `devices`, `device_gui`, and `device_descriptors` tables), and re-pairing the sensor.  At that time, the fingerprint in the `device_gui` table reflected these additional clusters.  I think the GUI might as well, but I'm not sure about that.

I blacklisted the resources for homebridge-hue, so they don't bother me (for now).  Please let me know what info to provide for debugging this issue (obviously, after the routing issues are tackled ;-)

manup, 2018-11-23:
Interesting, can you please post a screenshot of the clusters. I figure the issue must happen somewhere in the addSensors() function?

ebaauw, 2018-11-23:
[image]

ebaauw, 2018-11-23:
>  I figure the issue must happen somewhere in the `addSensors()` function?

I think that function just does its thing, based on the fingerprint/list of clusters it retrieves.  Somehow the clusters of the joining Hue motion sensor were linked to the Xiaomi switch.

ebaauw, 2018-12-13:
I think the commit mentioned above does the trick.

Note that the resources were added to the broken `lumi.ctrl_ln2.aq1`, that resets (and rejoins the network) every 60-65 seconds, so there definitely were two sensors joining at the same time.

closed 2018-12-17 by ebaauw