dresden-elektronik/deconz-rest-plugin #968: Question: Rejoin attack possible?

Counterfeiter, 2018-11-23:
Hello,

I have a security question, is it possible to send device rejoin command and be part of the network?

I read this articel and the answer of ubisys:

https://m.heise.de/security/meldung/Deepsec-ZigBee-macht-Smart-Home-zum-offenen-Haus-3010287.html

Basicly it says: The zigbee spec is unsave, don‘t implement rejoin options in zigbee < 3.0 gateways.

Would be great to know if my gateway is as save as it can!

Best regards

Sebastian

manup, 2018-11-23:
Fun fact in the video a RaspBee is used ;)

>I have a security question, is it possible to send device rejoin command and be part of the network?

Yes, but only a **secured rejoin**.

>Basicly it says: The zigbee spec is unsave, don‘t implement rejoin options in zigbee < 3.0 gateways.

Nope, the important detail here is **insecure rejoin**  shouldn't be supported not *rejoin* in general.

Simplified version: A insecure rejoin is one which doesn't use the network key acquired during setup join.

Insecure rejoin is disabled in the RaspBee/ConBee firmware. Further we are also working on supporting install codes which secures also joining devices during setup. Currently not many devices support install codes but this will change in mid-future.

Counterfeiter, 2018-11-23:
Okay, good to know.

I don‘t get the point of install codes. Is it basiclly a AES key but manufacture dependent?!

closed 2019-03-30 by stale[bot]

wang70880, 2021-04-07:
> Fun fact in the video a RaspBee is used ;)
>
>
> Yes, but only a **secured rejoin**.
>
>
> Nope, the important detail here is **insecure rejoin** shouldn't be supported not _rejoin_ in general.
>
> Simplified [...]

I know this is an old thread, but seeing your discussion, I have some questions and comments.

- I heared that ZigBee 3.0 disables insecure rejoin, is that true? Where can I find official docs to address this?
- In ZigBee 3.0, since we introduce install codes, the security of link keys has been enhanced. Why does ZigBee 3.0 still want to remove insecure rejoin? I think keeping insecure rejoin can increase utility.
- Many ZEDs of different brands have disabled insecure rejoin, and only keeps secure rejoin. I don't think it is a good idea. Do you agree?