dresden-elektronik/deconz-rest-plugin #969: Fehler bei Installation Debian debian-9.6.0-amd64

mister2212, 2018-11-23:
Hallo
ich bin mit meinem iobroker auf einen intel nuc umgezogen. jetzt bekomme ich deon deconz usb stick nicht mehr installiert. bekomme immer den einen fehler paketarchitektur passt nicht. Wo bekomme ich eine passende installationsdatei her? Vielen dank für die Hilfe
[image]

ebaauw, 2018-11-23:
Try the Ubuntu package, see https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages.

mister2212, 2018-11-25:
funktioniert auch leider nicht. egal ob ubuntu oder raspbian
[image]

closed 2019-04-01 by stale[bot]