dresden-elektronik/deconz-rest-plugin #970: sensor unavailability not showing up

vindaalex, 2018-11-24:
I have installed the marthocoo deconz addon in HassOs and connected a bunch of tradfri lights (7) and aqara temperature sensors (6). When a light looses power it will show up in phoscon as unavailable. If however a sensor goes offline it will remain 'online' and keeps the last refresh time (tested now up to 13 hours). When the gateway is restarted sensors first will show as unavailable until new data is received thus it seems that 'unavailable' is possible for these sensors.

System setup:
rpi3b
Hassos 1.12
Conbee stick
marthocoo deconz addon 2.05.47

Is it possible to include a check if the sensors are alive and if not switch the sensor to unavailable. 

Thanx

manup, 2018-11-24:
Notable change: With 2.05.48 sensors are marked available after restart if they were reachable in the last 24 hours.

Marking sensors unavailable is tricky we need to add this function on a per device base.

vindaalex, 2018-11-24:
Thank you for the feedback. I use the aqara temperature sensors, do you expect that this function is implemented any time soon?  Or do you have any other suggestion how I could verify that sensors are still alive?

I would like to use the sensors for thermostat controll and therefore I want to check if sensors are working properly. My first implementation was a check of the update interval in home assistant only the refresh of deconz is not updated in HA when the temperature did not change. This results at some sensors in intervals of 3-4 hours while the deconz interval is a few mintues up to about 1 hour.

ebaauw, 2018-11-24:
I check that the sensor’s `state.lastupdated` holds a recent value.  As @manup indicated, “recent” differs per sensor: the Hue motion sensor resources are updated (at least) every five minutes, Xiaomi sensors might take over an hour between updates.  You should even be able to automate this in deCONZ rules, using a `stable` or `not stable` condition on the sensor’s `state.lastupdated`, but I never tried this myself.

Looking at `state.lastupdated` works well for regular sensors (motion sensors, door sensors, leak sensors, smoke sensors, etc), but not for ZHASwitch sensors.  Typically their `state` is only updated when a button is pressed.  Also the ZHAConsumption and ZHAPower sensors for Xiaomi plugs and wired switches only update when the device is on.  These sensors do report back regularly, though, reporting their battery status (every five minutes for the Hue dimmer) or through the Xiaomi special report.   Unfortunately, there’s no way to see this through the REST API.  I’ve been fantasising about adding `config.lastupdated`, but haven’t yet found the time even to investigate this.

vindaalex, 2018-11-24:
Thank you for the replies. It seems to be an easy question but not that easy to solve. If I understand it correctly, the above metnoined options are based on updates from the sensor itself not a check (like a ping, poll) to the sensor initiated by deconz. Would it be possible to check these battery operated sensors this way?

ebaauw, 2018-11-24:
> [...]  a check (like a ping, poll) to the sensor initiated by deconz. Would it be possible to check these battery operated sensors this way?

No, see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/End-device-Polling.

closed 2019-03-31 by stale[bot]

Rantaki, 2019-06-10:
Is there a plan to make this work? My use case is to have leak sensors that trigger an alert if there is a leak. It is very important to know if the sensor is not online, for a reason or another. Otherwise the system cannot be trusted.

ottelo9, 2019-06-15:
I use this sensor template for all my Hue Motion Sensor. Template creates battery level and correct battery symbol (0...100% battery symbol). If the last_updated value of the temperature is 10 minutes older than actual time then the sensor battery value = 110% and battery symbol = ?

my sensor name:
binary_sensor.einfahrt_sensor_move = Presence Detection
sensor.einfahrt_sensor_temp = Temperature

configuration.yaml:
`sensor: !include sensors.yaml`

sensors.yaml:
```
# Weather prediction
- platform: yr

# Batterie Level Motion Sensoren
- platform: template
  sensors:
    einfahrt_sensor_battery:
      friendly_name: "Einfahrt Sensor Batterie"
      value_template: >
        {% if as_timestamp(now()) > (as_timestamp(states.sensor.flurog_sensorog_temp.last_updated) + 600) %}
          110
        {% else %}
          {{ states.binary_sensor.einfahrt_sensor_move.attributes["battery_level"]}}
        {% endif %}
      icon_template: > 
        {% set battery_level_xod1 = states.binary_sensor.einfahrt_sensor_move.attributes.battery_level|default(0) %}
        {% set battery_level_xod1 = (battery_level_xod1 / 10)|round * 10|int %}
        {% if as_timestamp(now()) > (as_timestamp(states.sensor.einfahrt_sensor_temp.last_updated) + 600) %}
          mdi:battery-unknown
        {% elif battery_level_xod1 >= 100 %}
          mdi:battery
        {% elif battery_level_xod1 > 0 %}
          mdi:battery-{{ battery_level_xod1 }}
        {% else %}
          mdi:battery-outline
        {% endif %}
      unit_of_measurement: '%'
```

ottelo9, 2019-06-26:
I made a change after few days. I changed this:

sensor.einfahrt_sensor_light to sensor.einfahrt_sensor_temp (Temperature)

in my code. Because if the light level is 0, then the motion sensor doesnt seem to send light update packages to deconz. But temperature packages comes every x minutes.

Rantaki, 2019-07-14:
I may have missed something, but I don’t understand how ottelo9’s solution is relevant for the problem. I don’t have problem (yet at least) with the logic used for determining whether the sensor is online or not. The problem is that deconz seems not to deliver the last update date and time (last connected) to Home Assistant. Without that it cannot be determined whether the device is working or not. Note that there is some last updated data available, but that is updated only when sensor value changes, i.e at the last water leak in my case - not very useful. The sensor seems to connect twice in a hour or so anyway (seen in deconz). That data should just be delivered to Home Assistant.