dresden-elektronik/deconz-rest-plugin #972: REST API exposes state.bri for smart plugs.

ebaauw, 2018-11-24:
Some smart plugs, like my favourite innr SP 120, provide a server _Level Control_ cluster, so the plug can respond to _Move with On/Off_ or _Step with On/Off_ commands sent by wireless switches.  They don't support variable output, though: _Current Level_ values 0 and 1 cause the device to turn off; any other value causes it to turn on.

The REST API plugin currently exposes `state.bri` for these plugs, but I don't think it should.  I currently suppress the creation of the _Brightness_ characteristic in homebridge-hue, but I think the creation of `state.bri` should be suppressed in the REST API plugin.  I think this should be based on the device type (e.g also for _On/Off lights_ with a _Level Control_ cluster, like the Busch-Jaeger RM01).

When creating a group of plugs (and/or _On/Off lights_, you don't want `action.bri` in the group either, see #415.

manup, 2018-11-24:
Some plugs like the Climax (also Bitron) power plug not only advertise but also do support dimming which is really nice to dimm classic lights bulbs.

I think this needs to be decided on per device base.

ebaauw, 2018-11-24:
> Some plugs [...] also do support dimming 

I'd hope they use a device type of _Dimmable plugin unit_ instead of _On/Off plugin unit_ or _Smart plug_.

closed 2019-07-29 by stale[bot]