dresden-elektronik/deconz-rest-plugin #973: xy values

ebaauw, 2018-11-24:
For the list: there's still a number of issues how the REST API plugin handled `xy` values:
1. Hue bridge cuts the API values to four decimals; the deCONZ REST API uses up to six decimals.
2. The API reports the ZigBee values divided by 65279 (0xfeff, the maximum allowed value for the _Current X_ and _Current Y_ attributes).  It should instead report the ZigBee value divided by 65535 (0xffff), as defined by the ZigBee standard (and as implemented by the Hue bridge).  When setting these values through the API, the API values should be multiplied by 65535 as well, but maximised to 65279 (corresponding to an maximum API value of 0.9961);
3. Some (non-Hue) lights hang when receiving a _Current Y_ value of 0 (I assume because of a divide by 0 bug in the firmware).  To workaround this, I think the REST API plugin should minimise the value to 1;
4. The web socket events contain the raw ZigBee values.  They should use the values divided by 65535 instead;
5. The `x` and `y` values are still issues through separate web socket events.  They should be combined in a single event (preferably with the other `state` attributes);
6. Typically, lights support way fewer colours than in the [0.0001, 0.0001] x [0.9961, 0.9961] range.  While there is a standard for a light to report their gamut, through the primaries in the _Color control_ cluster, not all manufacturers have implemented this.  The Hue bridge API now exposes this through `capabilities.control.colorgamut`, see https://developers.meethue.com/documentation/lights-api.  It would be nice if the deCONZ REST API would as well (currently I whitelist all Hue lights in homebridge-hue for this info).  I'm not sure, however, if the Hue bridge actually retrieves this info from the light, or whether it has whitelisted the Hue lights;
7. When requesting a non-supported colour, some lights display (Hue) the closest supported colour, while others (IKEA) display a default colour instead. Hue lights report back the actual colour displayed, other lights typically report back the requested colour.  The latter results in the API not reflecting the actual colour.  The former leads to weird behaviour: the API reflects the requested colour at first, but the colour changes "spontaneously" the next time the light is polled.  
[I need to test if this is still the case] To overcome this, the Hue bridge actually maps the requested colour to the closest in the light's gamut, updating the API values with these.  It doesn't (cannot) when setting colours through a group, though, so you'll see the API values (of the light) reflecting the requested values, until the light is next polled.  I have implemented the same logic in homebridge-hue (mapping the requested colour to the closest colour supported by the light), but it would be better to implement this in the deCONZ REST API plugin instead, so all apps can benefit.

ebaauw, 2019-01-18:
@manup, please take this into account while refactoring the colour logic.

ebaauw, 2019-03-15:
Commit fixes 1, 4, and 5.  It also combines the web socket notifications for a group `state` change.

> theses values are useless separate, but it s same for hue and sat

If `/config/websocketnotifyall` is set to `true`, each web socket notification for a light's `state` will contain all state attributes (also the ones that aren't changed).  If it's set to `false`, only the changed values are included in the notification.  Either way, your app needs to cache the previous value, to check whether the value has been changed, or to use the previous `sat` (`hue`) value when only `sat` (`hue`) is changed.  
Note that in my experience, `hue` and `sat` are useless, since they are relative to the light's supported colours, i.e. they don't match the RGB _Hue_ and _Saturation_ values on your device.  To make you app match the light actual colours, you need to use `xy` and translate the CIE 1931 values to RGB.

Smanar, 2019-03-15:
Right, for the moment, I never use Hue and Sat.
And thx for update ^^.

ebaauw, 2019-07-13:
Better keep this open for 6 and 7.

closed 2019-11-17 by stale[bot]