dresden-elektronik/deconz-rest-plugin #974: ZHASwitch capabilities

ebaauw, 2018-11-24:
Another one for the list: provide a `capabilities` like attribute (cf. #973) for ZHASwitch sensors, indicating which buttons and actions (`state.buttonevent` values) it supports.  Currently, I need to whitelist each switch in homebridge-hue for this information, but it is already available in the REST API plugin (although not explicitly).  It would be preferable to provide this information, for the benefit of all apps (I assume you need a similar whitelist in Phoscon as well).

wvuyk, 2018-12-01:
That would indeed help a lot, also for my plugin....

ebaauw, 2019-01-11:
How about something like providing an array with label and possible actions per button?  The associated `buttonevent` value would be the array position (starting at 1) * 1000 + the action.

E.g. (for the Hue dimmer switch):
```
{
  "capabilities": {
    "buttons": [
      { "label": "On", "actions": [0, 1, 2, 3] },
      { "label": "DimUp", "actions": [0, 1, 2, 3] },
      { "label": "DimDown", "actions": [0, 1, 2, 3] },
      { "label": "Off", "actions": [0, 1, 2, 3] }
    ]
  },
  "config": {
    "battery": 78,
    "group": "4604",
    "on": true,
    "reachable": true
  },
  "ep": 2,
  "etag": "812bcefe4fef1e89898bc73274d9f791",
  "manufacturername": "Philips",
  "mode": 1,
  "modelid": "RWL021",
  "name": "Living Room Dimmer",
  "state": {
    "buttonevent": 4002,
    "lastupdated": "2019-01-10T23:00:28"
  },
  "swversion": "5.45.1.17846",
  "type": "ZHASwitch",
  "uniqueid": "00:17:88:01:10:49:55:03-02-fc00"
}
```
or (for the Tråfri remote):
```
{
  "capabilities": {
    "buttons": [
      { "label": "On/Off", "actions": [2] },
      { "label": "DimUp", "actions": [1, 2, 3] },
      { "label": "DimDown", "actions": [1, 2, 3] },
      { "label": "Previous", "actions": [1, 2, 3] },
      { "label": "Next", "actions": [1, 2, 3] }
    ]
  },
  "config": {
    "alert": "select",
    "battery": 47,
    "group": "35438",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "49b8b0ca4d0b9a1bc68bb5da731ad1dd",
  "manufacturername": "IKEA of Sweden",
  "mode": 3,
  "modelid": "TRADFRI remote control",
  "name": "Living Room Remote",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2019-01-11T09:22:50"
  },
  "swversion": "1.2.214",
  "type": "ZHASwitch",
  "uniqueid": "00:0b:57:ff:fe:14:45:ca-01-1000"
}
```
or (for the `lumi.vibration`):
```
{
  "capabilities": {
    "buttons": [
      {"label": "1", "actions": [7, 8, 9] }
    ]
  }, 
  "config": {
    "battery": 100,
    "on": true,
    "pending": [],
    "reachable": true,
    "sensitivity": 1,
    "sensitivitymax": 21,
    "temperature": 2600
  },
  "ep": 1,
  "etag": "326387118598b7c38ffab873d5085c2e",
  "manufacturername": "LUMI",
  "mode": 1,
  "modelid": "lumi.vibration.aq1",
  "name": "Washroom Vibration",
  "state": {
    "buttonevent": 1007,
    "lastupdated": "2019-01-07T20:42:28"
  },
  "swversion": "20180130",
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:02:a5:21:24-01-0101"
}
```
For switches that don't follow the standard scheme of `buttonevent` values, like the Hue tab or the Xioami cube, we could (for now) simply omit the `capabilities` (or omit `capabilities.buttons`, and leave an empty `capabilities`).

manup, 2019-01-12:
I like that, but would suggest to use the full buttonevent values in the actions array. No calculations needed to figure out the actual event value. And a bit more future safe when it comes to actions > 9 like `1010`.

```
      { "label": "On", "actions": [1000, 1001, 1002,  1003] },
      { "label": "DimUp", "actions": [2000, 2001, 2002, 2003] },
      { "label": "DimDown", "actions": [3000, 3001, 3002, 3003] },
      { "label": "Off", "actions": [4000, 4001, 4002, 4003] }
```

>For switches that don't follow the standard scheme of buttonevent values, like the Hue tab or the Xioami cube, we could (for now) simply omit the capabilities (or omit capabilities.buttons, and leave an empty capabilities).

The Friends of Hue switches also use other numbers, like the Hue Tap (but more actions) albeit these can and should be mapped to 1002... schema. With explicit numbers like above even these could be supported in capabilities.

ebaauw, 2019-01-12:
> suggest to use the full buttonevent values in the actions array. No calculations needed to figure out the actual event value.

That makes a lot of sense, actually.  It's also a safeguard against breaking the order in the array.

> The Friends of Hue switches also use other numbers, like the Hue Tap (but more actions) albeit these can and should be mapped to 1002... schema.

They actually send the ZGP command code as `buttonevent` value.  Mapping makes sense for the ZGP switches, but not for e.g. the Xiaomi cube.

> With explicit numbers like above even these could be supported in capabilities.

In case of not mapping?  In that case, I would like an additional attribute to indicate whether they follow the "standard" buttonevent scheme, or something special (although for a ZGPSwitch, we could conclude that from the type).  Or maybe expose the cube's actions using a different attribute than `buttonevent` (but that would cause an impact to existing applications).

manup, 2019-01-12:
>Or maybe expose the cube's actions using a different attribute than buttonevent (but that would cause an impact to existing applications).

This might be the cleanest solution but I'm afraid too many people already using the current cube implementation so we can't break that (or support both).

The Friends of Hue switches are very similar to the Hue Tap (ZGPSwitch with a newer Enocean module) The module just support more actions like long press and release. With the initial commissioning telegram we can distinguish which switch it is.

I've added a few details here:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L890

I'd prefer mapping these commands to the 1002.. schema, but it would be interesting to know how the Hue api shows them, unfortunately the docs aren't updated yet.

ebaauw, 2019-01-12:
> but it would be interesting to know how the Hue api shows them

The Hue API just exposes the ZGP command the switch sends.  For the Hue tap:
- 34 = 0x22 = On/Off = press/release button 1;
- 16 = 0x10 = Scene 0 = press/release button 2;
- 17 = 0x11 = Scene 1 = press/release button 3;
- 18 = 0x12 = Scene 2 = press/release button 4;
- 98 = 0x62 = Press 1 of 2 = press buttons 1 and 2;
- 99 = 0x63 = Release 1 of 2 = release buttons 1 and 2;
- 100 = 0x64 = Press 2 of 2 = press buttons 3 and 4;
- 101 = 0x65 = Release 2 of 2 = release buttons 3 and 4.

See section A.4.1 of the ZGP spec:

CommandID | Command Name | Corresponding ClusterID | CommandID | Command Payload
-- | -- | -- | -- | -- 
0x00 | Identify | Identify | Identify | 0x003c
0x01 – 0x0F | Reserved
0x10 | Scene 0 | Scenes | View Scene | 0
0x11 | Scene 1 | Scenes | View Scene | 1
0x12 | Scene 2 | Scenes | View Scene | 2
0x13 | Scene 3 | Scenes | View Scene | 3
0x14 | Scene 4 | Scenes | View Scene | 4
0x15 | Scene 5 | Scenes | View Scene | 5
0x16 | Scene 6 | Scenes | View Scene | 6
0x17 | Scene 7 | Scenes | View Scene | 7
0x18 | Scene 8 | Scenes | View Scene | 8
0x19 | Scene 9 | Scenes | View Scene | 9
0x1A | Scene 10 | Scenes | View Scene | 10
0x1B | Scene 11 | Scenes | View Scene | 11
0x1C | Scene 12 | Scenes | View Scene | 12
0x1D | Scene 13 | Scenes | View Scene | 13
0x1E | Scene 14 | Scenes | View Scene | 14
0x1F | Scene 15 | Scenes | View Scene | 15
0x20 | Off | On/Off | Off | N/A
0x21 | On | On/Off | On | N/A
0x22 | Toggle | On/Off | Toggle | N/A
0x23 | Release | 
0x24 – 0x2F | Reserved
0x30 | Move Up | Level Control | Move Up |  
0x31 | Move Down | Level Control | Move Down |  
0x32 | Step Up | Level Control | Step Up |  
0x33 | Step Down | Level Control | Step Down |  
0x34 | Level Control/Stop | Level Control | Stop | N/A
0x35 | Move Up (with On/Off) | Level Control | Move Up (with On/Off) |  
0x36 | Move Down (with On/Off) | Level Control | Move Down (with On/Off) |  
0x37 | Step Up (with On/Off) | Level Control | Step Up (with On/Off) |  
0x38 | Step Down (with On/Off) | Level Control/ | Step Down (with On/Off) |  
0x39 – 0x3F | Reserved
0x40 | Move Hue Stop | Color Control | Move Hue | Stop
0x41 | Move Hue Up | Color Control | Move Hue Up |  
0x42 | Move Hue Down | Color Control | Move Hue Down |  
0x43 | Step Hue Up | Color Control | Step Hue Up |  
0x44 | Step Hue Down | Color Control | Step Hue Down |  
0x45 | Move Saturation Stop | Color Control | Move Saturation | Stop
0x46 | Move Saturation Up | Color Control | Move Saturation Up |  
0x47 | Move Saturation Down | Color Control | Move Saturation Down |  
0x48 | Step Saturation Up | Color Control | Step Saturation Up |  
0x49 | Step Saturation Down | Color Control | Step Saturation Down |  
0x4A | Move Color | Color Control | Move Color |  
0x4B | Step Color | Color Control/ | Step Color |  
0x4C – 0x4F | Reserved
0x50 | Lock Door | Door Lock | Lock Door | N/A
0x51 | Unlock Door | Door Lock | Unlock Door | N/A
0x52 – 0x5F | Reserved
0x60 | Press 1 of 1 | N/A
0x61 | Release 1 of 1 | N/A
0x62 | Press 1 of 2 | N/A
0x63 | Release 1 of 2 | N/A
0x64 | Press 2 of 2 | N/A
0x65 | Release 2 of 2 | N/A
0x66 | Short press 1 of 1 | N/A
0x67 | Short press 1 of 2 | N/A
0x68 | Short press 2 of 2 | N/A
0x69-0x6f | Reserved
0x70-0x9f | Reserved
0xA0 | Attribute reporting | Copied from the triggering GPD command | ZCL Report attributes command | Copied from the trigger- ing GPD command
0xA1 | Manufacturer-specific attrib- ute reporting | Copied from the triggering GPD command | ZCL Report attributes command | Copied from the trigger- ing GPD command
0xA2 | Multi-cluster reporting | Copied from the triggering GPD command | ZCL Report attributes command | Copied from the trigger- ing GPD command
0xA3 | Manufacturer-specific multi- cluster reporting | Copied from the triggering GPD command | ZCL Report attributes command | Copied from the trigger- ing GPD command
0xA4 | Request Attributes | Copied from the triggering GPD command | ZCL Request attributes command | Copied from the trigger- ing GPD command
0xA5 | Read Attributes Response | Copied from the triggering GPD command | ZCL Read attributes response com- mand | Copied from the trigger- ing GPD command
0xA6 – 0xAE | Reserved
0xAF | Any GPD sensor command (0xA0 – 0xA3) | Copied from the triggering GPD command | ZCL Report attributes command | Copied from the trigger- ing GPD command
0xB0-0xDF | Reserved
0xE0 | Commissioning | N/A
0xE1 | Decommissioning | N/A
0xE2 | Success | N/A
0xE3 | Channel Request | N/A
0xE4 – 0xEF | Reserved

manup, 2019-01-12:
Yes but is it the same for Friends of Hue switches?

ebaauw, 2019-01-12:
Yes.  Unfortunately, I cannot find the reference where I read that.  Probably somewhere on the Hue developer forum.

manup, 2019-01-12:
What a pity the other mapping is much nicer and generic. So we can be either be compatible or generic.

carloscae, 2019-03-24:
Hi guys,
I got a Friends of Hue Niko switch this week. Couldn't pair on Phoscon (tried as Hue tap), no problem as I have no urgency. Just letting you know I have it available in case I can be of some help on the integration. Cheers.

ebaauw, 2019-03-24:
> Apparently it's only pairing with to the coordinator and won't work if i move far from it.

Yes, unlike the Hue bridge, the REST API plugin doesn't (yet?) configure the ZGP proxies (Hue lights and other routers providing a ZGP 0xF2 endpoint) to forward the ZGP commands to the coordinator (ZGP sink).  See https://github.com/dresden-elektronik/deconz-rest-plugin/pull/396#issuecomment-363517377.  Effectively, the ZGP switches need to be in direct range of the RaspBee/ConBee.

> Button assignment on groups also won't work

For this to work, a ZGP sink needs to be configured to translate the ZGP codes to the corresponding ZCL commands.  The interesting case would be if we can configure a router (other than the coordinator) to do this; if only the coordinator will do this, you might as well use rules based on the buttonevent values, as the switch won't do anything when deCONZ is down.

> I get events (on Hassio) of every button + top 2/bottom 2 simultaneously (press and release events in every case).

That looks suspiciously like the Hue tap.  Could you check which `buttonevent` values?  Are they the same as the Hue tap?  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/974#issuecomment-453758775.  Could you double-check that deCONZ reports the same values as the Hue bridge?

> Pairing procedure: Hold top left for 10s, then press lower left+top right simultaneously.

That's different from the Hue tap, which just needs a hold for 10 seconds.  Which button depends on the Zigbee channel used.  I suspect the pairing sequence of the Niko switches might be different for different Zigbee channels as well.

Kane610, 2019-03-26:
Can we change "On/Off" to "Toggle"?

When can we see this in a release? :D

carloscae, 2019-03-26:
Hey @ebaauw. Sorry for the late reply (was out of town).

So, apparently I was clicking buttons too fast, simulating Hue app pairing procedure, but it actually only requires to the Top Left button to be held for 10 seconds. 

As for the button event values (first value is PRESS, second is RELEASE)

- Top Left: 16, 20
- Bottom Left: 17, 21
- Top Right: 19, 23
- Bottom Right: 18, 22
(Weird order, I double checked.)
- Top Left + Top Right: 100, 101
- Bottom Left + Bottom Right: 98, 99

So, quite similar to the values you linked. Just one button not matching.

ebaauw, 2019-05-05:
> but it actually only requires to the Top Left button to be held for 10 seconds.

You need to hold one button for 10 seconds to get the switch to pair.  Which button depends on the ZigBee channel used.  If they're like the Hue tap:
- Top right (1 dot) is for channel 11;
- Top left (2 dots) is for channel 15;
- Bottom left (3 dots) is for channel 20;
- Bottom right (4 dots) is for channel 25.

> Weird order, I double checked.

They're numbered in a circle, counterclockwise, just like the Hue tap.  With the Hue tap enclosure you cannot physically press top right.  It issues ZGP command 34 (_Toggle_) when the engergy harvesting mechanism is pressed, with top right as well as without any of the buttons.  I wonder if the Friends of Hue will issue _Tiggle_ without any button pressed as well.  Otherwise, the Hue tap issues the same buttonevents as the Friends of Hue switches, except for the release on the single buttons.  In particular, it will issue the 98-101 when pressing/releasing both bottom/top buttons (but you cannot physically press 2 dots and 1 dot because of the enclosure).

ebaauw, 2019-05-05:
Support for FoH switches was added in https://github.com/dresden-elektronik/deconz-rest-plugin/commit/5a3767b7c3d6802d619368a63b5f856549d7a1e0, which should be included in next version 2.05.56.  I'm not sure what the status is for configuring the routers' ZGP proxy function.  @manup?

In the next version of homebridge-hue I'll include support for FoH switches, as well as supporting the two-button presses for the Hue tap.  See https://github.com/ebaauw/homebridge-hue/issues/466.

> Can we simplify strings; change "On/Off" to "Toggle", "DimUp" to "Up", "DimDown" to "Down"?
> When can we see this in a release? :D

As of API v1.30, the Hue bridge now reports the switch capabilities, see https://developers.meethue.com/develop/hue-api/supported-devices/ (scroll down to section 4) and the listings below.  I have no idea what device uses the ZLLRelativeRotary sensor type.  Afaik, only the Living Colors and Living Whites remotes (pre-dating Hue) had dials.  The example seems constructed manually (it has a fake mac address).  Also, I don't see the `archetype` attribute (which would hold the button label), even though I'm on v1.31.

I wouldn't call the Hue bridge scheme "simple", but I would prefer deCONZ to expose the same scheme, rather than having to support two separate schemes.  This would also help dealing with deCONZ exposing the FoH switches with regular (?) `buttonevent` values instead of the ZGP command codes.  Maybe we could expose the 03 endpoint of the Xioami Cube as ZLLRelativeRotary (ZHARelativeRotary) sensor?  @manup, @Kane610, @wvuyk, what do you think?

Below the REST resources (my sorting).

For a Hue tap:
```json
{
  "capabilities": {
    "certified": true,
    "inputs": [
      {
        "events": [
          {
            "buttonevent": 34,
            "eventtype": "initial_press"
          }
        ],
        "repeatintervals": []
      },
      {
        "events": [
          {
            "buttonevent": 16,
            "eventtype": "initial_press"
          }
        ],
        "repeatintervals": []
      },
      {
        "events": [
          {
            "buttonevent": 17,
            "eventtype": "initial_press"
          }
        ],
        "repeatintervals": []
      },
      {
        "events": [
          {
            "buttonevent": 18,
            "eventtype": "initial_press"
          }
        ],
        "repeatintervals": []
      }
    ]
  },
  "config": {
    "on": true
  },
  "diversityid": "d8cde5d5-0eef-4b95-b0f0-71ddd2952af4",
  "manufacturername": "Philips",
  "modelid": "ZGPSWITCH",
  "name": "Hue tap switch 1",
  "productname": "Hue tap switch",
  "state": {
    "buttonevent": 34,
    "lastupdated": "2019-05-05T11:35:06"
  },
  "swupdate": {
[... 186 lines cut ...]
  },
  "config": {
    "battery": null,
    "on": true,
    "pending": [],
    "reachable": true
  },
  "diversityid": "73bbabea-3420-499a-9856-46bf437e119b",
  "manufacturername": "Philips",
  "modelid": "RWL021",
  "name": "Hue dimmer switch 1",
  "productname": "Hue dimmer switch",
  "state": {
    "buttonevent": null,
    "lastupdated": "none"
  },
  "swupdate": {
    "lastinstall": "2019-05-05T12:29:53",
    "state": "noupdates"
  },
  "swversion": "5.45.1.17846",
  "type": "ZLLSwitch",
  "uniqueid": "00:17:88:01:10:49:55:03-02-fc00"
}
```

carloscae, 2019-05-07:
Hey guys, keep in mind those switches can be mounted on a 2 button scenario, which means would be good to get the click/release of the combo buttons top left/right and bottom left/right.

related image:
https://www.vimar.com/cache/images/meta-article/w1200h627/vimar-friends-of-hue-philips-lighting-comandi-luci-dimmer-on-off-63zm67b.jpg

ebaauw, 2019-05-07:
The Hue bridge will issue the corresponding buttonevents (98, ..., 101) even though they don’t include them in the capabilities.  Also for the Hue tap (some people take the module out of the enclosure and place it in a wall switch).

closed 2019-09-11 by stale[bot]

ICTSMART, 2020-06-02:
Hi Guys,

Today I started migrating from HUE bridge to RaspBee II. Everything went well, apart from the NIKO FOH switches. After long trying all buttons for pairing (despite all information here), I managed to get one paired. But after pairing no events appear at my RaspBeeII at all. So deconz nicely shows a switch (2 button version?) but thats it. I am running with the following versions: 
Version 2.05.77 / 22-5-2020
Firmware 26520700
RaspBee II

Any ideas/suggestions are highly appreciated.

ICTSMART, 2020-06-02:
Apparantly, commissioning is essential to not only have the switch registered, but also to keep transmitting on the appropriate channel ;)
See "Complete commissioning" at https://phoscon.de/en/support#pairing-friends-of-hue-switch