dresden-elektronik/deconz-rest-plugin #976: Discovery issue, port undefined, login fails

markuspf, 2018-11-25:
I updated to the most recent deCONZ (`2.05.48`).

I am not a JS expert, so this report might be inaccurate.

Login to the Phoscon webapp fails (Login error.), whereas the old webapp works. After some stabbing around with the Firefox JS debugger I noticed that app.gw showed as `192.168.10.12:undefined` (my deCONZ runs on port 80, but in a different subnet from the network I am accessing it from), editing out ports in `pwa/login.js` made the webapp work, but this is obviously not the correct solution.

I guess somewhere there's an `if port is undefined don't store it` missing or similar.

manup, 2018-11-25:
Thanks for the report, yes normally this should be handled seems there is something missing.
I'll check it out.

manup, 2018-11-26:
There were indeed bugs, already fixed in http://dresden-elektronik.de/pwa and will be part of 2.05.49 too.

dergreg, 2018-11-30:
Any updates on this?

I updated my old deconz installation via dpkg and updated to the latest firmware. Now I'm not able to login into the PWA app anymore, the old password does not work anymore.

Also, a password reset does not work because it always says the gateway wasn't restarted. I restarted the deconz.service and also rebooted my PI.

I can login via delight/delight into the old app, but not into the new one.

Also gateway discovery only works in Chrome and not in Safari. Looking into the developer console it seems like it's trying to find the gateways via HTTPS and then fails due to a certificate error.

manup, 2019-01-16:
The issues were fixed in recent versions, closing the issue for now.

closed 2019-01-16 by manup