dresden-elektronik/deconz-rest-plugin #977: ConBee Stick does not find lights / plugs

LordVoodoo, 2018-11-25:
Hello,

I have set up the Deconz Gateway on a Z83/Debian in a Docker environment.
Unfortunatelly, I cannot add any lights. At the moment, I have tried two OSRAM Power plugs, both are recognized by the Philips Hue Bridge (v2), but not by DeConz.

I have also tried to find the plugs using the DeConz Software for Windows. Also no success.

Any idea how to continue?

Kane610, 2018-11-27:
Have you actually got deconz to find your conbee? Do you get a proper channel that is not 0?

manup, 2018-11-27:
Please update the firmware to the recent version as described here: https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

In some cases it is needed to use a USB extension cable between ConBee and PC to prevent signal distortion.