dresden-elektronik/deconz-rest-plugin #979: deCONZ ConBee setup and config when more serial devices connected to RPi

moritzthecat, 2018-11-26:
on same RPi3+ where deCONZ is installed I also have two 1-wire master connected via USB with serial/by-id device recognition. fhem is used to control the 1-wire devices. this is working fine and running stable. further fhem is using the deCONZ REST API to access the ZigBee devices. I am planning to add another USB serial device to the RPi to read my energy meter. 

since I am experiencing some sporadic problems with deCONZ (Version 2.05.48 / 18.11.2018, Firmware 262F0500) loosing the connection to the gateway and experiencing ttyUSB0 disconnects, I am not sure if all changes I made are correct and surviving reboot. So far a gateway restart in Phoscon is getting the connection to the gateway back. 

/boot/config.txt 
```
# uart required for deconz
enable_uart=0
dtoverlay=pi3-miniuart-bt-overlay
#
# disable Wifi and BT
dtoverlay=pi3-disable-wifi
dtoverlay=pi3-disable-bt
```

/lib/systemd/system/deconz.service
```
...
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 --dev=/dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_aaabbbcccddd-if00-port0
Restart=on-failure
...
```

for firmware updates I am using the manual update path where I need to disconnect the two 1-wire master from the USB ports - so FTDI driver can be unloaded. If possible I would like to improve the setup.

since I could not find a detailed description to setup deCONZ ConBee using USB devices by-id (e.g. config file ?) I would appreciate a hint where to look or some guidance to make this work.

manup, 2018-11-27:
We are working to improve updates in deCONZ and GCFFlasher so that the connected device is to be chosen and hence multiple USB devices are no issue anymore also some tricks to powercycle the device into bootloader mode will be ruled out.

Since it's a bigger tasks this will take a while ETA is for early 2019.

moritzthecat, 2018-11-28:
ahhh, super. appreciate if a config file for deCONZ could be introduced. 

In mean time if you could pls. check /adapt if above settings need modifications ?

A320Peter, 2018-12-08:
I am also excited about this. I have a Z-Wave stick in my Raspi together with ConBee. I didn't know about this problem before I bought the Z-Wave stick. Unfortunately if there is a power cut or Raspi needs to be rebooted for any reason there is a very good chance that ConBee will be allocated to some other ttyUSB port and won't work. I don't always have physical access to my Raspi to unplug the USB sticks and plug in ConBee first during the reboot so this issue is very critical.

With Linux 'udev' command it is possible to create a symlink for ConBee based on the stick's serial number and product id and assign a static device name which never changes. However we still need to know where is the config where the software is looking for the USB port ttyUSB0? So we could modify it to the symlink name.

closed 2019-04-14 by stale[bot]